betaflight/betaflight #7796: Concerns about the D_MIN and Absolute Control defaults

Cleric-K, 2019-03-15:
I had some difficult experiences with tuning my quad.

I'll save you the whole process that led me to the conclusions but just present you with the results.

### 1. D_MIN
My whole journey started while trying to eliminate bouncebacks after flips and rolls. It turned out that it is a _cross-axis_ bounce back - for example I make a _flip_ but at the end I have a _roll_ wobble (and yaw, but let's focus on roll).

[image]

In the above example we have the end of a back flip. The back motors fire up to stop the flip. For whatever reason, they produce unequal moment around the CG which introduces parasitic roll (`rcCommand`s on roll and yaw are zero).

> The new DSHOT RPM Telemetry opens our eyes of why exactly this happens:
> [image]
> Because of motor delay at the moment when the two motors are commanded to kick in, m1 is still at _higher_ RPM than m3. The kick commands are very steep so the motors have to apply maximum acceleration. m1 has a kind of head start because of its higher initial RPM, so m1 keeps being ahead of m3 while both accelerate at their maximal rate (this rate should be more or less the same for same motors/props).
> For all this time m1 produces _higher_ thrust than m3 which results in the parasitic roll (and yaw) effect.
>
> (It will be interesting to see in the future if the RPM data could be used in the PID loop: for example, if the PID controller wants to give the motors equal sharp commands but the motors have different starting RPM, it could try accelerate the faster motor a little _less_ until they equalize and then continue with full acceleration for both)

This is normal but my problem was that no amount of tuning on the roll axis allowed me to arrest the parasitic roll in time - it always grows way too big such that it is very visible on the camera.

After many tests I suspected `d_min`. For my budget build I have to use somewhat higher PID values (in the 60-70 range).

Since at zero gyro/setpoint, dmin keeps D at the min value (which is 20 by default), this means that at the moment the parasitic roll is initiated we start with D=20 which for my build is way too low to have any effect. By the time the gyro grows enough that D has some more effective value, the roll is already quite pronounced.

After I set `d_min=0` I was able to tune my D high enough that it was perfectly capable to arrest the parasitic roll before it grows too much. I know that I can simply set `d_min` to a value little less than D and it will be OK, but that's not my concern (as I'll explain at the end).

### 1. Absolute Control

Yet, even after I've resolved the mystery with `d_min` I still got from time to time horrible bouncebacks. After some debugging it turned out that the `setPoint` values assume values that are somehow offset from the `rcCommand`:

[image]

Here again we have the end of a back flip. Note that the `rcCommand` is correct - it stops at zero - but the `setPoint` overshoots!

I looked at the sources and the only places I found that have the power to modify the `setPoint` was the `accelerationLimit()` and `applyAbsoluteControl()`, of which only the latter seems to be capable of such offsetting effects.

I did some tests on the floor (without powering the motors, just short logs with stick comands, with zeroed I to avoid windup) with and without AC and it really does modify the `setPoint`.
Here is with AC:

[image]

Note how even that the sticks are centered (near 1500us) there's some offset in the actual `setPoint`.
After disabling AC the behavior is as expected:

[image]

When `rcCommand` is zero, the `setPoint` is also zero.

I know this is not a relevant test for AC itself (because of the artificial setting), but it confirmed to me that AC has the power to modify the `setPoint`. So I think this explains the `setPoint` overshoot at the end of the flip above.

I have not yet had the chance to fly and confirm this (sorry), but I decided to raise attention to this as soon as possible in the face of the nearing release date.

In my previous logs I didn't observe such `setPoint` modifying behavior but I was running nightly build there. Last test (where the overshoot was seen) was with RC1. Sure enough I found the reason in #7718 - the `abs_control_gain` was changed from `0` to `10`.

Maybe I'm in a corner case with my budget build that needs higher than default PIDs (and thus the default `d_min` was no enough) but the amount of tests, deep debugging in bb logs and even reading into the source code that I had to do, makes me worried about the user experience for people who have more practical approach to tuning.

Usually things like bouncebacks will be attacked by PID tuning but in the case of `d_min` and `AC` no amount of PID tuning alone is able to solve the problem. My concern is that it is hard (at least from my experience) to reach the conclusion that these settings are _preventing_ to reach a good tune.

Unfortunately I don't have a clear idea at this point what should be done.
I _don't_ understand `Absolute Control` and `ITerm Rotation` (didn't have the time yet do delve into the details). So I don't understand why it causes this `setPoint` overshoot at the end of the flip. Maybe it's a matter of safer AC defaults.

As of `d_min` I guess it is more a matter of education. I hope that there'll be some videos soon to highlight the importance of _setting it too low_. Otherwise, for people who just flash 4.0 without reading the changelog too much, it could be insurmountable obstacle to tune their D if their build requires higher value but don't realize that `d_min` is keeping it low in certain cases.

McGiverGim, 2019-03-15:
I didn't have any problem with d_min after some tuning, but I have problems too with Absolute Control: after each roll I have a wobble in the pitch axis. The strange is that a lot of people has tested the defaults in the Slack chat and until now I was the only one with problems. Now it seems that we are two :)

joelucid, 2019-03-15:
Absolute control does input setpoint corrections if the quad doesn't follow the setpoint in a timely manner. It has something similar to iterm relax built in - so the axis which has the primary move should not get overshoot. But if there is crosstalk abs control will correct an error which can manifest as bounce back.
Unfortunately in this these pics the scale isn't visible so it's not possible to say whether the response on pitch was adequate from ACs perspective. If @Cleric-K could post the log I can verify if that is the case.

Cleric-K, 2019-03-15:
Here you go: [ac_overshoot.zip](https://github.com/betaflight/betaflight/files/2972805/ac_overshoot.zip)

Unfortunately at that time I didn't yet know what the problem was, so I didn't have the ITERM_RELAX debug mode turned on. If I have the chance to fly this weekend I'll try to record with debug.

I can confirm that the overshoot happens only with very rapid commands (maybe I should have mentioned that at the beginning). Slower flips/rolls are not affected.

loomerco, 2019-03-15:
@joelucid you have my log as well in you slack PM I’ve had very similar experience. Ended up with AC gain of 1 which now seams ok:
https://betaflightgroup.slack.com/archives/DF348DRQQ/p1551971910001400

jcutcher01, 2019-03-17:
Great report @Cleric-K ... following...

Cleric-K, 2019-03-17:
@joelucid here are some more tests: [los_test.zip](https://github.com/betaflight/betaflight/files/2975585/los_test.zip)

That was a quick LoS test with default PIDs (only d_min disabled) with and without AC.
Also two tests with my more or less good PIDs (again no d_min) with and without AC.
All use `debug_mode = ITERM_RELAX`.

I still don't understand AC, but from what I see the more pronounced problems happen when motors max out. My motors are Samguk 2206 2400KV, which seems to be not powerful and responsive enough when stopping flips and rolls (my max angular speed is 720deg/s - rate=0.9, super=0.75).

Let me clarify: as probably all know, stopping rolls and flips require more sharp motor commands (unless we use completely linear rates) because when returning the stick from its max position we are at the very steep part (super or expo) of the rates curve. Thus the setpoint moves very rapidly in the order of several hundreds deg/s which is simply impossible for the (my) motors to follow that quickly. That's why initiating flips/rolls have more modest motor commands (since the setpoint increases more gradually through the middle of the rates curve) and much more sharper motor commands when stopping.

So, from what I see, the AC axisError accumulates exactly at these flip/roll stopping maneuvers because the gyro has such a hard time following the very fast setpoint change.

I suppose that people with better motors might not have such problems because I imagine that these high-end motors are capable in following even such fast setpoint changes and thus no excessive AC error accumulates.
With this in mind it will be interesting to compare what motors are using people who do not experience problems and people like us who do experience problems.

In a nutshell, I believe that AC overcorrection occurs at these moment where it is impossible for the motors to follow the setpoint (they are already maxed out). In that case AC pushing the setpoint even more (hoping that that will help accelerate the rotation) is of no use.
I might be totally wrong in the above (sorry again for not yet taking the time to study AC in depth).

(I'm planning to experiment with the _setpoint acceleration limiter_. It might be useful in my case)

Cleric-K, 2019-03-19:
@joelucid 
Just want to check if I get the basics of AC right:
In very rough overview I think that AC basically tries to make the integral of the gyro to be equal to the integral of the setpoint.
As an simplified example: let's imagine rectangular rcCommand (instant rise and fall) which holds 360deg/s for 1 second. The integral of this command would be exactly 360 degrees or one full rotation of the craft. The goal would be to make sure that the integral of the gyro is also 360 degrees by slightly correcting the setpoint and Iterm if needed. Do I get this right?

joelucid, 2019-03-20:
One thing you could try is to set the pidsum_limit to 400. That gives the motors a bit of headroom to address crosstalk without air mode having to scale the motor mix down to 1. This has helped me in similar cases.

Abs control ensures that the attitude of the quad is brought to were it would have been without axis crosstalk. So if there has been some it means a bit of a bounce back will occur to fix attitude. So we should focus on reducing this cross talk.

klutvott123, 2019-03-21:
When experimenting with absolute control in BF 3.5 I had a similar experience. After flips and rolls the quad would do the wobble dance. This was on a quad with weak racerstar 2205 2300KV on 3S.
Since AC became enabled by default in BF 4.0 I have been using it again. This time it's the same frame, but with T-motor F40 pro II 2400KV on 4S and the wobble isn't that noticeable. Maybe it's just able to do the correction quicker because of the more powerful setup. I do notice the effect of absolute control and wouldn't be without it now. 

I also have concerns about D min being enabled by default. For the relatively powerful setups a lot of people have, the default D gains might just be too high, but because of D min they seem fine in normal flight. If I understand correctly, the purpose of D min was to be able to have lower D gains and less noise during normal flight, not higher gains during flips and rolls. Don't get me wrong, I do appreciate the work that has gone into this feature, and it works well on my weaker backup quad with default settings, but on my main quad it's the first thing to be disabled. IMO D min disabled and more conservative D gains might have been better as default. Then people could just tune D as normal, and those that take the time to read the release notes/tuning guide can enable D min if they want.

joelucid, 2019-03-21:
Yeah I think you have a valid point that dmin might reduce the quads ability to react to external disturbances like this crosstalk. You might not have to turn it off - there are ways to configure it to be activated earlier, @ctzsnooze can likely make a suggestion. 

I’d be grateful if you could check whether pidsum_limit=400 makes a difference for you. It does for me and might be appropriate for the defaults.

McGiverGim, 2019-03-21:
The D Min can be used in both ways:
1. To reduce noise. In this way you will have very low D Min and you will suffer some more propwash.
2. To stop high speed rolls (useful for freestyle). In this way you need a "normal" value of D for D Min and a very high value in D to help stop this kind of movements. The noise will be high.

About Absolute Control, I have tested defaults with two quads (both budget components, one old, with racerstar 2205 motors, the other more new, with dys 2306 motors). Both of them have a pitch wobble each time I make a fast roll. Disabling AC the wobble is gone.

Cleric-K, 2019-03-21:
I'll try to test `pidsum_limit` and `acc_limit` this weekend If I get the chance.

I still think that the problem is that weaker motors spend much more time with big difference between *gyro* and *setpoint* (and it's only thanks to Iterm relax that this does not cause big I accumulation). So with this in mind, limiting the pid sum will likely increase the error even further, maxing out the axis error of AC. That's why I was thinking about `acc_limit` because this should help the *gyro* track better the thus limited *setpoint*. But we'll see, maybe I'm missing something in this thought line.

About d_min.
@ctzsnooze here're some thoughts. I would be happy to hear everyone's opinion.
* Would it be reasonable to be able to boost D not on per axis basis, but rather getting the _largest_ factor for all axes and then boost each axis' D with _that_ factor? This might be able to help in cross-talk. For example, currently when performing a flip, the factor in pitch will be high, but on roll - zero - thus the roll axis is not capable of addressing quickly any cross-effects (only as soon as it's gyro starts to rise and the factor with that). On the other hand, if the _max_ factor is used, roll will get pitch's bigger factor.
* Making a factor also from throttle setpoint movements. The motivation for this is quick throttle changes, which produce nose dips/jumps. Classically, the advice is to use _antigravity_, but truth is that D is much quicker to react to such disturbances. But with no stick movement, wobbles caused by quick throttle changes would be addressed efficiently only when the gyro rises enough that D can be boosted to adequate value.

Unfortunately, the way d_min is implemented now would require some ugly changes if the above is to happen. d_min is nicely embedded now in the per-axis `for()` loop. To be able to get the _max_ factor that loop would have to be somehow separated in two parts - once, to survey the max factor, and second to apply it to each axis' D.

But anyway, the question is whether this seems reasonable at all.

joelucid, 2019-03-21:
@Cleric-K, acc_limit is not really a good idea. I would not use that. The problem is not that the gyro can't follow the set point. We have solved that with iterm relax. It's rather that crosstalk on the other axis does not have relax enabled but doesn't have more motor range available for the correction. Airmode knows how to deal with that and will scale the motor mix so that the correction still goes to the motors. But this scaling shrinks all pids and thus makes the correction less effective. And also since at least one motor is maxed out only one motor will now have the burden of the correction.

Here's an example what it looks like with pidsum_limit at 400. As you can see now the red and blue motors both get a correction.

[image]

Cleric-K, 2019-03-21:
Aha! Great explanation. That graph really tells it.

Thank you! I'll test at first opportunity.

klutvott123, 2019-03-25:
@joelucid I tried pidsum_limit = 400. Can't see any difference. Didn't see any decrease in performance either.

joelucid, 2019-03-25:
We've found a bug in abs control. the binaries here: https://github.com/joelucid/jflight have the fix. PR is here: https://github.com/betaflight/betaflight/pull/7866. So far this seems to solve the wobbles. Would be cool if you could take it for a spin!

klutvott123, 2019-03-25:
@joelucid Yes, I noticed that one. I will give it a try as soon as the weather allows it. Weather report says snow and wind for this week at least. I'll look for an opening. 😄

Cleric-K, 2019-03-25:
@joelucid Great! I'll have to try it also, but that will be on the next weekend at best :/ Hope other's will be able to test before that.

I know it's been confirmed already by @klutvott123 but just for the record, here's my results also (with `pidsum=400`:
[image]

klutvott123, 2019-03-26:
I tried the latest absolute control fix today. Seems to be working. Tested with absolute control on and off. I used to have a nasty wobble on pitch axis when doing 180 turns at high speed which seems to be gone or greatly reduced now. Didn't do any logging, but i suspect @Cleric-K will have some data for us. 😸

jcutcher01, 2019-03-26:
That’s great news. I still have not been able to fly yet. With this abs ctrl fix is it still ok to run pidsum_limit at 400 or bring it back to default?

closed 2019-04-22 by Cleric-K