betaflight/betaflight #7797: BF 4.0 rc2 GPS rescue = drop when < 50m from home instead of landing

laurentopia, 2019-03-15:
**Describe the bug**
When switch mode to GPS rescue, the quad drops and shows flight summary like a disarm

**To Reproduce**

set aux3 to gps rescue
fly a little
switch on gps rescue

**Expected behavior**
Land even if <50m from home

**Flight controller configuration**
https://pastebin.com/4KUkC3bZ

**Setup / Versions**
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Mar 14 2019 / 11:28:29 (360afd96d) MSP API: 1.41
BN-880 properly connected and recognized with steady 12 sats
BN-880's mag also connected
VTX leaf

etracer65, 2019-03-15:
Yes, this is normal. Please read the documentation.

https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode

closed 2019-03-15 by etracer65

laurentopia, 2019-03-15:
This behavior makes it difficult to test GPS reliability in small area before moving to the field.
Also if the rescue mode is triggered above home it will drop, which is dangerous.
Also the ergonomy of a function which has two diametrically opposite behaviors is highly questionable.
Therefore, for sake of consistency, safety and testability, it is better that landing behavior be kept regardless of distance to home.

etracer65, 2019-03-15:
No, it would be extremely dangerous to have it activate close to people - namely the pilot. Also GPS Rescue doesn't "land" and there is no "landing behavior". It simply flies back into range so you can retake control. After it passes the predefined distance, it starts to descend in a slope as it approaches the home point and will intentionally crash. It cannot land as it is impossible to determine heading unless moving forward when using GPS. So it can't hover and descend.

laurentopia, 2019-03-15:
I agree that a quad moving at a diagonal autonomously is very dangerous.

But a quad descending at a controlled speed with no horizontal movement is less dangerous than a quad dropping and reaching terminal velocity.
So instead of descending at a diagonal, BF could fly horizontally to a point above home then descends vertically at a set speed.

mikeller, 2019-03-15:
@laurentopia: What part of 'it can't hover and descend' do you not understand?

laurentopia, 2019-03-15:
@mikeller there is no need to be snarky, I simply missed that part of the explanation. Heading can be extracted from the MAG, which BF can read.

etracer65, 2019-03-15:
So what do we do with the 99.9% of the flight controllers that don't have a MAG? That's why GPS Rescue was designed to work without one.

It's an open source project. If you think you can do it better or have ideas to improve, feel free to submit a pull request.

laurentopia, 2019-03-15:
Will do.
Reading imu.c ... it'll take me a while to even understand how it all fits together, what's the best channel to ask about the code?

mikeller, 2019-03-16:
@laurentopia: Try #general in the Betaflight Slack (registration at http://slack.betaflight.com/).

krzysztofmatula, 2019-03-31:
Today similar case happened to me while trying to test GPS rescue.
I engaged the GPS rescue at 95m from home, which resulted in immediate disarm and crash.
I have `gps_rescue_min_dth` set to 100 (default).

When during failsafe I can agree that crashing may be better than hitting the pilot.

But when testing and activating the GPS rescue by a switch (and fully controlling the machine) I'd rather prefer the mode to be ignored than immediately causing disarming and crashing...

What do you think?

etracer65, 2019-03-31:
I think you successfully tested the behavior of activating it closer than your configured `gps_rescue_min_dth`. It did exactly what it was supposed to.

Jahnkeanater, 2019-04-14:
I don't think disarming the quad and showing a flight statistics page it a great idea.

If GPS rescue is accidentally triggered with a switch I think keeping the quad armed, Leveling out and showing a warning on the OSD is a better idea. Maybe disarm after a few seconds but not immediately.

To recover from this the pilot would have to disable GPS rescue, disarm and rearm mid air. The pilot should be able to recover by simply disabling GPS rescue mode.

krzysztofmatula, 2019-04-14:
I'd also vote for not disarming when GPS rescue is triggered by switch.
Just doing nothing would be fine. Only showing a warning would be even better.

Lebensgefahr, 2022-08-24:
I crashed my three drones of this great settings. It is really hard to watch of the distance to home point when you are nervous about broken video signal. This function should work as RTH only and should not to make the drone drop.  It can do nothing if the distance is lower than 50 m but do not drop. It is really annoying.