betaflight/betaflight #7798: Feedback Oscillations

Bradw-17, 2019-03-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

etracer65, 2019-03-15:
Closing as no information was supplied.

closed 2019-03-15 by etracer65

Bradw-17, 2019-03-15:
Sorry about that, hit enter for next paragraph and it decided to submit my issue early lol

Bradw-17, 2019-03-15:
https://files.slack.com/files-pri/T221J1F6J-FH0F4NWR1/screen_shot_2019-03-12_at_6.28.59_pm.png

https://files.slack.com/files-pri/T221J1F6J-FH0JNR0N6/image.png

etracer65, 2019-03-15:
You have too much filtering turned off for the amount of noise in your quad - particularly the dterm lowpasses. I would recommend using the latest RC2 and resetting filtering to defaults and work back from there.

Bradw-17, 2019-03-15:
The same amount of filtering was applied to earlier builds. I believe I was flying ~build #1490 with no issues. It was after updating to Jflight, whichever build was current on March 12th. that these issues arose. At the same time, people testing the master builds noted similar issues in the BF Slack. The dropbox link posted above includes a log from my quad, as well as another pilots log whom I have no acquaintance with, yet he found a similar issue after flashing to RC-1.

Bradw-17, 2019-03-15:
Reading through the issues, I have found a similar claim: https://github.com/betaflight/betaflight/issues/7793

etracer65, 2019-03-15:
And the answer is the same there - put back the filtering and then try reducing it as you test. Also if you're not using and actual build like RC2 it's difficult to do anything. "Jflight" isn't a release.

etracer65, 2019-03-15:
And don't use a `diff` from earlier dev releases - things change. Start with a clean full-chip erase and configure from scratch.

Bradw-17, 2019-03-15:
I understand JFlight isn't a release. But if the issue was only present in Joe's build, I would have opened an issue on his repo instead of the master. This issue has presented itself in RC-1 as the included logs show. After doing a full chip erase, and then pasting the diff of a working, well flying quad after a firmware update, these issues have presented themself. The other issue I referenced ( https://github.com/betaflight/betaflight/issues/7793 ) mentions similar problems under similar circumstances. So my concern is that this is not a simple matter of lack of filtering, due to filtering being identical between builds. Joe's interpolated_ff and ff_spread do create a difference, I agree. But since this was present in an RC-1 build as well, I thought it worth mentioning.