betaflight/betaflight #7800: Blheli test code _67 causing esc / Fc resets with rc1

loomerco, 2019-03-15:
**Describe the bug**
When using FC firmware 4.0.0_rc1 with blheli test code ending _67 I got reproducible mid flight resets on 2 very similar quads:
It looks like esc’s #1,2,4 are resetting to a point where they stop work, but along that the FC also looses some functions like adc (vbat shows zero), beeper stops working, arm disarm works but only motor 3 works.

Setup is rpm filter, first Dterm and gyro lpf and dynamic filter. Running dshot 1200 8k/8k cpu OC.

-This does not happen with blheli test code _65
- this doesn’t happen with a hex from @joelucid _67
- looks like it doesn’t happen in 4/4, but wasn’t tested enough.
- dshot600 makes it much much worse.

**To Reproduce**
Steps to reproduce the behavior: see above.

**Flight controller configuration**
A diff is here:
https://betaflightgroup.slack.com/archives/CC07VN54L/p1552324864051000?thread_ts=1552324774.050000&channel=CC07VN54L&message_ts=1552324864.051000

**Setup / Versions**
 - Flight controller: Matekf722 running bf 4.0_rc1
 - Other components: esc’s are dys aria_32 35A running blheli rpm test code ending _67

A log of an event described:
https://betaflightgroup.slack.com/archives/CC07VN54L/p1552319110044600?thread_ts=1552318725.044400&channel=CC07VN54L&message_ts=1552319110.044600

Some more info from bench tests:
https://betaflightgroup.slack.com/archives/CC07VN54L/p1552421140312300

mikeller, 2019-03-15:
@loomerco: Can you try if this is still a problem when you set `dshot_bidir = off`.

joelucid, 2019-03-15:
@sskaug

sskaug, 2019-03-16:
My tests on the bench with Matek F7 4.0.0 RC1 and a soon to be released 32.6.8 show:
- 4k and Dshot 600 is rock solid
- 8k and Dshot600 gives 4 errors every time I give the status command (FC probably loaded and misses some)
- 8k and Dshot1200 runs well, but has a few errors
So 4k and Dshot600 seems to be robust and should be the recommended setup at this point.

joelucid, 2019-03-20:
@loomerco I think I fixed this issue today. Please try the binaries in my jflight repository.

loomerco, 2019-03-20:
@joelucid OK I will try it later today. I also see very few errors in latest master so I will check it as well.

loomerco, 2019-03-20:
@joelucid Joe I bench tested - Dshot bidir is not working with this hex. tried dshot600/1200 and 4/4 and 8/8.
I'm not getting any rpm data (not shown).

joelucid, 2019-03-20:
You need to use the new command dshot_telemetry_info to retrieve the rpm info @loomerco.

loomerco, 2019-03-20:
OK working. will test soon

loomerco, 2019-03-20:
@joelucid no resets today, but made a mistake: i forgot i reverted one of my quads to blheli X.65 so testing was:
2 X batteries on latest master + blheli X.67 (rc1 gave resets with 67).
3 batteries on Jflight + blheli X.65.
both flying 8/8 +dshot1200.

I believe I can easily try to reproduce later indoors with the worst scenario: 8/8 +dshot600. will comment if I manage.

joelucid, 2019-03-20:
Please try as I would like to get the fix integrated.

loomerco, 2019-03-20:
@joelucid Jflight + test X.67 looking very solid! I did a short hover with DSHOT600 (used to reset immediately with RC1) without reset. Also did some bench tests. hopefully I mange to squeeze a flight tomorrow.

Bench results:
8/8 + dshot600 shows very few errors:
`Dshot reads: 1184241
Dshot invalid pkts: 6
Dshot irq micros: 2

Motor     RPM   Invalid
=====   =====   =======
    0     147     0.00%
    1     136     0.00%
    2     150     0.00%
    3     138     0.00%`

8/8 +Dshot1200 shows zero errors:
`Dshot reads: 1213829
Dshot invalid pkts: 0
Dshot irq micros: 2

Motor     RPM   Invalid
=====   =====   =======
    0     147     0.00%
    1     136     0.00%
    2     150     0.00%
    3     137     0.00%`

I did manage to causethe ESC's to reset at a very specific scenario: slowly throttling them up to the point where the motors just start to react but unstable. this was with 8/8 + DSHOT600:
`Dshot reads: 1274325
Dshot invalid pkts: 362379
Dshot irq micros: 3

Motor     RPM   Invalid
=====   =====   =======
    0     147     0.00%
    1      15   100.00%
    2     150     0.07%
    3      15   100.00%`

Looking good!

closed 2019-04-27 by stale[bot]