betaflight/betaflight #7805: [BF 3.5.x] Unstable Cycle Time with TELEMETRY enabled

swilkens, 2019-03-16:
**Describe the bug**
Flashing BF 3.5.x (currently 3.5.7) on defaults shows unstable Cycle Time when TELEMETRY is enabled.

**To Reproduce**
 - Flash BF 3.5.7 on BETAFLIGHT F3 FC
 - Full erase
 - Connect
 - Notice Cycle Time varies between ~130 and ~310

[image]

[image]

 - Disable TELEMETRY and option
 - Save & Reboot, reconnect
 - Notice Cycle Time remains more stable

**Expected behavior**
Stable Cycle Time

**Flight controller configuration**
Just for verification, an empty diff:

```
# diff
# Betaflight / BETAFLIGHTF3 (BFF3) 3.5.7 Mar 15 2019 / 21:36:59 (e9130527c) MSP API: 1.40

profile 0

rateprofile 0
```

**Setup / Versions**
 - BETAFLIGHT F3 purchased from multirotorparts
 - 4x Aikon 30A V3.1

**Additional context**
 - Reduction of GYRO / PID loop freq. to 4 / 1 kHz drops CPU load to 15% but does NOT stabilize the Cycle Time
 - Motors not attached

swilkens, 2019-03-16:
A closer look shows disabling TELEMETRY does improve the situation, but doesn't remove the instability altogether.

Below logs are with debug_mode = CYCLETIME on 4.0.0-RC2, peek values are roughly the same ~400.

TELEMETRY ON:
[image]

TELEMETRY OFF:
[image]

0crap, 2019-03-17:
I see the same on a OMNIBUS F3 board.
Even adding a digit on Cycle Time at regular interval.
Meaning it goes above 1000!
I'm running 4K/4K.

When I have time I'll try to flash old Jenkins builds, to see when this behaviour starts.

swilkens, 2019-03-18:
@0crap Do you see the same spikes in Cycle Time in the blackbox logs? The visual indicator on the configuration interface is indicative but not very absolute.

I'm also not sure if the spikes I'm still seeing when I disable TELEMETRY are an artifact of the logging or if the cycle time is actually spiking.

I'm currently going through versions to find out if I can find a point in time when this started, because I don't remember the Cycle Time being so unstable on my F3 board in the past.

0crap, 2019-03-18:
I can't check BBL because this clone board has no BBL at all. :-(
On 4.0 RC2 and default settings I already see the Cycle Time jump from 3 to 4 digits.
So also with default settings it's jumping to >999.

0crap, 2019-03-18:
Only some really old stable versions don't show it. (CPU only at 12%.)
But if you go more recent the cpu load also creeps up.
With that also Cycle Time becomes itchy.
From 3.3.3 to 3.4.0 there is large jump in "default" CPU usage.

Guess what we see is in for a long time and probably not worth the devs time to troubleshoot.
Also because support for F3 boards will be dropped after BF 4.x if I'm correct.

swilkens, 2019-03-20:
> Only some really old stable versions don't show it. (CPU only at 12%.)

I can drop the PID loop to 1 kHz to obtain 15% CPU load and still see strong spikes in the Cycle Time, so I'm not convinced it's directly CPU load related. Still trying to track down a cause.

0crap, 2019-03-21:
There you have a point...
I only have one F3 FC, without BlackBox, makes it hard to look for a clue.
I do remember I see the itchy Cycle Time also on older firmware.
My guess it's already in for some time. not 4.0 related.

Hope you can find a direction for the devs.

closed 2019-04-27 by stale[bot]

Juzzle1, 2019-09-02:
I've been seeing the same issue with my F3 controller (Singularity).
Cycle time in betaflight configurator appears to be extremely erratic with telemetry enabled.
CPU load doesn't seem to be affected much with telemetry disabled or enabled (22-24%) with 1.3kHz PID and 2.6kHz gyro, but cycle time fluctuates between 380 and 800 with telemetry enabled and barely moves from 380 with it disabled.

 I hooked up a motor ouput to my scope and you can see this is real:

Telemetry enabled:
[image]

Telemetry disabled:
[image]

Notice my scope also measures the PWM update freqency as 1.225kHz with telemetry enabled and 1.3kHz with it disabled - this is consistent and not just from this particular screen grab.

The quad seems to fly OK, but I presume this isn't normal?

yoodog, 2019-09-02:
I'm not sure man, but my F745 board doesn't do that Cycle time: 250 +/-2 at all times with or without telemetry enabled. I imagine f3 should behave same in terms of cycle time