betaflight/betaflight #7817: AIKON F4 target - missing PIN definitions for UART1 TX and UART4 RX?

awolf78, 2019-03-16:
@DieHertz On the Aikon F4 mini UART1 TX and UART4 RX have a pad on the FC but they are not defined in target.h?
Any chance you can let me know which pins they are connected to?

DieHertz, 2019-03-17:
They are designated for SOFTSERIAL 1 and 2, just check it out.
You can probably add full UART1 and UART4 definitions and let the user decide when to use UART1 and when SOFTSERIAL 1, same for UART4 and SS2.

awolf78, 2019-03-18:
Interesting, so update target.h to this? No problem if pins (e.g. PA9) is defined for a hardware UART or a softserial?

```#define USE_UART1
#define UART1_RX_PIN            PA10
#define UART1_TX_PIN            PA9

#define USE_UART2
#define UART2_RX_PIN            PA3
#define UART2_TX_PIN            PA2

#define USE_UART3
#define UART3_RX_PIN            PB11
#define UART3_TX_PIN            PB10

#define USE_UART4
#define UART4_TX_PIN            PA0
#define UART4_RX_PIN            PA1

#define USE_SOFTSERIAL1
#define SOFTSERIAL1_TX_PIN      PA9

#define USE_SOFTSERIAL2
#define SOFTSERIAL2_RX_PIN      PA1

jflyper, 2019-03-18:
@awolf78 
> No problem if pins (e.g. PA9) is defined for a hardware UART or a softserial?

Ah. I think it is okay, as the physical pad ensures the pin is used only for one of the functions.

I'm not sure how configurator responds to this configuration, though.
You might want to check it out.

DieHertz, 2019-03-18:
Think about it, I'm afraid UART initialization will eat up both RX and TX, leaving one of them unavailable for the softserial.

awolf78, 2019-03-18:
Thanks guys, that's what I thought - just double checking. No target update, will just reconfigure target.h UARTs for my purposes. Thanks again for your help.

closed 2019-03-18 by awolf78