betaflight/betaflight #7820: betaflight 3.5.x blackbox log header not complete on Openlog

trunglp, 2019-03-17:
Dear, 

Today i use [openlog ](https://user-images.githubusercontent.com/38026441/54485749-32c9ad80-48b1-11e9-9927-3174c47fb198.jpg) to record log file. and I flash it to [OpenLog_v3_Blackbox](https://github.com/cleanflight/blackbox-firmware)

I flash firmware 3.5.7 to ["Betaflight F4 Noxe Flight Controller" ](https://user-images.githubusercontent.com/38026441/54485641-c9489f80-48ae-11e9-9f97-6889476c0051.jpg)

I use UART1 for blackbox  baudrate  is **115200**
and config "gyro frequency" and "PID log frequency" to: **8k/8k**
And use "Serial Port device", and use "blackbox logging rate": **2khz** 

But my log error head same image

[image]

everyone  fix it for me. Thank you.

McGiverGim, 2019-03-17:
I don't see any problem in your image. Can you concrete more where is the problem?

trunglp, 2019-03-17:
@McGiverGim  i'm sorry if it is not an error, because i open my log in blackbox explorer, i don't see it show

[image]

This is link my log
[LOG00032.TXT](https://drive.google.com/file/d/1eJIV2CWDVvfViCNOgQkwLouAnCzZteWL/view?usp=sharing)

McGiverGim, 2019-03-17:
I'm out of home, I will try to open it tomorrow with latest master.

etracer65, 2019-03-17:
I really doubt you can log at 2KHz over a slow 115200 baud serial port. Try a much slower logging rate like 0.5K.

mikeller, 2019-03-18:
@trunglp: @etracer65 is right, 2 kHz is much too fast to log at 2 kHz. You might be able to do that if you build a custom version of openlog with the logging rate set to 1000000 baud (the maximum it supports) and then configure Betaflight to output the log at 1000000 baud as well in CLI:

```
serial <your blackbox port number> 128 115200 57600 0 1000000
```

McGiverGim, 2019-03-18:
I have tried to open the log, and is only 300ms of one frame. The header is ok, but it seems the rest of data is not. I think etracer and mikeller were right, and the problem is the logging rate.

trunglp, 2019-03-18:
i said thank you @mikeller @McGiverGim @etracer65 i will change blackbox baudrate and logging rate  config is slower

trunglp, 2019-03-19:
today i change blackbox baudrate to 1000000 and test logging rate from 500Hz and 1Khz.  and my log working.. thank you, i will close thread.

closed 2019-03-19 by trunglp