betaflight/betaflight #7835: inverted yaw spin  / no yaw spin, quad pushes to the ground

T3flon, 2019-03-19:
**Describe the bug**
at the invert yaw spin, there is no yaw spin. Instead, the quad is pressed to the ground.
test with two quads

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
at the invert yaw spin, there is no yaw spin. Instead, the quad is pressed to the ground. yaw spin only happens with minimal stick movement

**Flight controller configuration**
Omninext F7

**Setup / Versions**
 Betaflight 4.0 RC2

**Additional context**
Add any other context about the problem here.

etracer65, 2019-03-19:
If the quad is inverted it would seem likely that it would be thrusting downwards.

Maybe you need to better explain your problem. Also provide a `diff` of your settings as requested when the issue was opened.

T3flon, 2019-03-19:
if the quad is inverted. there is no yaw spin. only thristing downwards

T3flon, 2019-03-19:
# diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Mar 14 2019 / 11:35:05 (360afd96d) MSP API: 1.41

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id 

# name
name Mark 2 6s

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2075 0 0
aux 1 1 1 1475 2050 0 0
aux 2 13 3 1425 1575 0 0
aux 3 35 3 1725 1825 0 0
aux 4 36 2 1450 1550 0 0
aux 5 39 3 1950 2075 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_to_use = SECOND
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = 109,-113,-22
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set max_check = 2000
set serialrx_provider = CRSF
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_switch_mode = STAGE2
set vbat_scale = 124
set vbat_divider = 11
set ibata_scale = 262
set small_angle = 180
set pid_process_denom = 1
set osd_vbat_pos = 2081
set osd_tim_1_pos = 2103
set osd_current_pos = 2094
set osd_mah_drawn_pos = 2087
set osd_warnings_pos = 2441
set osd_esc_rpm_pos = 2433
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 3
set vtx_freq = 5880

# profile
profile 0

set dterm_lowpass2_hz = 0
set iterm_relax = OFF
set iterm_relax_type = GYRO
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 201
set pitch_rc_rate = 201
set yaw_rc_rate = 168
set roll_expo = 33
set pitch_expo = 33
set yaw_expo = 18
set roll_srate = 61
set pitch_srate = 61
set yaw_srate = 64
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end

T3flon, 2019-03-19:
> 
> 
> If the quad is inverted it would seem likely that it would be thrusting downwards.
> 
> Maybe you need to better explain your problem. Also provide a `diff` of your settings as requested when the issue was opened.

I know that too ^^
but there is no yaw spin, maybe a little bit

etracer65, 2019-03-19:
Yeah, still not getting what your problem is. The quad doesn't know whether it's inverted or not. I assume you have yaw control in normal flight. Perhaps provide video showing the problem. A blackbox log would also be needed.

T3flon, 2019-03-19:
OK. if I'm inverted and want to spin a yaw, nothing happens. Minimal yaw spin with small stick movement. but if I want to make a strong / fast inverted yaw spin, the quad will almost stop and it will just thrusting to the ground.
that's not normal

T3flon, 2019-03-19:
video I create later, blackbox does not exist

etracer65, 2019-03-19:
Blackbox log is mandatory.

T3flon, 2019-03-20:
https://youtu.be/8lpEcAJz__Q

T3flon, 2019-03-20:
@0.26 inverted yaw spin with low stick movment 
@0.29 inverted yaw spin with maximal stick movment

T3flon, 2019-03-20:
on the second try you can clearly see how the quad suddenly slows down and pushes itself down. It should be noted at what altitude the experiment takes place

T3flon, 2019-03-20:
if that's not enough, the experiment is repeatable and it happens with 2 quad both have the f7. a 6s quad and a 4s quad

etracer65, 2019-03-20:
You're going to have to supply a blackbox log for further assistance. The flight video can't differentiate what the flight controller is doing vs. your command inputs. For example in the second yaw spin it sounds like you haven't  lowered the throttle while inverted (in fact sounds like full throttle) which would reduce yaw authority and thrust you into the ground. Without a blackbox log it's impossible to tell. Being that nobody else is reporting a problem and you say it's happening on multiple quads, that would imply either a setup problem or a problem with how you're performing the move.

T3flon, 2019-03-20:
They are quite right. I have to say that I've done inverted yaw spin hundreds of times. But that feels totally wrong. blackbox log I submit later.
"For example in the second yaw spin it sounds like you haven't lowered the throttle while inverted (in fact sounds like full throttle) which would reduce yaw authority and thrust you into the ground"
I can assure you that the throttle was completely down. I only used yaw.

etracer65, 2019-03-20:
You should also go back to defaults and test with a minimal configuration. No RPM filter, default PID's and filter. Even default rates. Also try `set abs_control_gain = 0`.

T3flon, 2019-03-20:
Okey
1. Betaflight 4.0 (RPM filter + default pids)
2. Betaflight 4.0 (Default pid's + my rates)
3. Betaflight 4.0 (default pid + default rates

and try set abs_control_gain = 0

joelucid, 2019-03-20:
That does seem strange. Looking forward to the logs.

T3flon, 2019-03-20:
Betaflight 4.0 (RPM filter + default pids)  inverted yaw spin at about 2 min. I can not say it more precisely. blackbox always crashes
[Betaflight 4.0.zip](https://github.com/betaflight/betaflight/files/2989489/Betaflight.4.0.zip)

T3flon, 2019-03-20:
Betaflight 4.0 default tests will be done tomorrow and handed in later.
What I can say, it only happens when a certain stick movement is achieved. before it feels relatively normal.
but it is clear, the quad suddenly slows down its rotary motion and begins to thrust on presumably all the engines.
The test is really not that easy, because I have to intercept the quad incredibly fast if I want to provoke the error.
the thrust that generate the 1750 kw engines directly is really scary

joelucid, 2019-03-20:
Can't really look at that log - it seems terribly corrupted. Could you log at a lower frequency so the log is readable? Overall the logs shows good tracking.

T3flon, 2019-03-20:
log at 1khz.
I do not know exactly what's going on, that's not the case with Betaflight 3.5 .. and even with Betaflight 4.0 that has come and gone with the versions
[Betaflight 4.0 RPM.zip](https://github.com/betaflight/betaflight/files/2989809/Betaflight.4.0.RPM.zip)
2 Log Betaflight 4.0 (RPM filter + default pids)

joelucid, 2019-03-20:
Still can't read the logs. Please try to create logs that can be read and then highlight the log section where the problem is by embedding a screenshot here.

T3flon, 2019-03-20:
do my best. have now put on 500 hz. Let's see if it gets better.
As I said in the beta, it was already gone a few times ...

Tomorrow I test first without RPM filter. Let's see if we can narrow it down then

etracer65, 2019-03-20:
I was able to open the second log and in every case where you're doing a hard yaw you're also applying significant throttle. That's why it's happening. The quad is doing exactly what you're commanding. Red graph is throttle input, yellow is yaw.

[image]

[image]

[image]

T3flon, 2019-03-21:
you're right. I'm doing something wrong. yes I changed my yaw rates in winter ... but then I do something like that ... sorry since your time has stolen for such a crap
This also explains why the slow invert yaw spin works

test tomorrow still betaflight 4.0 and let a stickcam record
I'm really sorry, could have sworn that I do not do such a stick movement
i will test again tomorrow and will then close the request

T3flon, 2019-03-21:
[image]
... I set the stick low threshold to 1020.
That made that the quad gave thrust, when tried to make the inverted yaw spin.
why I have moved the sticks so funny I can only explain myself that I flew all the time over the winter even with a stick low threshold of 1000.
have it tested that too and so was an inverted yaw spin not possible. then the quad behaves badly in the inverted yaw spin.

McGiverGim, 2019-03-21:
This is a LOT of throttle been applied according to the blackbox capture of etracer. Maybe you have changed some mixing on the radio or something similar?

T3flon, 2019-03-21:
I was also very surprised.
But with stick low threshold of 1000 the inverted yaw spin is even worse.
In principle, it all started because I had with my new 6s quad frightful desync's.
Since I did not know where it came from I tried everything possible from ..
among other things just stick low threshold.
Apparently, I got used to it a very strange control over 3 months ....
Could have sworn that I only use yaw.
Just look at it again. after I flew a few packs.
Someone knows why betaflight blackbox always crashes?

mikeller, 2019-03-21:
> Someone knows why betaflight blackbox always crashes?

You are probably logging too fast - try reducing the logging rate. If you are logging to an SD card, try using a different brand / type of card as well - they don't all work the same.

etracer65, 2019-03-21:
I'm not sure why you're messing with "stick low threshold" (`min_check`). This has nothing to do with de-syncs or fixing them. This is simply the threshold below which the throttle is considered to be zero. Setting to to 1000 is an extremely bad idea as it will likely prevent you from arming. This setting has nothing to do with you applying throttle during yaw - particularly when you're applying 60% or more throttle in many cases. It just really looks like you're moving the left stick diagonally up and right when trying to yaw.

T3flon, 2019-03-21:
it was definitely stick low threshold.
I have it now on had it change to 1060 briefly. I caught a few more times, I move the stick so funny.
I have to get used to it again. Can it really only explain to me that I am so funny moves the stick.
Now the inverted yaw spin works the way it should.

But now I have noticed another thing, if I have the rpm filter active, sometimes the OSD disappears for a second.
sometimes signs are everywhere. sometimes the battery simbole is suddenly in the middle of the screen.

Many thanks to all who helped solve the problem!!!!

closed 2019-03-21 by T3flon