betaflight/betaflight #7840: Betaflight 4.0 - Foxeer F722 dual gyro - can't setup vtx channels Raceband 2 and Raceband 3 through the radio.

goodnerdboy, 2019-03-19:
**Describe the bug**
Betaflight 4.0 - Foxeer F722 dual gyro - can't setup vtx channels Raceband 2 and Raceband 3 through the radio.

**To Reproduce**

Install and configure Betaflight on a Foxeer F722 dual gyro, setup taranis telemetry and last version of lua scripts on your radio (mine is a x-lite).
Try to set R2 or R3 through the lua scripts, it displays "save" then "retry", then reload the previous values.
Tried also with Betaflight 3.5.7 and it doesn't work too.

I tried the exact same thing with a clracing F7 and it works just fine.

Is anybody able to reproduce this? I am really not sure wheither this is a bug or a configuration problem.

**Expected behavior**
to be able to set all the vtx channels through lua scripts/radio.

**Flight controller configuration**
```
# diff

# version
# Betaflight / FOXEERF722DUAL (FXF7) 4.0.0 Mar 17 2019 / 13:12:42 (10a27bcb8) MSP API: 1.41

# start the command batch
batch start

board_name FOXEERF722DUAL
manufacturer_id 

# name

# resources

# timer
timer C09 0
timer C08 0
timer C06 0

# dma
dma pin A09 1
# pin A09: DMA2 Stream 2 Channel 6
dma pin A08 1
# pin A08: DMA2 Stream 1 Channel 6

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 2 1024 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 10,10::CT:7
led 1 11,11::CT:7
led 2 12,12::CT:7
led 3 13,13::CT:7
led 4 13,15::CT:7
led 5 12,14::CT:7
led 6 11,13::CT:7
led 7 10,12::CT:7
led 8 6,12::CT:7
led 9 5,13::CT:7
led 10 4,14::CT:7
led 11 3,15::CT:7
led 12 3,13::CT:7
led 13 4,12::CT:7
led 14 5,11::CT:7
led 15 6,10::CT:7
led 16 6,4::CT:7
led 17 5,3::CT:7
led 18 4,2::CT:7
led 19 3,1::CT:7
led 20 3,3::CT:7
led 21 4,4::CT:7
led 22 5,5::CT:7
led 23 6,6::CT:7
led 24 10,6::CT:7
led 25 11,5::CT:7
led 26 12,4::CT:7
led 27 13,3::CT:7
led 28 13,1::CT:7
led 29 12,2::CT:7
led 30 11,3::CT:7
led 31 10,4::CT:7

# color
color 6 0,0,255
color 7 60,0,255
color 8 359,0,255

# mode_color
mode_color 6 0 7
mode_color 6 1 7
mode_color 7 0 7

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 2 2 1300 1700 0 0
aux 3 27 1 1300 2100 0 0
aux 4 13 1 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set min_throttle = 1070
set dshot_idle_value = 450
set dshot_burst = OFF
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set current_meter = ESC
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vtx_channel_pos = 55
set osd_current_pos = 2466
set osd_warnings_pos = 2474
set osd_avg_cell_voltage_pos = 2485
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_freq = 5732

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end
```

**Setup / Versions**
 - Foxeer F722 dual gyro;
 - Foxeer CleanTx2 with power and channels unlocked;
 - R-XSR;
- T-motor 4in1 55A;

**Additional context**
Sorry if this is not the place to report this, but as my radio setup works with another board i assumed it was more board related than lua related.
My diff is with rpm filter but i also tried with a straight from defaults setup and the same happens.

mikeller, 2019-03-20:
@goodnerdboy: This is odd - there should not be a difference in how this works for the same version of Betaflight on two different boards.

What firmware versions are you running on the R-XSR?

goodnerdboy, 2019-03-21:
@mikeller i am so sorry.
I thought I updated all my r-xsrs but it turns out this one was still with an old firmware. Updated it and well everything works as expected.
Apologies for the non-issue and my dumbness.
Still it is interesting that with that old firmware it looked like everything was working but these frequencies.

closed 2019-03-21 by goodnerdboy

mikeller, 2019-03-21:
@goodnerdboy: All good. The issue is that with the old firmware, only 2 out of 256 values were transmitted incorrectly. So everything worked fine **unless** the update included one of these values. And it looks like changing to the bands you wanted did exactly that...