betaflight/betaflight #7849: BF4.0.0 RC2 OMNIBUSF4SD SUMD Reciever at UART1 not working:

BEE-KA, 2019-03-21:
I got a Graupner Reciever using SUMD but It is not possible to connect in new version.
I got no Signal in Receiver tab.
After flashing back to BF 3.5.5 it is working again without Problems.

UART1 is configured as Serial Reciever
I configured in BF "configuration" Serial Reciever  and SUMD

**Flight controller configuration**
```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 14 2019 / 11:33:21 (360afd96d) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set serialrx_provider = SUMD

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
 - Flight Controller is a X-Tower Flycolor OMNIBUSF4SD from HobbyKing.
(https://hobbyking.com/de_de/flycolor-x-tower-40a.html)

mikeller, 2019-03-21:
@Scavanger: This could be related to #6215 (since that was the only change made to SUMD in 4.0). I don't have SUMD test gear, are you able to help out debugging this?

watageek, 2019-03-21:
same problem with SerialRx DSM on UART1 OMNIBUSF4SD

Perfect on 3.5.7
not working on RC2
[BTFL_cli_20190321_151654.txt](https://github.com/betaflight/betaflight/files/2993785/BTFL_cli_20190321_151654.txt)

mikeller, 2019-03-21:
@watageek: 

> set serialrx_provider = SPEK2048

This seems to be using a different protocol - which means it is not caused by a problem in the SUMD protocol stack.

Can you please open a separate issue for this?

mikeller, 2019-03-21:
@BEE-KA: Can you please test this:

[betaflight_4.0.0_OMNIBUSF4SD.zip](https://github.com/betaflight/betaflight/files/2993869/betaflight_4.0.0_OMNIBUSF4SD.zip)

watageek, 2019-03-21:
For DSM it goes crazy but we have something

BEE-KA, 2019-03-21:
@mikeller: sorry but this seems not to help.

```
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 22 2019 / 09:33:04 (8239950c7) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set serialrx_provider = SUMD

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
```

BEE-KA, 2019-03-21:
Seems to be a Problem with the UART1 because SBUS (i can switch the RX to SBUS or SUMD) 
4.0.0  SBUS , SUMD not working,  
3.5.7 SBUS not working, SUMD working
So only working configuration is 3.5.7 with UART1 and SUMD

etracer65, 2019-03-21:
I just tested uninverted SBUS on UART1 with an OmnibusF4 v3 using the OMINBUSF4SD target and it works normally on 4.0RC2. It's necessary to use an uninverted signal because the board's built-in inverter is on RX6.

mikeller, 2019-03-21:
@BEE-KA: Looks like the clone that you are using is having some extra circuitry on UART1 that does not work properly under Betaflight 4.0. Can I suggest you contact the manufacturer / supplier and ask them for a fix?

watageek, 2019-03-22:
Can you try the 1100 build
https://ci.betaflight.tech/job/Betaflight/1100/artifact/obj/betaflight_4.0.0_OMNIBUSF4SD-1100-6797d5dd2.hex

vs
https://ci.betaflight.tech/job/Betaflight/1101/artifact/obj/betaflight_4.0.0_OMNIBUSF4SD-1101-baaaaf9d8.hex

Scavanger, 2019-03-22:
I can't confirm this issue. 
I have an "Airbot Omnibus F4 Pro V3" (same Target: OMNIBUSF4SD) as debug hardware and SUMD works perfectly on all Hardware UARTs under BF 4 RC1/RC2.

watageek, 2019-03-22:
@BEE-KA 
this one solved my issue on dsmx
https://ci.betaflight.tech/job/Betaflight/1535/artifact/obj/betaflight_4.0.0_OMNIBUSF4SD-1535-1bdbd1cd7.hex

they are relentlessly working on perfection

BEE-KA, 2019-03-22:
Hi all,
first thank you for your Input.
https://ci.betaflight.tech/job/Betaflight/1100/artifact/obj/betaflight_4.0.0_OMNIBUSF4SD-1100-6797d5dd2.hex
Is working… (1100 Build)
Build 1101 is not working.

is it possible to check, what the exact differeces are?
And find a fix for Releasing? 
@mikeller i don´t have any contact to the manufacturer :-(

watageek, 2019-03-22:
yes the build 1535 should clear the issue
https://ci.betaflight.tech/job/Betaflight/1535/artifact/obj/betaflight_4.0.0_OMNIBUSF4SD-1535-1bdbd1cd7.hex

BEE-KA, 2019-03-22:
Just for Information: In my case 1535 not working. (as i know this is only clearing DSMX issue)

BEE-KA, 2019-03-22:
@Scavanger thank you for your Information. Can you figure out, why it is not working on all my Flycolor FC´s? (i have 3 of them) the answere should be between the Build 1100 and 1101

etracer65, 2019-03-22:
I don't have any SUMD equipment to test so I wrote a simulator using an Arduino that outputs SUMD protocol. Tested with 4.0RC2 on a real OmnibusF4 v3 (using OMNIBUSF4SD) connected to UART1 and it works perfectly in both 16 and 32 (SUMD v3) channel modes.  All channels working.

watageek, 2019-03-23:
@BEE-KA 
do you want to try this one.

[betaflight_4.0.0_OMNIBUSF4SD_Unremoved IOConfigGPIOAF from timerInit.zip](https://github.com/betaflight/betaflight/files/2998945/betaflight_4.0.0_OMNIBUSF4SD_Unremoved.IOConfigGPIOAF.from.timerInit.zip)

BEE-KA, 2019-03-23:
@watageek 
thank you very much. I tried this build but i am sorry, that it did not help :-(

BEE-KA, 2019-03-23:
I checked it with UART6 and there it is running with 4.0.0 RC2

watageek, 2019-03-23:
The gurus will sort it out. 
Happy flying.

mikeller, 2019-03-23:
@BEE-KA: As you stated yourself, this is not happening with original OMNIBUSF4SD, making this clearly a fault of the clone you are using. We do not have any contact with them either, so I guess your best bet is to return the hardware to your supplier as 'not working properly' and get a refund, and then buy an original flight controller from a manufacturer that actually supports their user and replaces defective hardware that they sold.

BlackPixxel, 2019-04-18:
I have an original Omnibus Fireworks V2, and there SUMD is also not working on UART1 with BF4.0, but it does work when I flash previous Betaflight versions.

schugabe, 2019-04-28:
I helped a friend to upgrade his quad to BF4 on the betaflight f4 board. With BF3.5 SUMD is working: the only configuration need is to enable SUMD and it works (on UART6).
But with BF4 this does not work.

schugabe, 2019-05-01:
With git bisect I managed to find the commit that breaks sumd for betaflight 4.0: 7665a424c7386ed90f7bd042e265a845a41fd63e

mikeller, 2019-05-01:
Thanks for bisecting, @schugabe. Looks like the issue is not with the SUMD protocol in itself, but with the RX hardware being sensitive to logic level on the RC output pin. What hardware were you using, and does the documentation for it state something about required logic levels on the RC pin (presumably only at power up, since this is an _output_ pin for the RX)?
Also, Betaflight isn't currently geared to allow custom powerup power levels to be set on arbitrary pins, so there probably won't be a quick fix.

schugabe, 2019-05-02:
The rx is the Graupner GR-16 HoTT. I did not find any information about required logic levels.

What do you mean with RC output pin? The output pin of the RX or on the FC?

mikeller, 2019-05-02:
@schugabe: The output pin of the RX that is used to send SUMD to the flight controller. This RX seems to be firmware upgradeable. So one possible explanation is that the boot loader in the RX uses the level on the SUMD pin to determine whether to boot into 'normal' or 'firmware update' mode. But without collaboration from Graupner this is just speculation.

schugabe, 2019-05-04:
@mikeller while looking on the commit I posted I noticed that the UART6 is not in the timerHardware array on the betaflightf4 target so the change in 7665a42 did not change anything for this pin. 

Your theory with the bootloader of the rx sounds reasonable but I checked the manual and the firmware update is done via a completely different pin.

So it's still not clear why sumd is not working. Or am I missing something here?

I created a unit test and verified that the sumd implementation is correct (I could create a PR for the unit test).

mikeller, 2019-05-04:
@schugabe: Good point about the UART6 not being in the `timerHardware` array. Not sure if there weren't other side effects of the old shotgun approach to pin initialisation.
Do you have access to a scope (or at a pinch a logic analyser), so that you could check the level of the RX pin when the system is powered up, before and after the change that causes SUMD to fail?
And yes please, pull requests for unit tests are always welcome!

schugabe, 2019-05-08:
@mikeller 
Result during power up for 3.5:
[image]

Result during power up for 4.0.2:
[image]

The high voltage level for 4.0 is much lower and the decoding function of the scope does not detect anything. For 3.5 the decode function shows the correct results.

Seems like the pin is not configured correctly.

jflyper, 2019-05-08:
Alright ... so, it looks like that this is related to #8094 also...

schugabe, 2019-05-08:
If I add the lines from 7665a42 back in sumd starts to work again. 
More specific adding just the call for timerHardware[0] back does the job.
```c
const timerHardware_t *timerHardwarePtr = &timerHardware[0];
IOConfigGPIOAF(IOGetByTag(timerHardwarePtr->tag), IOCFG_AF_PP, timerHardwarePtr->alternateFunction);
```

schugabe, 2019-05-08:
And two more images:
Just the PIN without anything connected for bf 3.5:

[image]

Same with bf 4.0.2:

[image]

mikeller, 2019-05-09:
@schugabe: Thanks for these - wait what, looks like this change disabled the pull up on the UART. :confused:

Ziege-One, 2019-05-11:
Hello 
I can confirm this issue. I have a REVO F4 (Flip32 F4) board. UART1 RX Graupner SUMD GR12L = no input. Change to UART6 RX it works. 
The Telemetry (HOTT) change from UART6 TX to UART1 TX and works on very well.
Hope the info is usefull.

mikeller, 2019-05-11:
@Ziege-One: Interesting that this should be dependent on which UART is used - the change that seemingly 'broke' it (7665a42) is not specific to an individual UART. Looks like #5675 is rearing its ugly head again - PA9, which is used for UART1_TX on the OMNIBUSF4SD is hard wired by STM to be capable of serving as USB voltage sensing pin, and the STM libraries do a bad job of configuring this.

Ziege-One, 2019-05-12:
On an OmnibusF4SD test fc i have this issue to.
 I fix it with:
`resource pinio 1 b8
set pinio_config = 1,1,1,1`
so SUMD works on UART1 RX
look at this 
#7523

mikeller, 2019-05-12:
@Ziege-One: Is this an original OMNIBUSF4SD, or is this a clone? According to my information, and @jflyper's testing, the original OMNIBUSF4SD isn't using the 'broken by design' schematic that the requires the hack described in #7523 to make it work.

Ziege-One, 2019-05-12:
it is a clone

mikeller, 2019-05-12:
@Ziege-One: Heh, that explains it. Thanks.

schugabe, 2019-05-21:
I investigated a bit further and the comment https://github.com/betaflight/betaflight/issues/7523#issuecomment-461044801 explains why 7665a42 broke sumd for this target: the PPM pin is configured as PULL UP and via the inverter the SUMD pin is pulled down and the uart signal is not getting high enough.

Yellow signal is the sumd input pin and pink is sbus/ppm input.
Serial RX/SUMD active: PPM pin is high and SUMD low => no sumd signal would be received (if rx was connected)

[image]

Serial RX deactivated and normal PPM input selected: PPM pin is low, SUMD pin high  => sumd signal would be received (if rx was connected)

[image]

The pinio_config also solves the problem on the BETAFLIGHTF4 board (no clone... I hope so since I bought it from a shop in germany for the regular price...)

The question is how this can be solved? Add an option for the targets that indicate that a certain UART is influenced by another pin and if the UART should be used for not inverated signals like sumd/inav the PULLUP for this pin is not activated?
The easiest fix would be to init the TIM10 for the PPM signal as before...

mikeller, 2019-05-21:
We'll probably need to look into adding a facility to configure 'unused' pins for pull up / pull down, just like the pinio based config does in an overcomplicated way.

How about something like `resource b8 pullup`?

@jflyper: Thoughts?

schugabe, 2019-05-29:
Using the resource command like this would compeletely change the usual syntax of this command so in the usual style it would be ```resource pullup1 b8``` but the usual index does not really make sense here and would make it uncessary complicated for the user to manage the rules. But most setups require probably only one explicit pullup/down setting so this would probably be no big deal.

I think a command like ```pin b8 PULLUP``` would be more self-explanatory and could integrate the complicated pinio settings because that pinio config is really not that great

mikeller, 2019-05-29:
@schugabe: Sorry for getting the order wrong, what I meant was `resource pullup b8` (the index is already optional).
I am against introducing new commands with new syntax for stuff that can be done with existing syntax.

mikeller, 2019-10-01:
@BEE-KA: Has this issue been addressed? If so, can you close this please?

closed 2019-10-06 by mikeller