betaflight/betaflight #7852: 4.0 RC2 OMNIBUSF4SD DSM broken on UART1

watageek, 2019-03-21:
DSM is broken on UART1 on OMNIBUSF4SD

4.0RC2

[BTFL_cli_20190321_151654.txt](https://github.com/betaflight/betaflight/files/2993865/BTFL_cli_20190321_151654.txt)

tried with a custom brew ... still broken

[common_pre.h.txt](https://github.com/betaflight/betaflight/files/2993872/common_pre.h.txt)

watageek, 2019-03-22:
issue solved with build 1535

watageek, 2019-03-22:
workrd as expected and then the channel went crazy, and then nothing

jflyper, 2019-03-22:
Please attach the most recent CLI `diff` output you have the issue with. The one in the first comment is against 6 month old firmware. Please also provide the information about your flight controller and your receiver as instructed by the pre-formatted text when you first opened this issue ticket.

watageek, 2019-03-22:
This is SPEKTRUM Clone (REDCON) satelite. Remote is an old DX4E. FC is an OMNIBUSF4SD (V3 i think)

jflyper, 2019-03-22:
Just tested with authentic Omnibus F4 V3, flashed with
```
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 22 2019 / 18:20:52 (360afd96d) MSP API: 1.41
```
loaded with your config diff.

With Hyperion's satellite connected to UART1, it works without an issue.

watageek, 2019-03-23:
Replicated the issue with an omnibus pro f4 v2 with satelite connected to uart1.

Same config used by jflyper. 
If I re-enable the change made in the build in timer.c #1100 to #1101 it works.

watageek, 2019-03-23:
I deeply apologize. I suspect data corruption on my side. I reflashed with 4.0RC2 (360afd9).

I will close this issue for good since it is obviously not related to the code.

closed 2019-03-23 by watageek

jflyper, 2019-03-23:
@watageek 

Tested with Omnibus F4 Pro V2, flashed with the same #360afd9 (somewhere in 4.0 RC), configured with your BTFL_cli_20190321_151654.txt, using the same Hyperion Satellite on UART1.

It works without an issue.

jflyper, 2019-03-23:
Now, are you saying that 4.0-RC2 works with your setup if you revert the change introduced in CI build "#1101", which is PR #6681 (commit 7665a42)?

jflyper, 2019-03-23:
@watageek Oh, is it working now?

watageek, 2019-03-23:
Yes. Each time...
What baffles me is, it works and I reflash, it doesn't work. This morning I had no issue at all with RC2 release. 

Rechecked everything prior to arm test and launch test on bench. No RX input on the FC. Bind is rock solid.

But 3.5.7 is flawless
Build 1100 is flawless
Rc2 with revert change flawless too.

The issue with uart1 on RC2 appears in less than a minute after the flash. The channel went crazy in receiver tab followed by failsafe engaged as set in the failsafe tab.

I was highly suspecting hardware failure. But I am not convinced.

I don't want to lose anybodies energy and time. 

Thank you for the follow up.

Since I am an isolated case for now, as you wish we might let it slip or investigate. 

Best regards

watageek, 2019-03-24:
> Now, are you saying that 4.0-RC2 works with your setup if you revert the change introduced in CI build "#1101", which is PR #6681 (commit 7665a42)?

@jflyper 
exactly

Occurring the same behavior on RC3
The official 4.0RC3 gives me no signal
4.0RC3"Recooked" is working flawless
same diff
[config_OBSD_4_rc3.txt](https://github.com/betaflight/betaflight/files/3000162/config_OBSD_4_rc3.txt)
[timer.c.txt](https://github.com/betaflight/betaflight/files/3000166/timer.c.txt)
[betaflight_4.0.0_OMNIBUSF4SD_Fix.hex.zip](https://github.com/betaflight/betaflight/files/3000172/betaflight_4.0.0_OMNIBUSF4SD_Fix.hex.zip)

mikeller, 2019-05-12:
@watageek: Can you please check if this fixes the problem for you:

```
resource pinio 1 b8
set pinio_config = 1,1,1,1
```

(from https://github.com/betaflight/betaflight/issues/7849#issuecomment-491553115)

Hardtail88, 2019-06-11:
I am experiencing this issue with omnibus pro f4 v2 and Spektrum quadrace satellites connected to uart1 on the OmnibusF4SD target.  The FC will not show receiver input when flashed with Betaflight 4.0, 4.02 and 4.03.  Betaflight 3.5.7 and all previous firmware versions work fine.  I have tried this on 2 identical quads.

watageek, 2019-06-12:
Sorry @mikeller for the late response, I don't have the F4 anymore.... I fried it. 

@Hardtail88 have you tried the cli command stated above?

Hardtail88, 2019-06-12:
Not yet, my programming expirence level is <=low.  to try the cli command is it a simple matter of copying and pasting in the cli followed by "save"?

"resource pinio 1 b8
set pinio_config = 1,1,1,1 "

mikeller, 2019-06-12:
@Hardtail88: Yes, that should work fine.

Hardtail88, 2019-06-12:
The cli command you provided solved my issue with both quads in BF 4.03.  Thank you very much!  Is this a step a will have to do on all future release of Betaflight?

Thanks again for your help!

mikeller, 2019-06-13:
@Hardtail88: We are still looking into how to best make it possible for users to set the firmware to work around these faulty hardware designs.