betaflight/betaflight #7854: Auto disarm on crash

marekhalmo, 2019-03-22:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

etracer65, 2019-03-22:
The problem with this is that it would be extremely difficult to make reliable and not disarm due to false detection. So making it not too sensitive to for false detection, but then making it "sensitive enough" to detect most crashes. For example, how many times have you clipped a branch or a race gate but were able to keep flying?

marekhalmo, 2019-03-22:
There could be some sensitivity setting. And of course it should be possible to disable this feature so if you are on the race  you wont disarm accidentally.

marekhalmo, 2019-03-22:
Also that is why I'm asking if it could detect the motor slip. There could be some half sec dead band so it won't trip accidentally..

etracer65, 2019-03-22:
I didn't say it was impossible - just extremely difficult. It would depend on whether anybody was interested in trying to implement this along with all the requisite testing (crashing) needed to ensure it was reliable. Feel free to give it a try if you're so inclined.

etracer65, 2019-03-22:
For example GPS Rescue already uses crash recovery as a "detection" of a crash and then disarms as a result. So it may be possible to use a similar approach. However I'm not sure many people would want a disarm over a possible recovery - particularly in the cases where continuing to fly high be an option.

marekhalmo, 2019-03-23:
This feature is oriented towards free style flyers that crash a lot and there are no race times. There is usually also no GPS on 5 inch freestyle quad. Why would you even want this on a long range or cinematic drone? That is why it must be possible to disable this feature.

mikeller, 2019-03-23:
I think this would be relatively simple to implement - just add as an alternative mode to the existing crash recovery, giving users the option to choose between 'recover' (existing) and 'disarm' (new). This way they can choose if they want the craft to try and recover after a crash, or disarm, and it's up to them to tune the detection sensitivity to avoid disarming in recoverable situations.

IB1387, 2019-03-23:
I fully agree with @mikeller. What I want to append is that it is however a good habit to train yourself to disarm in such situations yourself. That doesn't take away the usefulness of this feature request, as the fc can react faster than a human, but I fear the day where there will be people _relying_ on the "crash recovery"-based disarm functionality. 
I spent some time teaching novices to fly cheap amazon quads at an event and it became very apparent that the "disarm/throttle down reflex" in case of loss of control over the aircraft is something that has to be learned and doesn't come automatically. :-)

marekhalmo, 2019-03-25:
I agree that disarming the quad is very important. Not having that habit could lead to a painful injury. 
The quad could still beep if your arm switch is on.

krzysztofmatula, 2019-04-01:
Related topic is discussed on Inav since some time: https://github.com/iNavFlight/inav/issues/2331
To date there is no conclusion how to reliably detect if a quad has crashed or not...

marekhalmo, 2019-04-01:
Hello again. I don't understand why are you fighting against this..
There is a clear relation between gyro seeing rotation/acc forces and stick input so i see no reason why this should not work.

If you do 0 yaw. Hit a branch, the quad starts spinning and gyro measures 800 deg/sec - your quad has clearly crashed.

If you get 15g acc force during level flight - your quad has hit something

If your FC sends command to spin front motors at 4000 rpm, a 1/60s ago it was 4000 rpm, now it reads 0 rpm (some esc can even determine motor slip) - your props hit something

You need to disarm asap to prevent damage

It just needs a bit of thinking to figure out what is possible and what not + the effort should not be that much.

IB1387, 2019-04-01:
The likes of @etracer65 and other betaflight devs are intelligent people knowing their stuff. You can be sure if they say "it's difficult to reliably detect a crash" that it is. 
I for example had problems with my high rates when I first tried to use crash recovery. In the standard settings, the crash recovery algorithm prematurely activated, even when I just did a snap roll. That's the problem with theory and real life. What you say holds for a simulated environment without external forces, but if you add wind and other stuff like super rate to the mix (which relaxes the pid loop for larger stick movements), things become more difficult. And each new feature has to bring far more positive effects to the table than negative side effects. Do you want to have that auto disarm engage in case of a mid-air when racing or should it let you continue flying? Many crashes won't reach 15g of deceleration, where do you draw the line? Some quads may reach many g's of acceleration if you go to 100% throttle quickly? All these things have to be taken into consideration. 
It's not "just a bit of thinking", it's thinking very much about every edge case and then testing and validating every change you make. The benefit of having the feature has to be in line with the effort it takes to implement it.

If you ask me, I would vote to add to the crash recovery feature and just make an option to disarm instead of auto-level. If we don't enable that feature by default the user has to make sure it works for him when enabling it.

etracer65, 2019-04-01:
@marekhalmo 
> It just needs a bit of thinking to figure out what is possible and what not + the effort should not be that much.

Well then start here:
https://github.com/betaflight/betaflight/blob/master/src/main/flight/pid.c

mikeller, 2019-04-04:
@marekhalmo: #7924 adds what you are asking for. Can you please test and report back if this is working the way you want it to? Let me know what flight controller you are using, and I'll build the firmware for you. All you have to do is set `crash_recovery = disarm`.

PR #7924 (2019-04-14): Added 'disarm' option to crash recovery.
Fixes #7854.
closed 2019-04-14 by mikeller