betaflight/betaflight #7855: RPM - ESC Beeper Inteference

gvhlaw, 2019-03-22:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

gvhlaw, 2019-03-22:
Having music enabled in BLHeli32 is a known issue and problem with RPM code function.  It turns out that having an ESC beacon enabled also interferes with RPM and produces 100% errors in Dshot packets.  This emerged as an issue due to my quad having the ESC'S set to beep in a specific switch position at a value of 1500 - such that the ESC'S beep after a lipo is plugged-in until a TX is bound and the Aux Channel value drops to 1000.  The same problem would occur if an ESC Beacon is enabled for lost signal. Even after a TX is bound, significant errors persist in RPM data.  Even though no lipo is in use, the beacon
function persists in the FC during flashing of timer and DMA changes in Snippets and may interfere with same.  
[BTFL_cli_X220HV_RPM_Perfect.txt](https://github.com/betaflight/betaflight/files/2997986/BTFL_cli_X220HV_RPM_Perfect.txt)

etracer65, 2019-03-22:
This is not a bug or fixable. It's a limitation of the ESC firmware. BLHeli32 blocks when playing "beeps" through the motors. By "blocks" I mean it doesn't do anything else; won't respond to any DSHOT signals, won't respond with telemetry, etc.

And it really isn't a problem anyway as the motors can't "beep" while they're spinning.

gvhlaw, 2019-03-25:
Thanks Bruce - Joe asked me to report the issue - maybe just so folks will
know not to set a beacon on a switch at 1500 or the lost model beacon until
after they confirm that RPM is working properly.

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

closed 2019-05-01 by stale[bot]