betaflight/betaflight #7856: Diatone Mamba F405 Mini and FrSky D4R-ii

Vatlin, 2019-03-23:
First time I come to ask internet how to plug a reveiver after years of RC experiences !

I am trying to connect a D4R-ii receiver in CPPM mode to the Mamba F405 Mini, but nothing works...
I have tried to put all the informations in the pictures enclosed.
Any idea of what I am missing ? 
[image]
[image]
[image]
[image]
[image]

watageek, 2019-03-23:
add a diff or dump
have you choose ppm in the configuration tab in the configurator, 
ppm is not serialRX

etracer65, 2019-03-23:
As stated when you opened the issue - this is not the appropriate place to ask for help with configuration or setup problems. Issues are used to report problems problems or bugs with the firmware.

For configuration/setup questions try joining our slack: https://slack.betaflight.com/

Also consider our official Facebook group: https://www.facebook.com/groups/betaflightgroup/

Another good option is rcgroups: https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread

closed 2019-03-23 by etracer65