betaflight/betaflight #7860: Bi Directional DSHOT - No RPM in Status after spinning in Motor Tab

RobMcFly, 2019-03-23:
Hello Betaflight Team,

after spinning the Motor in the Motor Tab there ist 0 RPM in the Status:

Connect Battery waiting for Beeps. Connect USB
Status:
# Status
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=52degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3090, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000, BARO=LPS
System Uptime: 23 seconds, Current Time: 0000-01-01T00:00:00.000+00:00
CPU:8%, cycle time: 251, GYRO rate: 3984, RX rate: 33, System rate: 10
Voltage: 1490 * 0.01V (4S battery - OK)
I2C Errors: 2
Arming disable flags: RXLOSS CLI MSP
Dshot reads: 214656
Dshot invalid pkts: 5
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 7 0 0 4 11 5 12 6 13 7 14 8 15 8 16 9 16 
0 0 0 0 0 0 0 0 0 7 7 7 7 7 8 7 

"Exit" to leave CLI waiting for beep
Connect, go to Motor Tab. Spin every motor slowly:

Status:
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=59degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3090, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000, BARO=LPS
System Uptime: 71 seconds, Current Time: 0000-01-01T00:00:00.000+00:00
CPU:8%, cycle time: 249, GYRO rate: 4016, RX rate: 33, System rate: 9
Voltage: 1487 * 0.01V (4S battery - OK)
I2C Errors: 2
Arming disable flags: RXLOSS CLI
Dshot reads: 967884
Dshot invalid pkts: 3
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 14 7 7 7 7 14 14 7 14 14 14 0 0 11 18 12 19 13 0 14 1 15 2 16 3 16 4 
0 0 7 0 0 0 7 0 0 7 7 7 7 7 7 8 

Setup: OMNIBUS F7 NXT BF 4.0 RC2
Typhoon ESC. Test Code flashed / no Music

ESC flashing (on all four ESC :-) ) was succesfully. 
Hope this helps, Thanks for the great work
Kind regards

diff

# version
# Betaflight / OMNINXT7 (ONX7) 4.0.0 Mar 14 2019 / 11:35:05 (360afd96d) MSP API: 1.41

# start the command batch
batch start

board_name OMNINXT7
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

# end the command batch
batch end

#

etracer65, 2019-03-24:
The motors actually need to be spinning at the time you check the status to report any RPM. Not just that they have been spun. Also connect the battery *before* connecting the USB.

RobMcFly, 2019-03-24:
Thanks for your time...its working.

closed 2019-03-24 by RobMcFly