betaflight/betaflight #7861: Flight Time vs Armed Time

eumelOne, 2019-03-24:
As i am a racer and abuse my packs heavilly it would be really helpfull to se the real flight time instead of the armed time in the osd. My suggestion is the usage of a certain treshold value like launch_trigger_throttle_percent as indicator when the flight starts to start the timer. This will give us racers a more precise flight time.

OGFPV, 2019-03-25:
You already have the option to place, Flight Time or "Armed Time" on your OSD, or both if you like.

eumelOne, 2019-03-25:
There is nothing like "flight time" in BF. Either you can use "on time" which means the timer starts running right after you plug in your Lipo. Or you can use "armed time" which gives you either the overall arming time or the last arming time. But none of them gives you the accurate flight time. As Armed Time ≠ Flight Time. As racer you not flip your switch and raise the throttel. Somethimes you sit on the start ramp for 5-7 seconds. So the armed time gives you not an indicator for how long are you flying. 

I am flying my 6s1300mah at the edge. We are have to fly 2:00-2:20 minutes in german competitions. 2:20 is really hard with my setup and i have to maintain my lipo. So thats what i need the "real flight time" for.

mikeller, 2019-03-25:
@eumelOne:

> But none of them gives you the accurate flight time. 

I suspect what you want is not possible - we simply do not have a good way to detect the difference between 'almost flying but still on the ground' and 'flying'.
Can I suggest that you instead use 'total mAh consumed' to track how much is left in your battery? This will even take into account the consumption while sitting on tha starting block, ready to go, and the effect of different throttle levels while in flight.

eumelOne, 2019-03-26:
Why not? Shouldn't it be possible to use the same value like for the reset of the launch mode? Or introduce a new treshold based on the throttle position. This should give a good indication if the flight has started or the quad still sitting on the ground. 

The mAh used alone gives me no indication if i have to push or not in a competition.

mikeller, 2019-03-26:
@eumelOne: Feel free to open a pull request implementing a good way to detect being airborne.

> I am flying my 6s1300mah at the edge. We are have to fly 2:00-2:20 minutes in german competitions. 2:20 is really hard with my setup and i have to maintain my lipo. So thats what i need the "real flight time" for.

> The mAh used alone gives me no indication if i have to push or not in a competition.

Looks like you really don't know what you want. Nobody can help you with that, closing this.

closed 2019-03-26 by mikeller