betaflight/betaflight #7862: TFMini LIDAR values are -1.0 all the time (3.5.7)

andrejpodzimek, 2019-03-24:
While using [this LIDAR](https://www.sparkfun.com/products/14588), the Sensors tab shows -1.0 for "Sonar" all the time, so it looks like it doesn't work. When looking at the LIDAR with my cell phone camera (which can see infrared and shows it as pink/purple), the LIDAR does emit infrared light. The LIDAR is on UART 4. The FC is an [`ANYFCF7`](https://github.com/betaflight/betaflight/wiki/Board---ANYFCF7), the one with 10 big PWM outputs.

I'm using Betaflight 3.5.7, modified to support my octocopter with DShot.<sup>1</sup> (I don't think it's relevant, but I'll include all the config and code diff below.)

[image]

[image]

```shell
# diff all

# version
# Betaflight / ANYFCF7 (ANY7) 3.5.7 Mar 24 2019 / 10:09:51 (b2dc41a2b) MSP API: 1.40

board_name ANYFCF7
manufacturer_id 
mcu_id 003900263036510335383333
signature 

# reset configuration to default settings
defaults nosave

# name
name Kraken

# resources
resource MOTOR 1 A03
resource MOTOR 2 C09
resource MOTOR 3 B04
resource MOTOR 4 B05
resource MOTOR 5 B08
resource MOTOR 6 A02
resource MOTOR 7 A00
resource MOTOR 8 B03
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_RX 1 NONE

# mixer
mixer OCTOFLATX

# servo

# servo mix

# feature
feature GPS
feature RANGEFINDER
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 1 1 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 32768 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 6 2 115200 57600 0 115200
serial 7 16384 115200 57600 0 115200

[... 317 lines cut ...]
D10: FREE
D11: FREE
D12: I2C_SCL 4
D13: I2C_SDA 4
D14: FREE
D15: FREE
E00: SERIAL_RX 8
E01: SERIAL_TX 8
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: SERIAL_RX 7
E08: SERIAL_TX 7
E09: FREE
E10: FREE
E11: SPI_PREINIT_IPU 1
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: FREE
```

<sup>1</sup> *The table [shown here](https://github.com/betaflight/betaflight/wiki/Board---ANYFCF7#output-pinout) won't give you a DShot octocopter; the default configuration can use only 6 motors and the table extends this to 7 motors only (due to a conflict on `DMA1_ST4`). You need a configuration similar to the one above to get a DShot octocopter.*

andrejpodzimek, 2019-03-24:
Forgot to mention: This machine has had >3 3.5.x versions on it and the problem was the same each time. So it's not specific to 3.5.7.

etracer65, 2019-03-24:
Try testing by logging with `set debug_mode = LIDAR_TF`. Then debug items will be:
```
DEBUG_SET(DEBUG_LIDAR_TF, 0, distance);
DEBUG_SET(DEBUG_LIDAR_TF, 1, strength);
DEBUG_SET(DEBUG_LIDAR_TF, 2, tfFrame[4]);
DEBUG_SET(DEBUG_LIDAR_TF, 3, tfFrame[5]);
```
If this isn't showing anything then communication with the sensor isn't working.

andrejpodzimek, 2019-03-25:
@etracer65  Thanks for the hint. Because I don't have BlackBox set up (and have never used it), I found it easier to parse [the TFMini protocol](https://cdn.sparkfun.com/assets/5/e/4/7/b/benewake-tfmini-datasheet.pdf) [page 7]. I'm attaching my implementation. I think it shows more data than the debug output would. (The disadvantage is that it runs on `serialpassthrough`, not during a flight.)

**Long story short:** I'm seeing lots of invalid packets from TFMini. :confused: Packets that are valid show a height that makes perfect sense.

These are the first 10000 lines of my LIDAR reader's output: [lidar_reader_output.txt](https://github.com/betaflight/betaflight/files/3001082/lidar_reader_output.txt)

* There are numerous valid packets and the distance makes sense. The drone was at the edge of a table and the LIDAR was pointing down at the floor (real distance: 80.5 cm). Lifting the drone also gives sensible output changes [not shown here].
* There are numerous broken packets with invalid checksums (that are wrong, but not totally random; there seems to be a pattern of some sort). Sometimes they appear alone, sometimes in bursts. The bursts look somewhat regular/periodic.
* There is one case of a packet misalignment &mdash; instead of reading 2 bytes to get a valid packet header (`{0x59, 0x59}`), 10 bytes had to be read (`Read 10 B before getting packet header!`). I've seen more of these cases while experimenting.

The LIDAR reader I've used is below.

```c
// TFMini LIDAR protocol reader
// https://cdn.sparkfun.com/assets/5/e/4/7/b/benewake-tfmini-datasheet.pdf
//
// To build this tool:
// clang -Wall -Wextra -march=native -O3 -std=c11 lidar_reader.c -o lidar_reader
// You may need -lrt when building in an old environment.
//
// To set things up in the Betaflight CLI (LIDAR on UART4):
// serialpassthrough 3 115200
//
// After exiting the CLI, run the tool (e.g. with your FC on /dev/ttyACM0):
// ./lidar_reader /dev/ttyACM0

#define _POSIX_C_SOURCE 200809L
#include <errno.h>
#include <fcntl.h>
#include <inttypes.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <time.h>
#include <unistd.h>

#define FAILURE (-1)
#define CHECK(call) \
    ({                                                                         \
        ssize_t retval = (call);                                               \
        if (retval == FAILURE) {                                               \
            printf(                                                            \
                "LIDAR Reader: in %s on line %" PRIu32 " : %s : %s.\n",        \
                __FILE__, __LINE__, #call, strerror(errno));                   \
            exit(EXIT_FAILURE);                                                \
        }                                                                      \
        retval;                                                                \
    })
#define CHECK_EOF(call)                                                        \
    ({                                                                         \
        ssize_t retval = (call);                                               \
        if ((retval) == 0) {                                                   \
            printf(                                                            \
                "LIDAR Reader: in %s on line %" PRIu32 " : Unexpected EOF.\n", \
                __FILE__, __LINE__);                                           \
            exit(EXIT_FAILURE);                                                \
        }                                                                      \
        retval;                                                                \
    })

static const size_t BYTE_HEADER_0 = 0;
static const size_t BYTE_HEADER_1 = 1;
static const size_t BYTE_DIST_LOW = 2;
static const size_t BYTE_DIST_HIGH = 3;
static const size_t BYTE_STRENGTH_LOW = 4;
static const size_t BYTE_STRENGTH_HIGH = 5;
static const size_t BYTE_RESERVED = 6;
static const size_t BYTE_RAW_QUAL = 7;
[... 53 lines cut ...]
                alignment_read += bytes;
                if (buffer[BYTE_HEADER_1] == PACKET_HEADER) {
                    break;
                }
            }
        }
        // Check how many bytes it took to find the packet header.
        if (alignment_read != HEADER_SIZE) {
            CHECK(printf("Read %" PRIuMAX " B before getting packet header!\n",
                         alignment_read));
        }
        // Reading the rest of the packet.
        size_t left = PACKET_SIZE - HEADER_SIZE;
        do {
            bytes = CHECK_EOF(CHECK(read(tty, &buffer[HEADER_SIZE], left)));
            left -= bytes;
        } while (left > 0);    
        print_packet(&buffer);
    }
}
```

A noteworthy thing is [this other documentation](https://cdn-shop.adafruit.com/product-files/3978/3978_manual_SJ-PM-TFmini-T-01_A03ProductManual_EN.pdf) that mentions [page 5] two different data formats (*standard format* implemented above and a string-based *Pixhawk format*) with commands that switch between them. But a format mismatch is likely **not** the reason for broken packets shown in the output, because there's nothing that could be switching the TFMini between the two modes so many times &mdash; my reader doesn't write into the `serialpassghrough` port at all.

Should I perhaps order and try another TFMini, to rule out the possibility of this one being faulty? (But it's still more likely that there's bug in my reader. :grin:)

etracer65, 2019-03-25:
Regarding not getting any readings: the problem is that rangefinder support has been removed since 3.4. The sensor driver is still there but the sensor is never read. Even if it was read there's nothing Betaflight could do with the data except possibly supply altitude numbers since altitude hold, etc. have also long been removed. For this reason rangefinder was disabled.

andrejpodzimek, 2019-03-25:
@etracer65 I did not expect anything more than having the data in telemetry and OSD.

IIRC, the Betaflight configurator did not have a “Sonar” checkbox in the Sensors tab until recently. (Or maybe I just never noticed it…?) Now that the checkbox is there, it was the first time I looked at the data. Before that I assumed that lidar data was either reconciled with baro data or just unused. The “Sonar” checkbox made me somewhat hopeful that at least the data would be there.

The invalid packets from the TFMini look concerning to me. This looks like it wouldn’t work even if Betaflight did read the sensor. But if the `-1` is caused by the sensor being ignored (rather than disabled after an invalid packet), then this is probably not worth further debugging.

mikeller, 2019-03-25:
@andrejpodzimek:

> IIRC, the Betaflight configurator did not have a “Sonar” checkbox in the Sensors tab until recently. (Or maybe I just never noticed it…?)

I think you never noticed it - there has not been much ongoing development for sonar / LIDAR in a while.

andrejpodzimek, 2019-03-25:
I see. Perhaps I had had the LIDAR misconfigured to the point that the checkbox didn't appear.

As for the `-1`, is there still a way to get the data from the driver (something I can hack in reasonable time) or is this functionality too far gone?

(Now that I've seen all the invalid packets from the TFMini, I'm also curious whether this is a faulty TFMini (and if not, then what those "other" packets mean).)

Maybe I should try iNav in this case&hellip; But I'm not sure it would support an octocopter. (It took quite a few hacks and re-wiring to sort out the DShot / DMA / timer conflicts.)

mikeller, 2019-03-25:
@andrejpodzimek: I think the functionality will need a thorough check up to find and fix bugs that have crept in, and then it might make sense to wire it up to an altitude display (or a 'close to ground' warning or similar) in the OSD. It will be some work, for not so much practical gain, so not sure if there is a developer with time on his hands and the inclination to spend it on fixing this.

Re iNav, that might be a good option. The Dshot / DMA / timer issue will most likely be less of a problem there, as it seems to be geared towards using PWM as the motor protocol, which does not need DMA for motor output.

andrejpodzimek, 2019-03-25:
Digging into where the `-1` comes from: This can happen when the distance is out of range (`TF_MINI_RANGE_{MIN,MAX}`), but that's not the case (based on what I see on `serialpassthrough`). So at the moment my gues is that it's [this initial `RANGEFINDER_OUT_OF_RANGE`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L190) value, because [`rangefinderProcess()`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L300-L366) doesn't seem to be called. The Betaflight scheduler [doesn't call it](https://github.com/betaflight/betaflight/blob/aed1e51cf6588630ff847ba460ca4d0a6397a933/src/main/fc/tasks.c#L472) (and calls only [`rangefinderUpdate()`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L258-L267)), whereas [iNav does call it](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/fc/fc_tasks.c#L199). Betaflight also does't have [`calculateCosTiltAngle()`](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/flight/imu.c#L626-L629) to get the argument for `rangeFinderProcess()`.

> &hellip;it might make sense to wire it up to an altitude display&hellip;

Apart from the altitude display, I was also thinking, given the impressive precision of LIDAR, that it could calibrate the baro data interpretation, to improve baro readings once the drone climbs above the LIDAR range. (One would have to be careful with `cosTiltAngle`, of course.)

> &hellip;so not sure if there is a developer with time on his hands and the inclination to spend it on fixing this.

At the moment I do have a bit of spare time. If altitude-related functionality is merely "removed until rewritten and fixed" (rather than "removed forever"), I'd try and look into it. But I don't want to reintroduce stuff that Betaflight wants to get rid of; such effort would be bound to fail.

> Re iNav, that might be a good option. The Dshot / DMA / timer issue will most likely be less of a problem there, as it seems to be geared towards using PWM as the motor protocol, which does not need DMA for motor output.

iNav has DShot, albeit [with a scary warning](https://github.com/iNavFlight/inav/wiki/2.1.0-Release-Notes#important-notes). As for analog protocols &hellip; I like BLHeli_32 telemetry. From the BLHeli_32 side, telemetry could work with analog protocols &mdash; there's an *auto telemetry* option since version 32.51. But I think it wouldn't work with the usual "all ESCs on one RX" wiring; all ESCs would send telemetry all the time and clash. So telemetry depends on DShot.

Looking at the LIDAR drivers, I see that [Betaflight drops invalid packets](https://github.com/betaflight/betaflight/blob/54f59280bb47be29bcb2dff32c1c938c9d155081/src/main/drivers/rangefinder/rangefinder_lidartf.c#L213-L215) and [so does iNav](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/io/rangefinder_benewake.c#L116). So these may not be uncommon.

mikeller, 2019-03-26:
@andrejpodzimek: Sonar / LIDAR support was 'moved out of the way' when GPS rescue was introduced, to allow this to happen in a minimal, but workable form, without having to worry about interaction with other systems. It was never the idea to permanently remove this, it is just a case of finding somebody willing to work on it, as there is not much that 'altitude above ground' information can be used for in Betaflight. So you are very welcome to do work on fixing this up and making it work again!

Not sure if 'altitude above ground' could be used for altitude calibration - this will only work if the ground is flat (which we don't know), or the craft is not moving (which would require GPS information to be tied in as well). But even 'only' 'altitude above ground' information can be quite valuable, as this is what pilots really care about (and not absolute altitude) when they are trying to land smoothly.

ESC telemetry is not restricted to digital protocols - in fact, the 'separate wire' ESC telemetry protocol that is used by Dshot was defined by KISS (thanks @ronlix for the original design for this!) to work with OneShot42 before it was 'adopted' by Dshot when Dshot was defined. It works well there, but it has the limitation that OneShot42 is the fastest protocol it can be used with, as a fixed length pulse that is longer than the longest expected OneShot42 pulse is used to trigger telemetry transmission for an individual ESC.
This was never adopted by Betaflight, since the maximum 4kHz PID loop time possible with OneShot42 was considered to be slow when ESC telemetry support was added. But with iNav being much less focused on fast loop times, I think 'analog ESC telemetry' might be a valuable addition there - knowing things like ESC temperatures will be a welcome addition for long range flyers.

andrejpodzimek, 2019-05-02:
I'm finishing my second LIDAR build (a tricopter) slowly, but surely. (And poking this just to prevent stale bot from closing it.)

closed 2019-06-08 by stale[bot]

HenryHuYu, 2024-08-07:
> Digging into where the `-1` comes from: This can happen when the distance is out of range (`TF_MINI_RANGE_{MIN,MAX}`), but that's not the case (based on what I see on `serialpassthrough`). So at the moment my gues is that it's [this initial `RANGEFINDER_OUT_OF_RANGE`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L190) value, because [`rangefinderProcess()`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L300-L366) doesn't seem to be called. The Betaflight scheduler [doesn't call it](https://github.com/betaflight/betaflight/blob/aed1e51cf6588630ff847ba460ca4d0a6397a933/src/main/fc/tasks.c#L472) (and calls only [`rangefinderUpdate()`](https://github.com/betaflight/betaflight/blob/4134c9c7906ecbcde0c9e4a2c734db486da8c3c4/src/main/sensors/rangefinder.c#L258-L267)), whereas [iNav does call it](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/fc/fc_tasks.c#L199). Betaflight also does't have [`calculateCosTiltAngle()`](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/flight/imu.c#L626-L629) to get the argument for `rangefinderProcess()`.
> 
> > …it might make sense to wire it up to an altitude display…
> 
> Apart from the altitude display, I was also thinking, given the impressive precision of LIDAR, that it could calibrate the baro data interpretation, to improve baro readings once the drone climbs above the LIDAR range. (One would have to be careful with `cosTiltAngle`, of course.)
> 
> > …so not sure if there is a developer with time on his hands and the inclination to spend it on fixing this.
> 
> At the moment I do have a bit of spare time. If altitude-related functionality is merely "removed until rewritten and fixed" (rather than "removed forever"), I'd try and look into it. But I don't want to reintroduce stuff that Betaflight wants to get rid of; such effort would be bound to fail.
> 
> > Re iNav, that might be a good option. The Dshot / DMA / timer issue will most likely be less of a problem there, as it seems to be geared towards using PWM as the motor protocol, which does not need DMA for motor output.
> 
> iNav has DShot, albeit [with a scary warning](https://github.com/iNavFlight/inav/wiki/2.1.0-Release-Notes#important-notes). As for analog protocols … I like BLHeli_32 telemetry. From the BLHeli_32 side, telemetry could work with analog protocols — there's an _auto telemetry_ option since version 32.51. But I think it wouldn't work with the usual "all ESCs on one RX" wiring; all ESCs would send telemetry all the time and clash. So telemetry depends on DShot.
> 
> Looking at the LIDAR drivers, I see that [Betaflight drops invalid packets](https://github.com/betaflight/betaflight/blob/54f59280bb47be29bcb2dff32c1c938c9d155081/src/main/drivers/rangefinder/rangefinder_lidartf.c#L213-L215) and [so does iNav](https://github.com/iNavFlight/inav/blob/65b0ec142c5b27bb5f0e8ea14f8d0627cff7ca76/src/main/io/rangefinder_benewake.c#L116). So these may not be uncommon.

hi did u get rangefinder work on bf finally?