betaflight/betaflight #7863: Absolute Control broken

bexamous, 2019-03-24:
https://github.com/betaflight/betaflight/blob/1bdbd1cd72158ffebdb6f0084f1bdef75fe7835c/src/main/flight/pid.c#L1352

pidData[axis].P is calculated with errorRate, which is calculated with currentPidSetpoint before the call to applyItermRelax.

Problem is currentPidSetpoint is passed to applyItermRelax which then passes it to applyAbsoluteControl which modifies currentPidSetpoint.

errorRate needs to be calculated after the call to applyItermRelax or else absolute control doesn't really work.

This is a regression from:
https://github.com/betaflight/betaflight/commit/960442e91757092b81d186a5f513d83173125634

mikeller, 2019-03-26:
@bexamous: Can you please test to see if this has been fixed with 4.0.0-RC4?

aowi7280, 2019-03-26:
Build 1537 had terrible yaw bounce back. I turned the I up to 175 to get rid of most of it. I just tried build 1544 and it is better, but not perfect. Yaw I is default. I can tell abs_control_gain = 0 has no yaw bounce back and abs_control_gain = 10 has a small amount. I use 2100kv motors and 3S.

robert-hoffmann, 2019-04-09:
Has this been fixed, or temporarily just disable the time it takes to get it sorted out ?

mikeller, 2019-04-09:
@robert-hoffmann: See #7866.

robert-hoffmann, 2019-04-09:
I was more wondering if it's currently better to stick with i-term rotation, since defaults where rolled back to that in build #1553

Was hoping that the closing of this issue meant that abs control is the default again.

bexamous, 2019-04-09:
See here for discussion on defaults and why enabling iterm rotation again is not good idea: 
 https://github.com/betaflight/betaflight/pull/7901

Abs control, even with this issue fixed, seems to be at least partically to blame for some wobbles some have seen in testing, see betaflight slack for ongoing results.

robert-hoffmann, 2019-04-09:
Thanks for the heads up 👍 

Just as a sidenote, i'm running build 1551 with default pids/cutom rates, lot's of filters off (or reduced to PT1), and i-term rotation on, and can't say i've noticed anything strange yet.