betaflight/betaflight #7867: FrSky X EU LBT connection "flickers" on Matek F411-ONE

mankeli, 2019-03-24:
**Describe the bug**
Binding the RX & TX works fine, telemetry shows 100% all the time, but for some reason the connection randomly drops and failsafe values are activated on the FC. Then after few seconds it comes back on. This seems to happen constantly. At the workbench with low proximity, but also at the field while flying and copter is further away.

**To Reproduce**
Bind rx & tx, d16 eu-lbt. Observe the behavior in "receiver" tab in the configurator.

**Expected behavior**
Connection should work okay, without RXLOSS happening. Official receivers somewhat work even without antennas attached. 

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Matek F411-RX, AOI with spi rx.
 - Taranis X9D Plus. Can't find info which fw version it has.

Asizon, 2019-03-25:
For the moment,you need to disable telemetry to solve this problem. Frsky_X is not conpatible with telemetry an Lua for the moment.

mankeli, 2019-03-25:
I don't have lua scripts, but I'll try to disable telemetry. Though docs say that "integrated FrSky SmartPort telemetry (including support for MSP over Smartport / lua scripts in OpenTX);".

mankeli, 2019-03-26:
@Asizon Where I can disable telemetry so it should work? Atleast disabling telemetry output from the configuration tab doesn't help the situation.

Asizon, 2019-03-26:
Hi @mankeli I have to correct the documentation in reference to that problem. I am flying without problem in both crazybeeFr and MatekF411rx. You need to have latest version 4.0, activate Frsky_X, and disable telemetry in the configuration tab of Betaflight. With this you should not have any problem

phobos-, 2019-03-26:
@mankeli, could you record a dvr footage with debug_mode=RX_FRSKY_SPI, and debug enabled in the OSD? I'd like to see what's going on in flight. Receiver tab in the configurator can show some packetloss, but in the air it should be stable.

@Asizon, thanks for testing, I also don't have that problem.

mankeli, 2019-04-01:
My FC's usb port stopped responding few days ago, I'll see if I have time soon to try to get it working again.

closed 2019-05-16 by stale[bot]