betaflight/betaflight #7868: allow user to specify number of decimals for vbat osd

eamedia, 2019-03-25:
**Is your feature request related to a problem? Please describe.**
the constantly changing hundredth in vbat osd is annoying
since 4.0 my vbat osd reads 2 decimals eg. 15.55v

**Describe the solution you'd like**
allow me to specify to the tenths or hundreds for vbat osd reading eg. 15.6v

krzysztofmatula, 2019-03-25:
Maybe the 2 vs 1 decimal digit display could be selected automatically based on battery cell count?
For 1s or 2s display 2 digits (e.g. 3.33V)
For 3s+ display 1 digit (e.g. 15.6V)

AndersHoglund, 2019-03-25:
A more general approach would be to specify number of significant digits in total, not the number of decimals nor the number of cells . I.e. 2, 3 or 4 digits. For example "9.9", "9.92" or "9.921". Or "16", "15.5", "15.55" when applying correct rounding. Default could be three digits.

PR #7952 (2019-04-15): Show only 3 significant digits in vbat OSD element
Fixes #7868 Constantly changing hundredth in vbat osd is annoying. This PR changing this behaviour to show only 3 significant digits rounded to tenth, for example 12.3v or 3.70v.
closed 2019-04-15 by mikeller