betaflight/betaflight #7869: Enable passthru mode for quadrotors

HYC23, 2019-03-25:
I have a question about how to enable passthru mode for quads. I try the aux command in CLI, but there is no passthru mode in Modes tag on the betaflight software, and it is still airmode when I use transmitter to control it. 
I want to use this for testing a offboad controller, so is there a way to enable it for quads?

mikeller, 2019-03-25:
@HYC23: You need to select the 'Airplane', 'Custom Airplane', or 'Flying Wing' mixer to get access to the passthru mod: https://github.com/betaflight/betaflight/blob/6f93d526b72f9382305f53e275eeb5ca28ddae8e/src/main/msp/msp_box.c#L204-L206

Using 'Custom Airplane' and manually entering the mixer settings for a quad should work in your case.

HYC23, 2019-03-25:
> @HYC23: You need to select the 'Airplane', 'Custom Airplane', or 'Flying Wing' mixer to get access to the passthru mode:
> 
> [betaflight/src/main/msp/msp_box.c](https://github.com/betaflight/betaflight/blob/6f93d526b72f9382305f53e275eeb5ca28ddae8e/src/main/msp/msp_box.c#L204-L206)
> 
> Lines 204 to 206 in [6f93d52](/betaflight/betaflight/commit/6f93d526b72f9382305f53e275eeb5ca28ddae8e)
> 
>  if (mixerConfig()->mixerMode == MIXER_FLYING_WING || mixerConfig()->mixerMode == MIXER_AIRPLANE || mixerConfig()->mixerMode == MIXER_CUSTOM_AIRPLANE) { 
>      BME(BOXPASSTHRU); 
>  } 
> Using 'Custom Airplane' and manually entering the mixer settings for a quad should work in your case.

Thanks your reply.
I will try this method, but where should I enter the mixer? Is this form the betaflight or in my transmitter?

mikeller, 2019-03-25:
@HYC23: In Betaflight, in CLI, see https://github.com/betaflight/betaflight/blob/master/docs/Mixer.md#custom-motor-mixing.

closed 2019-05-01 by stale[bot]