betaflight/betaflight #7873: F7 VCP not completely transmitting buffer until empty

4712, 2019-03-25:
I can reproduce a windows client side read timeout issue here which seems to occur unpredictable every now an than. 
The bytes in the serial VCP buffer seems not be completely send, until empty.
This leads to failure of reading through the serial-4way-interface, but the interface code seems not the cause.

See attached test version of BLHeliSuite32. [BLHeliSuite32_32610TestVCP.zip](https://github.com/betaflight/betaflight/files/3003420/BLHeliSuite32_32610TestVCP.zip)

A "Test Data Flash read" loop is implemented, which expects 246 data bytes from address 0.

[image]

Mostly it fails here after several hundred cycles. The timeout is 500ms. 
[image]
Reading again bytes from the port fails, but sending the next MSP_DATAFLASH_READ gets the missing bytes in the buffer plus the next answer.

[image]

Tested here with OnniXNT F7 (BF 3.57 + 4.RC1,2,3) and Asgard32 F7 BF 3.3.0.
Noting else is attached to the board.

I do not see a similar behavior with F4 FC boards.

BTW: There is also a test loop for connecting the ESCs
[image]

mikeller, 2019-03-25:
@4712: Not sure how much point there is in discussing problems with Betaflight 3.3 or Betaflight 3.5, as there is little chance that any of these will see another release.

Have you had a look into making the changes to the serial 4way protocol that are required to make it work with timer resource management in Betaflight 4.0?

4712, 2019-03-25:
@mikeller: I only tested with 3.3, in order to verify that the issue is nothing newly invented...
It exists in the current build of BF 4.0 and exist in older releases as well and it exists independent from serial 4way interface code. 

As far as I can see, the serial 4way protocol is not changing anything about the timer management.
Also the usage of MSP command MSP_DATAFLASH_READ shows the issue as well and the implementation is completely independent from serial 4way interface code.

mikeller, 2019-03-25:
@4712: For next time, if an issue exists with current `master`, then there is little point in reporting it as existing with legacy versions of the firmware.

There have been a number of USB / VCP related changes in 4.0, so you should probably re-test with non-legacy firmware, and then report your findings, including the configuration used (as per instructions you were given when opening the issue), so it can be reproduced easily.

As for resource management, when I tested 4way serial with a Unified Target about a month ago, this did not work, potentially because the driver is trying to read the default motor assignments instead of reading the motor assignments that are actually configured. So unless this is fixed, there will be no more serial 4way interface support for F4 / F7 at some point in 4.0.

4712, 2019-03-25:
@mikeller: May be, I expressed it a bit unclear, when I wrote "Tested here with OnniXNT F7 (BF 3.57 + 4.RC1,2,3) and Asgard32 F7 BF 3.3.0."
Lets say it more clear: I tested with BF 4.0.0-RC1, BF 4.0.0-RC2, and BF 4.0.0-RC3 and other. Isn't RC3 recent enough? 
And as I wrote, nothing else connected to the FC board, so configuration is simply default.

mikeller, 2019-03-25:
@4712: Yes, I did not interpret this as a Betaflight version number, I thought you were referring to an ESC firmware with it.

So have you tested with a Unified Target or not? Because if we don't get this to work with Unified Targets then fixing this won't make much of a change, as serial 4way support will go away when all F4 / F7 targets are switched to use Unified Targets.

4712, 2019-03-25:
@mikeller: how do I test with a Unified Target?

mikeller, 2019-03-25:
@4712:

- download and install the Unified Target (STM32Fxxx) for your MCU type;
- download the config for your board from `unified_targets/configs` (assuming there is one), and paste it into CLI, followed by `save'.

If there is no Unified Target config for the board you are testing with, you can create one yourself: https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md

4712, 2019-03-25:
@mikeller: Thanks, I will give it a try.

4712, 2019-03-25:
@mikeller: that was almost easy ;) 
Did it with STM32F7X2 for OMNINXT7.
I got one error when doing the CLI batch:
`# resource ESCSERIAL 1 B07
###ERROR: INVALID RESOURCE NAME: 'ESCSERIAL'###`

Anyway, I tested the serial 4way interface just fine (beside of the here already mentioned VCP tx buffer issue),  even with a defective output, I had to remap.

[OMNINXT7.config.txt](https://github.com/betaflight/betaflight/files/3005269/OMNINXT7.config.txt)
(No idea if something is missing in the config)

While I'm hopefully successfully working on the workaround in BLHeliSuite32 for the legacy BF versions, it would be good to have the F7 VCP issue fixed in BF 4 release.  Unfortunately, I'm not I have no idea where to look for.

etracer65, 2019-03-25:
Unfortunately I think any bug related to this is deep within the STM HAL library in the USB endpoint handling. The higher level code has been examined quite thoroughly recently and seems unlikely that it could cause this problem.

mikeller, 2019-03-26:
@4712: One of the goals we want to achieve with Unified Targets is to make them as easy to use for users as the current targets are, if not better, once we've added support for loading the Unified Target configuration directly after flashing in configurator. So good to hear that the firmware side is almost there. :-)

Thanks for testing, that's one of the points of my list of things to get working for 4.0 ticked off.

Re the USB / VCP bug, I think what we found with `MSP_DATAFLASH_READ` is that reducing the frame size fixes the problem to some extent - is that an option that could be used for the 4way protocol as well?

4712, 2019-03-26:
@etracer65: this is what I feared...
@mikeller: for the 4way interface, I already tried to reduce the buffer size to 64 without success. I will check this for MSP_DATAFLASH_READ asap, but I assume, it will cure nothing. The fix for now is to send a simple dummy command and get the missing bytes right ahead of the next response.

4712, 2019-03-29:
@mikeller, @etracer65  I decreased the buffer to 64 bytes, which did not help, but it seems to work if we reduce the total tx frame size to max 63 byte. Sending 64 or a multiple of 64 bytes definitively fails.

So, it seems to be a bug related to the USB ep size. 
Searching the net for 64 byte bug in STM32 USB (HAL) driver I found 2 hints:
1. Sending a zero packet after multiple of 64 bytes
2. a hint for bug in stm32f7xx_hal_pcd.c (PCD_WriteEmptyTxFifo)

I tried a fix for PCD_WriteEmptyTxFifo by exchanging some lines with the ones of a newer STM HAL driver (en.STM32Cube_FW_F7_V1.15.0) version but it did not work. Maybe some more changes or upgrading the whole HAL driver might help here... or invent new bugs ;)

mikeller, 2019-03-29:
@4712: Thanks for looking into this. 63 bytes is small, and will most likely not be of much use if used for data flash transfers, but it could be enough for your purposes with 4way serial, and it is a fix that is available right now.

4712, 2019-03-29:
I did a quick and dirty upgrade of the HAL to STM32Cube_FW_F7_V1.15.0 (used the old usbd_cdc_hid.*) This worked to fix the 64 byte bug.

mikeller, 2019-03-29:
@4712: Sounds promising. Unfortunately we have had to make a number of tweaks to the currently used library to address some other bugs, and it will take some time to port them to the new version, plus it will need some (or rather a considerable amount of) exposure to get confidence that the new version will not break us in other, unexpected places. But we can look into updating the library after 4.0 has been released, to go into 4.1.
For now, and to make things work with 4.0, finding a workaround that reduces the buffer size to 63 bytes, is the best approach.

closed 2026-09-26 by blckmn