betaflight/betaflight #7874: Motors M2 + M4 won't spin w. camara control enabled

FranzPVal, 2019-03-26:
Both motors (M2/M4) won't spin after I set up my resource remapping on my AIO FC with integrated ESCs (Racerstar TattooF4S, Target LUXF4OSD).

`board_name LUXF4OSD
manufacturer_id 
mcu_id 004000273436510736343532
# resources
resource BEEPER 1 NONE
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SONAR_TRIGGER 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A01
resource CAMERA_CONTROL 1 B04
`
As soon I set up CAMERA_CONTROL 1 to B04 the bug is happen (I Tried DShot1200 and MultiShot). The motors will work again if I freed this resource.

To Reproduce:
1. update to BF 4.0.0RC2 or RC3 (tested) with full chip erase
2. set to DShot1200
3. test motors (should work)
4. apply resource remapping except last line
5.  (work now in 90° remapped order)
6. apply "resource CAMERA_CONTROL 1 B04"
7. test motors again (now both front motors won't spin anymore)

etracer65, 2019-03-26:
Not sure why you're trying to use B04 as the camera control. For that target B04 is the pin used for the buzzer and likely won't work for camera control (as there's likely a transistor in the circuit to drive the buzzer).

You'd probably be better off using A08 which for that target is MOTOR 6 pin which I see you've already resourced to `NONE`.

FranzPVal, 2019-03-26:
Yes - should try it but in earlier versions this havn't work (Cam Control on A08). So I soldered it to the MoFet gate pin...

mikeller, 2019-03-26:
@FranzPVal: Not sure that what you are trying to do is supposed to work - you most likely have a resource conflict between camera control (on B4) and motors 2 & 4 (on B0, B1). What does `dma show` tell you?

closed 2019-05-02 by stale[bot]