betaflight/betaflight #7875: BetaFlight OSD firmware on External OSD (MinimOSD) boards

linuxdude21, 2019-03-26:
Hey all,

I have seen and heard people using BetaFlight based FCs purely for the OSD.  Sometimes it's two BF FCs in the same quad, which is a head scratcher for me.  There are also a lot of us that have older quads that have FCs without integrated OSD chips (AB7456) and have external (Micro) MinimOSD Board running MWOSD code.  It works, but honestly, my favorite OSD firmware is BetaFlight's.  The ease to use the UI to configure everything and upload your favorite font is great.  Along with having CMS (i know MWOSD allow CMS as well).

I know this would be used very selectively and not a great demand.  But is it possible?  I know these MinimOSD based boards have Atmel procs. Not sure if that is a pitfall.  

Thanks,
Nick

mikeller, 2019-03-26:
@linuxdude21: Looking at the resources available on the 8 bit MCUs used on typical MWOSD boards, I'd say there is little chance to be able to run Betaflight in a meaningful way on there - and certainly not with the configurability and feature richness that you say is what users want.
I don't know if you are aware, but it is possible to use the CMS to configure Betaflight even with an OSD being generated on an external MWOSD (this works fine), or even to have Betaflight generate the entire in-flight OSD on such a device (this works ok, as long as there are only few elements used, due to the limitation of UART speed that these devices have).

linuxdude21, 2019-03-27:
@mikeller Thanks for your super quick reply like always.  

I am completely aware of CMS access via MWOSD menu.  But unaware of inflight data controlled by BF.  Can you send me a link or info on this?  I thought that the Atmel proc would be a bottleneck, but didn't think of the slow UART speed to the external OSD.  

The desire would be to have the ease of OSD element/display configuration and the BF OSD elements available on MWOSD.  I don't have a great understanding of MWOSD, but it's not a flexible as BF OSD is.  I could be completely wrong.  But if it is possible the BF controls the OSD through MWOSD I would like to see more info on that?

Thanks again.

mikeller, 2019-03-27:
@linuxdude21: The in flight data can be displayed by an MWOSD device by acting as a 'dumb' canvas device in just the same way as what is done when the CMS menus are displayed. I've done some experimentation with this, but at the 115200 baud that is the maximum speed supported by the Atmel MCU running Arduino code, it can not keep up with displaying more than a few numerical elements. I will have to dig to see if I can find the config that I used for this, I don't think there are any instructions to get this to work published anywhere.
This facility was introduced by @hydra, and used to drive the OSD on his SPRACINGF3OSD, so maybe he can help here?

> The desire would be to have the ease of OSD element/display configuration and the BF OSD elements available on MWOSD. I don't have a great understanding of MWOSD, but it's not a flexible as BF OSD [...]

As I said before, the flash space of the MCU used on MWOSD is definitely not capable of supporting more configurability than what MWOSD has - when building MWOSD with a lot of features enabled, it is currently necessary to use a minimised boot loader in order to make the firmware fit into flash, so we already are at the limit.

robert-hoffmann, 2019-04-08:
Hmmm, i have a spracing f3 deluxe with integrated OSD,
http://www.phasercomputers.com.au/wp-content/uploads/2016/08/sp-racing-f3-osd-manual.pdf

..it's a clone, and the OSD runs MWOsd, which is setup via a seperate port

To the point ..the board is actually currently displaying information from the BF OSD configuration. The only problem is some font issues, but i can turn stuff on/off in BFC and even reposition them and it displays mostly correctly on this board.

So i guess if this board can do it, why shouldn't a seperate MWOsd be able to do the same if wired simalarily ..though no clue how it's done on this board.

mikeller, 2019-04-08:
@robert-hoffmann: Would be interesting to know what settings you are using on the flight controller, and what firmware you are using on the MWOSD - it is potentially possible to drive the MWOSD's chip faster than than 115200 bps, there is just no way to do so from arduino, and the MWOSD firmware is written in arduino, so there is no easy way to get there.

robert-hoffmann, 2019-04-08:
In BF 4 i'm not using any setting in particular at all, i simply enabled osd, and the mwosd display disappeared, and the BF specific stuff appeared.

As far as mwosd itself, i don't really remember, i just pulled this controller out of a drawer and was surprised BF worked with it ^^

I think it's running mwosd 1.6 (or maybe scarab 1.6.8 ..probably mwosd), @11500 bauds, with just the `#define BETAFLIGHT` set in config.h

If you know of a command i can send to get the exact version, i'd be happy to try

robert-hoffmann, 2019-04-08:
If you have any tips for tuning/config, i'm all ~~ears~~ eyes ^^

Otherwise ..Well, it works just fine for what i need: RSSI, Voltage, Flight Time, Warnings

robert-hoffmann, 2019-04-09:
@linuxdude21 

If you flash your MWOSD with the settings i posted above, BF should "simply" work

You can find the last "free" version of it here:
https://github.com/ShikOfTheRa/scarab-osd/releases/tag/v1.6.8.0

Change the config.h to suit your needs, and then upload it to your MWOSD with the arduino ide
https://www.arduino.cc/en/main/software

mikeller, 2019-04-09:
@robert-hoffmann:

> After combing through the settings, i guess any MWOsd should work with betaflight just fine. I suppose all the magic happens here:

Yeah, I think 4 small OSD elements was about what I was able to make work relatively well in this configuration (SPRACINGF3 with on-board MWOSD) as well when I tried. Adding any more / bigger elements would cause the elements to flicker.

closed 2019-05-16 by stale[bot]