betaflight/betaflight #7876: Minor OSD stats screen layout issues

bizmar, 2019-03-26:
**Describe the bug**
In the OSD stats screen Max altitude is displayed offset from other lines
Max speed doesn't have a measurement unit.
Max ESC temp is displayed offset from other lines

**To Reproduce**
Put Max altitude on your OSD stats page. 
Put Max speed on your OSD stats page. 
Put Max ESC temp on your OSD stats page. 
Only tested in Metric with PAL

**Expected behavior**
Outputs to be padded correctly.
Value to have a measurement unit.

**Setup / Versions**
 BF 4.0  RC2

**Additional context**
[image]

etracer65, 2019-03-26:
Uh oh! Layout police are in the house! Just kidding of course. 😄 

Problem could be easily solved by having fewer OSD stats. 😉 But I'll look into it.

bizmar, 2019-03-26:
Stop adding random OSD elements! 😜 The little EQ meter thingy that shows your individual motor % is my favorite. 🙃

I need to set up a "movie" OSD profile with all the things enabled, do we have a vario graph?

etracer65, 2019-03-26:
No, just a numerical vario. Be aware that OSD elements are quite CPU intensive and with lots of them the OSD task can get quite long - impacting loop time stability. For an example look at the maximum time for the OSD task (do it a couple of times) to see the impact. The average doesn't represent it well because the OSD task runs in phases of drawing all the elements (the long part) followed by multiple loops of outputting characters to the OSD chip (relatively short parts).

PR #7877 (2019-03-27): Fix OSD stats alignment and formatting
Fixes #7876 

Corrects the alignment and formatting of the following stats:
* Max altitude
* Max speed
* Max ESC temperature
closed 2019-03-27 by mikeller