betaflight/betaflight #7878: 4.0-RC2 makes 2S whoop fly away

bluehallu, 2019-03-26:
**Describe the bug**

When lowering the throttle to 0 on a level position, the quad either falls down very slowly, maintains altitude, or on occasions actually gains altitude. Lowering I seems to help somewhat.

**To Reproduce**
Punch out and kill the throttle. 

**Expected behavior**
The quad falls

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
Eachine trashcan everything stock except gemfan 40mm 3-blade props. Clean flash with 4.0RC2.

**Additional context**
The trashcan uses Crazybee F4. This board comes from the factory with a nightly build of Betaflight 4.0 from December, so the problem must have been introduced in the last 3 months.

etracer65, 2019-03-26:
This is clearly not a "clean flash". Things that are at defaults don't show up in a `diff` so you've changed PID and filter settings. Reset to `defaults` and leave the PID and filter settings unchanged and test. The problem you're describing is caused by too much noise getting amplified by the D settings and you've changed both.

bluehallu, 2019-03-26:
@etracer65 By clean flash I mean that I did a full chip erase. My changes have mostly been in the OSD trying to fix the issue by lowering the pids. There's a facebook group for eachine trashcan owners and multiple people are reporting the issue when updating from the December build to the latest RC so I don't believe this is particular to my settings. I've updated the issue to reflect that the problem is not noticeable when flying indoors without airmode and on horizon mode.

etracer65, 2019-03-26:
None of that changes the advice. Go back to default settings for filters and PIDs.

bluehallu, 2019-03-27:
I will do that and report back

watageek, 2019-03-27:
@bluehallu 
What bugs me is airmode is off in your diff's feature. I would advice you to use a throttle up if you don't want to use airmode and to have a look on the suggested config in the tuning tips. 

But, If I was you I wouldn't use absolute control neither integrated yaw ( as a debug). Go the PID, old fashion way. 

One thing at time...

By the way etracer65 and joelucid has helped me a ton in the past. They are really keen minds. If he say default... go default. 🙄

OpenUAS, 2019-04-04:
@bluehallu it would be great if you find any time to report back. Since testing BF4.0RC6 I could not reproduce your behaviour on quad with CRAZYBEEF4FR (x) clean erased board.

RubenFelix, 2019-04-04:
I had the same problem. After disabling airmode, could not reproduce the wrong behavior anymore.
I will try once more to enable ir and test to make sure that something is not right with the airmode feature.

bluehallu, 2019-04-05:
I'll test this today. The reason airmode was off is that it's on a switch as I don't use it indoors.

bluehallu, 2019-04-05:
I tested this again after flashing RC6 with fully default pids and filters, the issue is exactly the same. I also tried without airmode and while this issue was gone, a different issue arised: when lowering the throttle the quad would suddenly yaw left about 45 degrees very rapidly and tilt right about 20 degrees. This behaviour wasn't random, it was the same every time. This happened regardless of acro or horizon mode   which means both configurations I used to use in this quad (acro with airmode / horizon no airmode) are  broken. I'm not sure these are connected though.

Going back to the motors winding up, I noticed some patterns I didn't notice the last time. If the quad has no momentum and is falling straight down, I can hear the motors winding up but not enough to cause a noticeable problem. If the quad is moving forward and I let it fall while looking at the ground, nothing happens. However, if I have momentum and I cut the throttle while maintaining a forward flight attitude, it becomes very noticeable. On a couple of occasions it was so bad that trying to go under a tree made it shoot up to the branches!

deepdigger, 2019-04-07:
Did you already try to turn telemetry off? I had a similar issue where my toothpick-build with crazybee f4 where sometimes the throttle kept stuck at 60% for a second or so... disabled telemetry and now everything worked fine (found the solution in comment #6172 seems to be a similar problem)

bluehallu, 2019-04-08:
@deepdigger Yes telemetry has always been off. I am sure this has nothing to do with the RX, I can hear the motors winding up organically. I'm pretty sure this is to do with filters or pid loop changes.

etracer65, 2019-04-08:
As stated earlier, the only thing that can cause a "fly-away" with airmode on vs. off is noise and vibrations affecting the PID controller. With airmode on the quad reacts much more strongly to movement. Vibrations are movement. You have too much vibration, not enough filtering, or too high PID gains (or a combination of all the above). If you can't produce a blackbox log to analyze the gyro noise then you're just going to have to adjust the filtering by guessing. Reducing the lowpass cutoff frequencies increases filtering. I would start with the dterm lowpasses.

bluehallu, 2019-04-08:
The quad is fine. I can revert back to the build it came with and it flies fantastically with no issues. The moment I update to latest version the issues appear.

etracer65, 2019-04-08:
That doesn't really answer any questions though. Filtering defaults have changed across versions. It's possible the old version you used had more conservative (meaning more) filtering so you can get away with the extra vibrations.

albertmonzo, 2019-04-09:
@etracer65 I've tried with new props no difference. I actually run a lighter version of the default filters (lpf1 110, lpf2 off, dlpf2 off) with that build and get good performance (https://www.youtube.com/watch?v=4ZL01HusmG4) and again, others that have tried the upgrade reported similar issues. I'm not saying there's a breaking change in 4.0, but rather that something about the defaults don't work for at least one model of 2S whoop and make it behave dangerously

bluehallu, 2019-04-09:
@etracer65 I've tried with new props no difference. I actually run a lighter version of the default filters (lpf1 110, lpf2 off, dlpf2 off) and significantly higher PID values (see video description) with that build and get good performance (https://www.youtube.com/watch?v=4ZL01HusmG4) and again, others that have tried the upgrade reported similar issues. I'm not saying there's a breaking change in 4.0, but rather that something about the defaults don't work for at least one model of 2S whoop and make it behave dangerously. Not just that, but I can't actually fix it by playing with filters or PIDs though that might just be me.

ctzsnooze, 2019-04-12:
Could you possibly upload a diff of the 4.0 code that flies well?  
Did you try pasting that diff (minus timer and DMA sections) into the current release of 4.0?

ArcsinX, 2019-04-15:
I have the same problem with Lizard 105s, 4s battery. In air mode it hovers well, but when I drop down throttle, it goes up like a rocket and continues to going up with fully down throttle, only disarm works to stop it. Also tried in stabilizer mode, and do not see that behavior.

ArcsinX, 2019-04-16:
The problem is still exist for me but in slightly different way:
1) It flies ok on 550mah 60C
2) If flyaway immediate after takeoff with 650mah 75c

When I switching to stab while it flyes away, it stops to flyaway and throttle response is normal. When I switch back to airmode, it starting to flyaway and throttle have no response.
So, something affecting throttle with huge value in some cases.

Could someone help me? What parameter I should change to remove or reduce throttle changing by betaflight?

I have the following changes for now:
```
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 150
set feedforward_transition = 0
set iterm_relax_type = setpoint
set iterm_relax_cutoff = 35
set transient_throttle_limit = 0
set p_pitch = 30
set p_roll = 28
set i_pitch = 90
set i_roll = 84
set d_pitch = 30
set d_min_pitch = 18
set d_roll = 28
set d_min_roll = 16
set d_min_boost_gain = 27
set d_min_advance = 0
set f_pitch = 90
set f_roll = 84
set tpa_rate = 75
set tpa_breakpoint = 1400
set tpa_mode = D
set p_yaw = 30
set i_yaw = 90
set d_yaw = 0
set f_yaw = 30
set iterm_rotation = OFF
set thrust_linear = 0
```

JurikRistjouw, 2019-04-16:
It seems i have a similar problem on an omnibus F7. 
- 3.5.7 defaults fly awesome
- 4.0.0 nightly build before RC2 flies good
- 4.0 final vibrates like mad when taking of and I allmost could not land it as it tried to stay up. Killed it when it was safe enugh.

these are the diffs
`# Betaflight / OMNIBUSF7 (OBF7) 4.0.0 Apr 11 2019 / 07:32:23 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF7
manufacturer_id 

# name
name Blackout miniH

# resources
resource SERVO 1 B11
resource SERIAL_RX 3 NONE

# timer

# dma

# mixer

# servo
servo 0 1300 1700 1500 100 7

# servo mix

# feature
feature SERVO_TILT
feature GPS
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# led
led 0 2,0:NW:C:1
led 1 1,0:NW:C:1
led 2 0,0:NW:C:1
led 3 13,0:NE:C:1
led 4 14,0:NE:C:1
led 5 15,0:NE:C:1
led 6 2,15:SW:C:6
led 7 1,15:SW:C:6
led 8 0,15:SW:C:6
led 9 13,15:ES:C:6
led 10 14,15:ES:C:6
led 11 15,15:ES:C:6
led 12 10,15:S:F:13
led 13 9,15:S:F:13
led 14 8,15:S:F:13
led 15 7,15:S:F:13
led 16 6,15:S:F:13
led 17 5,15:S:F:13

# color
color 2 120,0,255

# mode_color

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 4 1150 1275 0 0
aux 2 2 4 1350 1450 0 0
aux 3 8 2 1950 2100 0 0
aux 4 13 1 1625 2100 0 0
aux 5 15 1 1300 1625 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -310,-3,-104
set mag_declination = 1500
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = MULTISHOT
set align_board_pitch = 10
set bat_capacity = 1500
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set osd_warn_core_temp = OFF
set osd_vbat_pos = 1
set osd_rssi_pos = 2048
set osd_tim_1_pos = 2071
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2457
set osd_ah_sbar_pos = 2254
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2432
set osd_gps_speed_pos = 2262
set osd_gps_sats_pos = 2052
set osd_home_dir_pos = 2062
set osd_home_dist_pos = 2063
set osd_compass_bar_pos = 9
set osd_altitude_pos = 2240
set osd_power_pos = 2424
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2080
set osd_disarmed_pos = 2409
set osd_nheading_pos = 362
set osd_nvario_pos = 365
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set vtx_channel = 4
set vtx_freq = 5800
set vcd_video_system = NTSC

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end`

etracer65, 2019-04-17:
For those having a problem, please try the following:

1. Reset to `defaults` and do a minimal setup. In particular leave the PID and filter settings unchanged.
2. Test and confirm there's still a problem
3. Make paste the following into the CLI:
```
set d_roll = 25
set d_pitch = 27
set d_min_roll = 0
set d_min_pitch = 0
save
```
Then test again.

etracer65, 2019-04-17:
We would also like you to try some revised filtering defaults as a second test after the D gain changes above.

Again revert to `defaults` and do a minimal setup - leave PID settings and filters unchanged.

Then apply the following in the CLI:
```
# set gyro1 lowpass to 100-300 dynamic using PT1
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 300

# disable gyro2 lowpass
set gyro_lowpass2_hz = 0

# set dterm1 lowpass to 100-150 dynamic using PT1
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 150

# set dterm2 lowpass to 100hz with PT1
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100

save
```
Then test to see if the flyaway issue is resolved.

JurikRistjouw, 2019-04-17:
I will try that. will that also work with the newest nightly 4.1?

etracer65, 2019-04-17:
Yes

JurikRistjouw, 2019-04-17:
Solution 1:

```set d_roll = 25
set d_pitch = 27
set d_min_roll = 0
set d_min_pitch = 0
save```

this did solve the flyaway issue, but still vibrating like mad. Will try second solution

JurikRistjouw, 2019-04-17:
@etracer65 
Sir, you just made my day! It flies like butter with option 2.... It's like night and day difference.

```
# set gyro1 lowpass to 100-300 dynamic using PT1
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 300

# disable gyro2 lowpass
set gyro_lowpass2_hz = 0

# set dterm1 lowpass to 100-150 dynamic using PT1
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 150

# set dterm2 lowpass to 100hz with PT1
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100

```
here is my diff, and blackboxes (if it helps)

diff:
```
# Betaflight / OMNIBUSF7 (OBF7) 4.1.0 Apr 16 2019 / 19:16:40 (55a5d70cc) MSP API: 1.42

# start the command batch
batch start

board_name OMNIBUSF7
manufacturer_id 

# name
name Blackout miniH

# resources
resource SERVO 1 B11
resource SERIAL_RX 3 NONE

# timer

# dma

# mixer

# servo
servo 0 1300 1700 1500 100 7

# servo mix

# feature
feature SERVO_TILT
feature GPS
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 2 115200 115200 0 115200

# led
led 0 2,0:NW:C:1
led 1 1,0:NW:C:1
led 2 0,0:NW:C:1
led 3 13,0:NE:C:1
led 4 14,0:NE:C:1
led 5 15,0:NE:C:1
led 6 2,15:SW:C:6
led 7 1,15:SW:C:6
led 8 0,15:SW:C:6
led 9 13,15:ES:C:6
led 10 14,15:ES:C:6
led 11 15,15:ES:C:6
led 12 10,15:S:F:13
led 13 9,15:S:F:13
led 14 8,15:S:F:13
led 15 7,15:S:F:13
led 16 6,15:S:F:13
led 17 5,15:S:F:13

# color
color 2 120,0,255

# mode_color

# aux
aux 0 0 0 1600 2100 0 0
aux 1 1 4 1150 1275 0 0
aux 2 2 4 1350 1450 0 0
aux 3 8 2 1950 2100 0 0
aux 4 13 1 1625 2100 0 0
aux 5 15 1 1300 1625 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 300
set acc_calibration = -304,28,-113
set mag_declination = 1500
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = MULTISHOT
set align_board_pitch = 10
set bat_capacity = 1500
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set osd_warn_core_temp = OFF
set osd_vbat_pos = 1
set osd_rssi_pos = 2048
set osd_tim_1_pos = 2071
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2457
set osd_ah_sbar_pos = 2254
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2432
set osd_gps_speed_pos = 2262
set osd_gps_sats_pos = 2052
set osd_home_dir_pos = 2062
set osd_home_dist_pos = 2063
set osd_compass_bar_pos = 9
set osd_altitude_pos = 2240
set osd_power_pos = 2424
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2080
set osd_disarmed_pos = 2409
set osd_nheading_pos = 362
set osd_nvario_pos = 365
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set vtx_channel = 4
set vtx_freq = 5800
set vcd_video_system = NTSC
set name = Blackout miniH

# profile
profile 0

set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 150
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100
set d_pitch = 27
set d_roll = 25
set d_min_roll = 0
set d_min_pitch = 0

# rateprofile
rateprofile 0
```
[LOG00003.zip](https://github.com/betaflight/betaflight/files/3091171/LOG00003.zip)

ArcsinX, 2019-04-18:
> For those having a problem, please try the following:
>
> 1. Reset to `defaults` and do a minimal setup. In particular leave the PID and filter settings unchanged.
> 2. Test and confirm there's still a [...]

For me behaviour is the same as default values, still flyaways in airmode (sometimes immediate after takingoff, sometimes after some movements). In stab all is ok (as it was before).

Planning to try second solution.

ArcsinX, 2019-04-19:
> We would also like you to try some revised filtering defaults as a second test after the D gain changes above.
>
> Again revert to `defaults` and do a minimal setup - leave PID settings and filters [...]

It helped partly only. It flies ok on 4s 550mah 60C battery, but flyaway with 4s 650mah 75C battery.
The second battery heavier by 10% (68g vs 62g). I have several batteries of both types and with the above parameters on all 650mah batteries it tries to flyaway in airmode and on all 550mah it flies well.

With betaflight 3.5.7 it flies ok on all battery types.

Here is the video how it looks like (I was ready on switch to change mode to stab). I slowly raise throttle, immediate after takeoff it flies away and I switched to stab to stop it.

https://www.youtube.com/watch?v=Bxi8wdtUHao

Could the problem be here? src\main\flight\mixer.c
```
    if (motorMixRange > 1.0f) {
        for (int i = 0; i < motorCount; i++) {
            motorMix[i] /= motorMixRange;
        }
        // Get the maximum correction by setting offset to center when airmode enabled
        if (airmodeEnabled) {
            throttle = 0.5f;
        }
    } 
```
Is setting throttle to 50% here is usefull?

JurikRistjouw, 2019-04-19:
I must concur with @ArcsinX It did fly perfect in Angle mode... Today i flew in Acro again and it was horrible compared to default 3.5.7. If needed I will post the blackbox. But for now i saw there was a new Nightly with a fix for dual gyro. Will test that again with default settings.

JurikRistjouw, 2019-04-22:
ok, update: i watched Joshua's video: https://www.youtube.com/watch?v=wLAfYNTYPiw&t=3s
This works without any additional settings as previously posted. Smooth as butter on acro and angle.
[blackbox_and_diff.zip](https://github.com/betaflight/betaflight/files/3104584/blackbox_and_diff.zip)

Here are my diff and blackbox. enjoy.

JurikRistjouw, 2019-04-22:
IMHO, close issue

ArcsinX, 2019-04-23:
> ok, update: i watched Joshua's video: https://www.youtube.com/watch?v=wLAfYNTYPiw&t=3s
> This works without any additional settings as previously posted. Smooth as butter on acro and angle.
For me this works worse than second solution from @etracer65.
It is still trying to flyaway on 650mah battery and unstable on 550mah (sometimes it rises throttle but recovers).

JurikRistjouw, 2019-04-23:
Wow, that is interesting. for me it did wonders. tried it on an UR65 and it seems to made no difference, but flew good. You did confirm it flew good in 3.5.7 right? Is that still the case? I wonder if some hardware fault might have slipped in for you or some voodoo?

ArcsinX, 2019-04-24:
Finally I found changes with which I have no problems on both battery types on my lizard 105s. May be this could be helpfull for someone.

```
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 300
set transient_throttle_limit = 10
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 150
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 100
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0
set tpa_rate = 75
set tpa_breakpoint = 1400
```

P.S.
Anyway I think that BF could sometimes set throttle to 50% in airmode independent on user input is a dangerous bug.

JurikRistjouw, 2019-04-24:
very intersting @watageek Should we move to that thread?

schugabe, 2019-04-24:
I also have a 2S brushless whoop and did not have a great experience with 4.0 defaults values. The problem was constant slow oscillations and if it is a bit windy the quad is flying completely bad with the  default values. 
After a slight bump in the wall the quad did start to climb without any additional throttle. And I'm not sure if a fly away would have happened in other situations since I disarm the quad fast if it acts weird (I often fly inside).

During my tuning I noticed two things: I term is way to high for my quad and the filtering min settings need to be lower and the max much higher.

My current settings are listed below. It's still WIP. The dyn_lpf_gyro_max_hz  was the result of the max FFT reported in the OSD.

I don't know if something is wrong with my quad but I term of 5 seems crazy low but so far it is flying good with the settings. I have new props on and made sure that nothing is vibrating on the frame (for reference the quad is a betafpv 75 pro 2).

```
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 100
set dyn_notch_width_percent = 10
set dyn_notch_q = 100
set dyn_lpf_gyro_min_hz = 130
set dyn_lpf_gyro_max_hz = 800
set dyn_lpf_dterm_min_hz = 130
set dyn_lpf_dterm_max_hz = 270
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 100
set anti_gravity_gain = 2000
set iterm_relax_type = GYRO
set p_pitch = 35
set i_pitch = 5
set d_pitch = 45
set f_pitch = 45
set p_roll = 35
set i_roll = 5
set d_roll = 45
set f_roll = 45
set f_yaw = 20
set abs_control_gain = 2
set d_min_roll = 18
set d_min_pitch = 18
```

mikeller, 2019-04-28:
@bluehallu: This should be fixed by #8043. Can you test / confirm / close please?

closed 2019-06-04 by stale[bot]