betaflight/betaflight #7883: Brutal stop during flight

CcLaFleche, 2019-03-27:
**Describe the bug**
Brutal stop during flight, Voltage 0.0V, Consumption 0 mAh, but data are wrong. Impossible to arm. This is the third time, but it happens in a random way so no black box. Everything is back to normal after a reboot or reconnection of the battery. another person had the same problem in a comment on FB while he was testing his Quad on a fixed pushy.

Betaflight / MATEKF722 (MKF7) 4.0.0 Mar 8 2019 / 03:23:32 (37dd396) MSP API: 1.41
board_name MATEKF722
manufacturer_id
start the command batch
batch start
name
resources
resource MOTOR 4 A08
resource MOTOR 6 NONE
timer
dma
mixer
servo
servo mix
feature
feature GPS
beeper
beeper -ON_USB
beacon
beacon RX_SET
map
serial
serial 0 16384 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200
led
color
mode_color
aux
aux 0 0 1 900 1250 0 0
aux 1 1 2 900 1750 0 0
aux 2 46 2 900 1250 0 0
aux 3 13 3 1750 2100 0 0
aux 4 26 3 900 1250 0 0
aux 5 32 7 900 1250 0 0
aux 6 33 6 1750 2100 0 0
aux 7 34 7 1750 2100 0 0
aux 8 35 4 900 1250 0 0
aux 9 36 0 1750 2100 0 0
aux 10 49 4 1750 2100 0 0
adjrange
rxrange
vtx
rxfail
display_name
master
set gyro_sync_denom = 2
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = -4,-45,42
set align_mag = CW270FLIP
set mag_calibration = 41,124,245
set max_check = 2000
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set min_throttle = 1000
set dshot_burst = OFF
set dshot_bidir = ON
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 2000
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 3000
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_descent_dist = 50
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 60
set osd_cap_alarm = 3000
set osd_vbat_pos = 2444
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2519
set osd_flymode_pos = 2478
set osd_throttle_pos = 2296
set osd_vtx_channel_pos = 2488
set osd_current_pos = 2465
set osd_mah_drawn_pos = 2497
set osd_gps_speed_pos = 2273
set osd_gps_lon_pos = 2529
set osd_gps_lat_pos = 2545
set osd_gps_sats_pos = 2081
set osd_home_dir_pos = 2095
set osd_home_dist_pos = 2060
set osd_altitude_pos = 2337
set osd_warnings_pos = 2313
set osd_pit_ang_pos = 2360
set osd_battery_usage_pos = 2505
set osd_stat_max_alt = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5732
set vtx_pit_mode_freq = 5732
profile
profile 0
set dterm_lowpass2_hz = 0
set p_pitch = 65
set p_roll = 55
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35
set launch_control_mode = PITCHONLY
set launch_angle_limit = 50
rateprofile
rateprofile 0
set roll_rc_rate = 218
set pitch_rc_rate = 218
set yaw_rc_rate = 218
set roll_expo = 95
set pitch_expo = 95
set yaw_expo = 95
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

CcLaFleche, 2019-03-27:
[image]

McGiverGim, 2019-03-27:
Disable the RPM filter or update to the latest RC. Earlier versions had a problem of dshot collisions.

CcLaFleche, 2019-03-27:
Ok, i will disable rpm filter. The other person who reported the problem was under the RC3 but I have no more info on its configuration and or settings. I'm still waiting for his answer...
RC3 version also is concerned ?
Thank you for your time !

McGiverGim, 2019-03-27:
Yes, RC3 is affected. Was fixed in RC4 here https://github.com/betaflight/betaflight/pull/7865

Please test with RC4 to see if that fixes the problem.

CcLaFleche, 2019-03-27:
ok, Thank you very much !

McGiverGim, 2019-03-27:
The setting that produced the problem and is needed for RPM filter was: set scheduler_optimize_rate = ON

If your friend has this setting maybe the problem was this bug.

mikeller, 2019-03-27:
@CcLaFleche: You were told to go somewhere else if what you want is to waste peoples' time after being told what the problem of your setup is.

closed 2019-03-27 by mikeller