betaflight/betaflight #7894: Add Airbot F7 target - AIRBOTF7 (AF7)

GhostMaster9, 2019-03-29:
Hi, can you add a new target for the new airbot FC? AIRBOTF7 (AF7).

I made a DUMP from 3.5 target it has installed, maybe it helps, im dont know the schematic.

DUMP here: https://ufile.io/defdr

jflyper, 2019-03-29:
@GhostMaster9 

I believe Airbot F7 is still under development and not a final product.
Support for the board will be added when the board is released to the public.

How did you get one of those?

GhostMaster9, 2019-03-29:
> @GhostMaster9
> 
> I believe Airbot F7 is still under development and not a final product.
> Support for the board will be added when the board is released to the public.
> 
> How did you get one of those?

Im testing it, but i wanted to use BF4.0, how can i use the generic target for F722 boards with it?

mikeller, 2019-03-29:
@GhostMaster9: Ask the manufacturer to provide you with a configuration for it. Without access to the board's designer's plan for resource allocation working out a working configuration will take hours of guesswork and testing (which requires access to a board).

Asizon, 2019-03-30:
Hi Rui!welcome,Im going to try to compile 4.0 for you,but i dont have this Fc for the moment i cant test it

SebiWolze, 2019-04-04:
Hey, got the same FC, need that target. Can I help you somehow?
T

Asizon, 2019-04-05:
Hi @SebiWolze , I have created 4.0 .hex for you test, this is already tested by @GhostMaster9  with good results. Now you can get unified target config and past it in any version of STM32F722 Target.

[betaflight_4.0.0_AIRBOTF7.zip](https://github.com/betaflight/betaflight/files/3046790/betaflight_4.0.0_AIRBOTF7.zip)

SebiWolze, 2019-04-05:
Hey @Asizon - sucessfully flashed, looks good, but I don´t know if thats correct but in the setup Page: 
When I tilt the FC in right direction to the front, it tilts left in betaflight, if I tilt it to the back, it tilts  right in Betaflight..  
Gyro Issue with that target?

Asizon, 2019-04-05:
Yes because its not a official target, it is only for flight controller test purposes :)

GhostMaster9, 2019-04-05:
> Hey @Asizon - sucessfully flashed, looks good, but I don´t know if thats correct but in the setup Page:
> When I tilt the FC in right direction to the front, it tilts left in betaflight, if I tilt it to the back, it tilts right in Betaflight..
> Gyro Issue with that target?

I rotated in betaflight, it's perfectly fine... it's 10 seconds fix.

SebiWolze, 2019-04-05:
> I rotated in betaflight, it's perfectly fine... it's 10 seconds fix.

Never done before, how to do that?

joelucid, 2019-04-17:
I have this board and I'm trying to run BF 4.0 on it. @asizon's hex does install but the flash chip isn't recognized. Anybody know what to do?

Asizon, 2019-04-18:
@joelucid I have knowledge of that problem, I do not know exactly what should be going to check the code. Anyway I think that in a few days they will raise PR to add target Airbot F7 oficial

Asizon, 2019-04-20:
@sebiwolze, about?for the moment you can test 4.0 without Flash and. Blackbox,all working

Asizon, 2019-04-23:
> Never done before, how to do that?

@SebiWolze are you a brand tester?you dont know how to rotate a Fc or Gyro on Bteaflight?

SebiWolze, 2019-05-02:
No, I am not, I know how to change that direction.
Is there any snippet for RPM Filters now, did you try that?

jflyper, 2019-05-02:
@SebiWolze How did you get the board?

SebiWolze, 2019-05-02:
Airbot Sponsoring..

Asizon, 2019-05-07:
Airbot F7 target availabe in 4.1 abd 4.0.3 when it merged
https://github.com/betaflight/betaflight/pull/8208

Asizon, 2019-05-07:
This issue can be close @ghostmaster9

closed 2019-05-07 by GhostMaster9