betaflight/betaflight #7897: Motors start spinning in CLI

Jensi, 2019-03-30:
**Describe the bug**
Whyle copying in some "code" into the cli the motors startet spinning up to idle and stoped ~2sec later.

**To Reproduce**
I flashed my FC with BF 4.0.0 RC4
then i copied the CLI part for freestyle out of the "4.0 Tuning Notes" (https://github.com/betaflight/betaflight/wiki/4.0-Tuning-Notes) into the CLI and typed Enter.
While the "code" was executet the motors started spinning with ~idle throttle.

this (motors spinning up) also occurred while copying the "dump all" of my first FC (after setting everything up) to my second quad
both are fresh builds and exactly the same.

**Expected behavior**
I don't want spinning motors on my bench if i not force it ^^
I had the props off so everything was ok but someone will not...

**Flight controller configuration**
FC: Matek F722 SE
BF: 4.0.0 RC4
Rx: crossfire on RX2/TX2
VTx: smartaudio on TX3
Camera: RC micro swift 3 on RX4/TX4
ESC: Aikon 4in1 with Telemetry on RX6

mikeller, 2019-03-30:
@Jensi:
- **Do not connect a battery when configuring your flight controller!** When resources are reconfigured this can have unintended consequences, including spinning / burnt out motors;
- do not use `dump all` for backup / restore. This will result in a not working or poorly performing configuration.

bizmar, 2019-04-02:
Were you using pre-release BLheli32 firmware?

Jensi, 2019-04-02:
@mikeller I know the risk and the receiver is only working with plugged in battery so that is why it was plugged in.

@bizmar no I used the latest non pre. 32.6? I believe (.6 it is I think but not sure about the 32...)

mikeller, 2019-04-03:
@Jensi: So if you know that motors may spin up when configuring your flight controller in CLI with a battery plugged in, then you should not be reporting this in the first place. Closing this.

closed 2019-04-03 by mikeller

Jensi, 2019-04-03:
@mikeller i know that the CLI is "dangerous" and bad thinks can happen.
That doesn't mean that they shoud.
especially if in the BF4 Tuning Notes is a part of "Code" that lets the motors spin up... and these 4.0 Tuning Notes are promoted on the front page of BF
I was save, because i know the risk.
One other might not!
That is why i reported it.
It's something different if someone is typing random things into the CLI or a official post like: "copy this into your CLI and your copter will fly amazing" 
(witch it does! So thanks to everyone who is working on BF :D )

mikeller, 2019-04-03:
@Jensi:

> That doesn't mean that they shoud.

The point here is not whether configuration (and in particular resource) changes performed in CLI _should_ cause undesired motor output signals that can cause motors to spin up if a battery is connected or if they _should not_.
The point here is that they _can_, and there is absolutely no way to avoid this.
This is why pasting snippets into CLI should *never* be done with a battery connected.
Besides posting even more disclaimers in even more places there is nothing that can be done to ameliorate this situation.

jorgemiarnau, 2019-08-27:
Happened to me as @Jensi predicted :/ wasn't aware you had to disconnect battery (and unfortunately had my props on this time and cut my finger). Didn't see the battery warning at all, maybe it should be implemented in a similar way to when you spin up the motors in betaflight?