betaflight/betaflight #7902: BF4.0.0 RC4 OMNIBUSF4SD DSMX Receiver at UART1 not working

TheFrisco, 2019-03-31:
When running BF 4.0 RC4 on OMNIBUSF4SD, my DSMX receiver on UART1 isn't working. Did work in BF 3.5.5. Anything special I need to do because its 4.0 or RC?

Thanks Robert

Here is my DIFF:
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Mar 22 2019 / 01:31:49 (1bdbd1cd7) MSP API: 1.41

board_name OMNIBUSF4SD
manufacturer_id 

# name
name TheFrisco

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1700 0 0
aux 2 13 1 900 1300 0 0
aux 3 35 1 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT600
set align_board_yaw = -45
set bat_capacity = 450
set vbat_max_cell_voltage = 450
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 310
set yaw_motors_reversed = ON
set small_angle = 180
set osd_vbat_pos = 2465
set osd_tim_2_pos = 2487
set osd_flymode_pos = 2349
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2090
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF

# profile
profile 0

# rateprofile
rateprofile 0

set thr_expo = 40
set roll_rc_rate = 180
set pitch_rc_rate = 180
set yaw_rc_rate = 200
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 20
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65

mikeller, 2019-03-31:
@TheFrisco: Is this happening with a genuine OMNIBUSF4SD or with a clone?

TheFrisco, 2019-03-31:
> @TheFrisco: Is this happening with a genuine OMNIBUSF4SD or with a clone?

It's a Super_S F4 stack. Does that answer your question?

mikeller, 2019-03-31:
Yes. Looks like it's a problem of an incrrrectly designed clone then. See #7849.

Duplicate of #7849.

closed 2019-03-31 by mikeller

TheFrisco, 2019-03-31:
How can this be a design fault by the chip when it works in 3.5.7? Feels like a copout to close this issue without a resolution!

watageek, 2019-03-31:
@mikeller
You may redirect them to this link.

https://www.rcgroups.com/forums/showthread.php?3271171-Betaflight-4-0-Omnibus-F4-Pro-V2-%28gearbest%29-Serial_RX-DSM2-DSMX-not-working-on-UART1

Hope it solve the issue.

Best regards

mikeller, 2019-03-31:
@TheFrisco:

[image]