betaflight/betaflight #7903: LED not working on BETA4 RC4 - OMNIBUS F4SD

Barux, 2019-03-31:
**Describe the bug**
LED not lighting up, they remains switched off (they re working with older Dev. Fw) 

**To Reproduce**
Flash BETA4 RC4 on omnibus F4SD (not tried on other targets) 

**Expected behavior**
Led control with knob as other betaflight versions

**Setup / Versions**
 - Flight controller [Hobbywing G2 OMNIBUS F4SD ];
 - Other components [Matek leds];

mikeller, 2019-03-31:
@Barux: Unless you supply the information that you are asked to provide when opening an issue nobody will be able to help you.

jflyper, 2019-04-02:
LED strip works without an issue on genuine Omnibus F4 V3 (target OMNIBUSF4SD) flashed with 4.0-RC4 and immediately configured for LED strip operation.

I think this is an isolated case with Hobbywing G2 board and/or configuration problem. Should be solved by hardware supplier.

Closing.

closed 2019-04-02 by jflyper