betaflight/betaflight #7904: ibus telemetry 2 wire

tSaM89, 2019-03-31:
Hi I am using the betaflight f4 fc and the flit10 ibus receiver. There is a small pcb included to combine the two wire communication (receiver) into a one wire communication (FC). In betaflight I can only set up a one wire communication (Telemetry Output iBUS) which requires the pcb with a resistor and a diode to connect the TX and RX of the receiver. I think it would be a lot easier to use if we could simply use the RX and TX channel of a uart port on the FC. This way there would be no need to add electrical components. I didn’t check the source code, but I think it would be a very small software change.

What do you think?

See IBUS section:
https://github.com/betaflight/betaflight/blob/master/docs/Rx.md

FLIT10:
https://www.banggood.com/de/Flit10-2_4G-10CH-Micro-Telemetry-Flysky-Compatible-Ibus-Receiver-for-FS-I6X-FS-i6S-Turnigy-Evolution-p-1260238.html?cur_warehouse=CN

mikeller, 2019-03-31:
@tSaM89: Not really sure what you are trying to do, but single wire should already work for iBus - just connect the single wire to (uninverted) TX on the flight controller, and set `serialrx_halfduplex = on`.

tSaM89, 2019-03-31:
Sorry, maybe I did not explain my problem good enough. I want to use the ibus telemetry option. This way I could transmit the battery voltage back to my radio control. My problem is that my flit 10 module is using a two wire interface. Their solution is a second pcb to turn my fc one write to the two write communication on the flit 10 module. I would like to get rid off the second pcb by using also TX and RX on my fc. I think it should work with a two wire communication, or did I miss something?

mikeller, 2019-03-31:
@tSaM89: So you want to use iBus to control the craft, **and** use telemetry to get data back to the TX? This should work just fine with what I described above. If you want to get rid of the second PCB you can do so as well. Just connect the telemetry wire to TX of an UART, and connect the RX wire to RX **on a different UART**. Why you would want to do so and use two UARTs for what can be done with one by using the second PCB is not clear to me though.

etracer65, 2019-03-31:
It just calls for a resistor and a diode - hardly another "pcb".

mikeller, 2019-03-31:
@etracer65: Potentially. I could not find any documentation on what the little PCB that comes with the RX actually does, did you?
The thing is though that since iBus telemetry already is bidirectional single wire, connecting RC and telemetry to the flight controller over the separate 'RC' and 'telemetry' wires that come out of the RX will require two UARTs (or for the signal to be combined on a single wire and then split again, for no good reason).

etracer65, 2019-03-31:
@mikeller https://github.com/betaflight/betaflight/blob/master/docs/Rx.md#combine-flysky-ibus-telemetry-and-serial-rx-on-the-same-fc-serial-port

mikeller, 2019-04-01:
@etracer65: Heh, good point - stands to reason that the 'flit10' RX will work the same.

McGiverGim, 2019-04-01:
Exactly. This is not the first time some users asks for that. I don't know which is the problem with the tiny extra board, maybe use less space:

https://github.com/betaflight/betaflight/issues/7164

tSaM89, 2019-04-01:
Using two uart is fine for me. I will try it later.

tSaM89, 2019-04-01:
I could not get it work. Even only the 'sensor' wire connected to one TX of the FC did not work.

closed 2019-04-01 by tSaM89

tSaM89, 2019-04-06:
Update:

I installed the "mini pcb" and it did not work. Later I tried to use TX3 instead of TX2 and it worked. I do not know where the difference is on my FC (Betaflight f4).  I think without the extra pcb it would also work in full-duplex.

mikeller, 2019-04-06:
@tSaM89: Looking at https://www.fpvmodel.com/u/tmp/goods_desc/2017-09/20170929175657-6325.jpg, it seems that TX3 is inverted (as it shows it connected to SmartPort on a FrSky XSR). This suggests that your signal is inverted, which would be unusual for FlySky hardware. Or the BETAFLIGHTF4 schematic from the supplier is simply wrong, and TX2 is inverted...