betaflight/betaflight #7908: Betaflight 4.0 RC4 & DalRC F722

Strangedave, 2019-04-01:
I just had a pretty spectacular freak out. This was the second flight on a rebuild of parts from a previous build being installed into a new frame. I did a 40 second test hover, everything was good. downloaded the log. then turned off the Gyro Lowpass 2 Filter, which I have done on this setup previously in a different frame. The second the ESC booted up the number 2 motor went full throttle and pulled the rig outta my hand and continued to spin/bounce around until the prop exploded. I did try plugging the rig in again, the problem did not repeat after 3 attempts. There is no blackbox log either since the quad was not armed.

Steps to reproduce the behavior: Behavior could not be reproduced.

The flight controller is set for two stage arming, so it was definitely not armed.

**Flight controller configuration**
```# diff

# version
# Betaflight / DALRCF722DUAL (DLF7) 4.0.0 Mar 26 2019 / 19:12:28 (173e958da) MSP API: 1.41

# start the command batch
batch start

board_name DALRCF722DUAL
manufacturer_id 

# name
name FLOSS 3.0 R

# resources

# timer

# dma
dma pin C06 1
# pin C06: DMA2 Stream 2 Channel 7
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 4 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1125 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 31 3 1900 2100 0 0
aux 5 35 2 1700 2100 0 0
aux 6 36 6 1700 2100 0 0
aux 7 49 3 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -71,-33,-85
set mag_hardware = NONE
set baro_bustype = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_idle_value = 450
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = KILL
set ibata_scale = 139
set ibata_offset = -500
set yaw_motors_reversed = ON
set small_angle = 40
set pid_process_denom = 1
set transient_throttle_limit = 10
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2081
set osd_rssi_pos = 2093
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2414
set osd_crosshairs_pos = 2286
set osd_current_pos = 2086
set osd_mah_drawn_pos = 2097
set osd_craft_name_pos = 2381
set osd_gps_speed_pos = 2401
set osd_gps_sats_pos = 2425
set osd_warnings_pos = 2346
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 5
###ERROR: CORRUPTED CONFIG: vtx_power = 5###
set vtx_freq = 5880
set vcd_video_system = NTSC

# profile
profile 0

set dyn_lpf_dterm_min_hz = 200
set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set feedforward_transition = 5
set iterm_relax_cutoff = 35
set p_pitch = 35
set i_pitch = 90
set d_pitch = 37
set p_roll = 34
set i_roll = 80
set d_roll = 36
set p_yaw = 24
set i_yaw = 110
set d_yaw = 2
set abs_control_gain = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35
set launch_trigger_allow_reset = OFF
set launch_angle_limit = 35

# rateprofile
rateprofile 0

set roll_srate = 71
set pitch_srate = 71
set yaw_srate = 69
set tpa_breakpoint = 1750

# end the command batch
batch end```

**Setup / Versions**

Floss   3.0
--
TMotor F-40 Pro v2 2400kv
Runcam microeagle
Crossfire Nano
Tramp HV
Lumenier AXII   LHCP
DALRC Rocket   50A 4-in-1 BLHeli32 firmware 32.6.8
HQProp DP 5x4.3x3 PC V1S
Dalrc F722 
Buzzer from   bangood
Microphone from bangood
battery 4s   100c 1300mah

bizmar, 2019-04-01:
this sounds to me like this Blheli issue https://github.com/bitdump/BLHeli/issues/353

bizmar, 2019-04-01:
This sounds to me like this BLheli32 issue [bitdump/BLHeli#353](https://github.com/bitdump/BLHeli/issues/353)

Strangedave, 2019-04-01:
Certainly looks like the same problem. Ill update to that FW. and report back if it happens again.