betaflight/betaflight #7909: Request to Add "KLEE" as board manufacturer for unified targets

LexioTech, 2019-04-01:
Hi there,

I'd like to request a manufacturer ID for the new unified targets with the following specification

ID: "KLEE"
Name: "Kevin Lee"
URL: "https://whitenoisefpv.com/"

manufacturer ID raw for Manufacturers.md: 
|KLEE|Kevin Lee|https://whitenoisefpv.com/|

He is the manufacturer of the SYNERGYF4 flight controller (added to the maintenance branch 3.5.x and released with 3.5.7) and since I am doing the 4.0 and unified target_config update, I was tasked to have the actual manufacturer ID listed instead of "CUST".

Since the deadline to push the updated target pull request for the SYNERGYF4 into master is close, please reply to this at your earliest convenience!

Thanks in advance!

LexioTech, 2019-04-01:
Just some additional info: The released maintenance target has the same manufacturer ID, so it would be awesome if we could keep it for consistency between versions! Thanks!

mikeller, 2019-04-01:
@LexioTech: https://whitenoisefpv.com/ does not seem to show any affiliation with / ownership by a 'Kevin Lee'. I think this will be confusing to users. Is there a better link to products manufactured by 'Kevin Lee'?

LexioTech, 2019-04-01:
@mikeller Literally every product in his store is manufactured by him (Kevin Lee) aka WhitenoiseFPV, his name (yes only Kevin) and email is in the "About us" section. I don't know it helps at all but he has a YouTube channel where the assembly of some of his products/prototypes are available: https://www.youtube.com/user/KSLee106

Other than that, I think this is the most logical website to link, his own store where the people can actually buy his creations. I don't know what else could we link there.

mikeller, 2019-04-01:
How about this:

|KLEE|Kevin Lee (WhitenoiseFPV)|https://whitenoisefpv.com/|

LexioTech, 2019-04-02:
@mikeller Okay, thats a good idea :)

PR #7912 (2019-04-03): Added 'KLEE' / Kevin Lee (WhitenoiseFPV) manufacturer id.
Fixes #7909.
closed 2019-04-03 by mikeller