betaflight/betaflight #7920: Climb rate efficiency indicator in OSD

NIHERASE, 2019-04-02:
**Is your feature request related to a problem? Please describe.**
Climb efficiency depends on a lot of factors like motor KV, prop, throttle position, air pressure, wind speed (to fight against) and etc (please correct me if I'm wrong).

I'd like to have fast feedback on my current climb efficiency so I can find the best throttle value and angle for climbing by trying to maximize the indicator value. 

Perhaps this could be used to compare props by comparing best efficiency indicator value achieved.
Perhaps mountain divers would find this feature useful. 

**Describe the solution you'd like**
I'd like a simple indicator in OSD that is calculated as `current vertical speed (cm/s)` / `power consumption (W)`.

**Describe alternatives you've considered**
Same thing could be implemented in radio using telemetry. It would have to say the indicator value and so it would be much slower than reading a value from OSD. I think that slow response would make this feedback loop much fiddlier. 

Would you guys be interested in such feature? Could it be solved differently?

mikeller, 2019-04-03:
Sounds reasonable. One caveat: Most ways to establish a reliable altitude (and change thereof) seem to be beheft with issues, so getting a reliable value for `current vertical speed` won't be easy. Any suggestion for how to do this?

Another thought: How about 'maximum climb rate' / 'maximum climb throttle value' statistics elements (displayed after disarming)? This would work as a way around having to spot the peak values in the live display.

NIHERASE, 2019-04-03:
Ha, I did not expect that calculating vertical speed would be difficult. Honestly I just though that numerical vario could be used to do this.

I think it's possible to plot a graph of efficiency/throttle values using blackbox data. I'll try doing that and see if it makes any sense.

`maximum climb rate` and `maximum climb throttle value` would be really nice to have but I'm not sure how to implement them properly. I expect that a naive implementation would not show realistic values, for example after a punch the throttle is lowered but the aircraft is still gaining altitude due to it's inertia, during this time the indicator would incorrectly show great efficiency values.

mikeller, 2019-04-03:
@NIHERASE: Numerical vario is often done based on the output of the baro sensor. These are very susceptible to things like vibrations or air streaming around them, and can often be thrown off by tens of meters, in particular during punch outs. So I would suspect that any momentary calculation of climb rate during that time will probably get a lot of error from this.

The idea of using blackbox data for this sounds good - this would allow for analysis of the overall data, and suppression of parts of the log where the correlation of baro data with other values makes it likely for the baro to be influenced by outside factors.

closed 2026-09-26 by blckmn