betaflight/betaflight #7922: BF4 RC5, OminbusF4V6, remapped motors, 2 motors not working when dshot_bidi = ON

wunderlins, 2019-04-03:
**Describe the bug**
I have a 4in1 Typhoon32 ESC, it is 90° CW roated, therefore I have to remap the motor resources. When I test the motors with dshot_bidi = OFF in the motor tab, they work as expected. If I turn dshot_bidi = ON, motor 2 and 3 stop working.

**To Reproduce**
Original (as of OMINBUSF4V6):

```resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 B05```

My Configuration:

```resource MOTOR 1 None
resource MOTOR 2 None
resource MOTOR 3 None
resource MOTOR 4 None

resource MOTOR 1 B01
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 A03```

**Expected behavior**
Motor 2 and 3 should turn like 1 and 4 (otherwise it's  bit an unfair work distribution ;)

**Flight controller configuration**
```# diff

# version
# Betaflight / OMNIBUSF4V6 (OBV6) 4.0.0 Apr  1 2019 / 11:36:02 (88665a678) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4V6
manufacturer_id 

# name
name Rooster 2

# resources
resource MOTOR 1 None
resource MOTOR 2 None
resource MOTOR 3 None
resource MOTOR 4 None
resource MOTOR 1 B01
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 A03

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY

# beeper
beeper -SYSTEM_INIT

# beacon

# map

# serial
serial 1 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 925 1200 0 0
aux 1 1 0 900 1200 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 2 1350 1650 0 0
aux 4 35 1 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1300
set pid_process_denom = 1
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set thr_mid = 30
set thr_expo = 20
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 81
set tpa_rate = 15```

**Setup / Versions**
 - Flight controller Ominbus F4 V6, Betaflight V 4.0.0 RC5 (OminbusF4V6)
 - Typhoon32 4in1 ESC with BLHeli_32 v. 32.6 - Multi installed

Asizon, 2019-04-03:
Hi,the motor resources are personal,depend of what esc are you using. Wich is exactly your problem?trying to test Dshot bidir and rpm filter?

etracer65, 2019-04-03:
Did you do `set dshot_burst=off` as required? Also the version of BLHeli_32 firmware you list doesn't support bidirectional telemetry. You need to use a beta version with the appropriate support.

Sounds like you didn't read and follow the instructions carefully. Please start here:
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter