betaflight/betaflight #7925: SPRacing F3 Disarming when moves sticks too fast.

henriquewatanabe, 2019-04-04:
When the multicopter is armed and take off to flight and move the sticks of aileron or elevator too fast or agreesive position, and the drone starts move, but in this time the board disarme and you have to turn off the switch and re-arm again.

1 - Arm the quadcopter.
2 - Take off.
3 - Move the sticks of aileron or elevator too fast.
4 - when the drone calculates the angular motion he desarm.
5 - It only occurs when the gyro calculate the angular motion. If you move the sticks of controller
and hold the drone it looks fine.

List of parts:
SP Racing F3 Deluxe. (FC)
Firmware - Betaflight 3.5.7
ESC - Sky RC 12a
Receiver - FlySky i6 mode PPM
Motors: 2300kv GOOLRC

Standard settings. Loop time and Gyro

Observation. If the Super rate is 0.70 up. Cause this issue.

McGiverGim, 2019-04-05:
When opening an issue, please read the instructions and give the information that appears in the text, like a `diff` for example.

About your problem, maybe is the runaway takeoff prevention, more info here: https://github.com/betaflight/betaflight/pull/4935

Try to deactivate it to see if the problem is gone, and then, by safety, enable it again and try to adjust some parameters, maybe the `runaway_takeoff_deactivate_throttle_percent`.

etracer65, 2019-04-05:
Make sure to fly a little before trying hard moves. And as noted, check the settings and read about the behavior on the wiki page.

henriquewatanabe, 2019-04-08:
set runaway_takeoff_prevention = OFF
WORKS!!!! 
Thanks you so much.
Problem Solved!!

etracer65, 2019-04-08:
No, that's not the solution. It just confirms that you were inadvertantly triggering the protection. Instead read about the feature and configure it properly then reenable the safety feature. Without it enabled you will have no protection for possible dangerous runaway events that can be caused by incorrect settings, or even props installed incorrectly. The key point is that you need to fly for a bit immediately after takeoff to deactivate the feature. Also make sure the `runaway_takeoff_deactivate_throttle_percent` is appropriate for your quad. It should be set to the approximate hover throttle percentage for your quad.

Read the documentation here: https://github.com/betaflight/betaflight/wiki/Runaway-Takeoff-Prevention

Testmangh, 2019-04-08:
> Without it enabled you will have no protection for possible dangerous runaway events that can be caused by incorrect settings, or even props installed incorrectly.

That saved me one time, I forgot to remap the motors (FC 180°) was doing a simple test with the drone armed (little roll and pitch) and it went crazy =D but it disarmed itself really (almost immediatly) fast preventing any damage to myself or my hause...

henriquewatanabe, 2019-04-08:
> No, that's not the solution. It just confirms that you were inadvertantly triggering the protection. Instead read about the feature and configure it properly then reenable the safety feature. [...]

Yes, I will do that, I understood what you explained.

henriquewatanabe, 2019-04-08:
Humm I have a question, why some CLI commands cannot be put in graphical part?

mikeller, 2019-04-09:
@henriquewatanabe:

> Humm I have a question, why some CLI commands cannot be put in graphical part?

That depends - some of the parameters are not meant to be changed for 'normal' usage, they are meant for testing / debugging only, and will result in bad flight performance / dangerous behaviour if they are changed by users not knowing what exactly they do.
For other parameters it is simply a matter of not having enough volunteers for work in configurator to be able to add support for all parameters, so we have to be content with only getting the most important ones added.

Testmangh, 2019-04-09:
@mikeller I want to start contributing but I don't even know how to start... I want to add a motor reamaping tool on the Motors tab and a VTX tab... I'm going to start reading about chrome apps today.

Do you recommend any articles or something else for me to read?

mikeller, 2019-04-09:
@Testmangh: Thanks, welcome on board! The current configurator is a 'dual home' app really. While we still retain compatibility with Chrome Web apps (as many users are still using configurator in this way), we also build configurator as an NW.js standalone app using gulp, so NW.js and gulp are helpful topics to read up on as well.

closed 2019-05-16 by stale[bot]