betaflight/betaflight #7926: GoPro remote control integration

ValkA, 2019-04-05:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide information about your flightcontroller and other components including how they are connected/wired;
4. Add the used configuration and firmware version;
5. Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/); and
6. Remove this Text :).

closed 2019-04-05 by mikeller

ValkA, 2019-04-05:
@mikeller, I wrote the Idea from my phone and by mistake it removed the text before posting, hope you would open it again after the edit :)

mikeller, 2019-04-05:
Probably a good idea to not use your phone to write up requests for those grandiose ideas. ;-)

From the look of it, http://euerdesign.de/2015/12/28/ultralowcost-diy-gopro-remote only supports a button. You can just use a GPIO with pinio to control this - this already exists in Betaflight: https://github.com/betaflight/betaflight/wiki/Pinio-and-PinioBox

ValkA, 2019-04-05:
Very nice! Looks like its easier than I thought it would be. So can I just define a new box name and configurator will see it? What if I want to use one GPIO pin for various commands (~4)? Is that possible using a PinioBox?

mikeller, 2019-04-06:
@ValkA: Not sure you you'd want to control more than one command with a GPIO pin that can either be on or off, or how you'd want to submit more than one command on a device that only has one push button for input?

If you want to write your own firmware for the ESP8266, you can just implement the serial RunCam device protocol defined in https://github.com/betaflight/betaflight/blob/master/src/main/io/rcdevice.c, and it will work with Betaflight out of the box.

ValkA, 2019-04-07:
Indeed using a serial protocol instead of a button was the idea. 
Will look into RunCam protocol which will save a lot of hassle, thanks!

theely, 2019-11-10:
What about extending this ida/feature a bit?
Looks like GoPro can also be interfaced via bluetooth allowing to read the camera status and control it (one example: https://github.com/KonradIT/gopro-ble-py) 

Would it be possible to add a BF peripheral (e.g. GoPro control) that using a bluetooth to UART card would allow to read the GoPro status and display it on the OSD (e.g. Rec sing while recording) and also control the GoPro using configurable modes?

ValkA, 2019-11-10:
Its possible, unfortunately I haven't found time yet to implement the first idea but you can acieve that by writing a firmware that mimics runcam protocol for a small BLE device, for example - https://youtu.be/-1SO9DQjnQE

mikeller, 2019-11-10:
@ValkA: Have you tested this on an actual setup? Since Bluetooth is in the same frequency range (5.8 GHz) that VTX normally uses, you might encounter interference issues - typically boards with integrated Bluetooth module (like the SPEEDYBEEF4) are designed in such a way that the power is cut for the Bluetooth module as soon as the craft is armed.

ValkA, 2019-11-11:
That would definately need to be tested, but BLE shouldn't transmit continuously, its very efficient compared to classic bluetooth so I think if there will be any interferences, they'll be short pulses when sending gopro commands

michael-blue2, 2020-04-25:
Anybody testing this? There are some borda now coming with bluetooth. I have some arduino bluetooth modules on the way to test the control how good it is and reliable. And only enable it when pressing a button so only that time will kind of mess up with the signal