betaflight/betaflight #7928: Motors not working on RC PikoF4 OSD

Rex700, 2019-04-06:
**Describe the bug**
Motor not working on upgrading from 3.5.x (all working) and not not running on 4.0 RC1-6 (blheli suite32 detect all esc and can modify or update)

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - FuriousFPV Piko F4 OSD
- Aikon AK32Pin
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Rex700, 2019-04-07:
Updated diff

mikeller, 2019-04-07:
@Rex700:

> resource MOTOR 1 B00
> resource MOTOR 4 A03

Try resetting to defaults, and reapplying your configuration **without these two lines*.

Rex700, 2019-04-07:
Reflashed RC6 and only setted up to Dshot600 and not working

# diff

# version
# Betaflight / FF_PIKOF4OSD (PKOS) 4.0.0 Apr  3 2019 / 13:55:47 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

board_name FF_PIKOF4OSD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set motor_pwm_protocol = DSHOT600

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

mikeller, 2019-04-07:
What does `resource show all` output?

etracer65, 2019-04-07:
@Rex700 Define "not working". Have you tried going to the Motors tab and tried spinning each motor? Because obviously with the minimal setup of only setting the motors protocol you're not going to be able to arm with your radio.

Also, do you have your motors connected in any way other than the default for the board.? I see you had motor resource commands in the first `diff` - why was that?

Rex700, 2019-04-07:
@etracer65 yes i tried in motors tab nothing happens , the esc doesnt make all the beeps on the startup detecting dshot multishot or whatever
the remapping weas because i had motor one in 4 and 4 in 1
now soldered correctly and same ploblem

for example now i have uploaded the development #1460 of the 4.0 and all runs perfect

Rex700, 2019-04-07:
tested again in RC6 and the motors work on Multishot but not on Dshot

etracer65, 2019-04-07:
Well if you can determine that some older builds work, it would be really helpful to narrow down which build starts showing the problem. We don't have that exact hardware setup to test and it doesn't seem to be happening with other boards/ESCs.

The simplest was is to do a bisect where you start with known good (1460) build and a known bad (current?) and then pick one in the middle to test. If it works then that becomes the new lower limit. If it doesn't then it's the new upper limit. Then pick the one in the middle of new limits and repeat. Using this method it should take at most about 6 tests to narrow down the build. We need to find the build where the problem starts (ie. it's fine the build before).

Rex700, 2019-04-07:
In 1470 only working 2 motors 
So the problem is between 1460 and 1470 
Will focus later wich is the problematic

Rex700, 2019-04-07:
Found the last working  build 1468 and in 1469 is not working properly (only 2 motors run) and in RC's any motor isnt working

etracer65, 2019-04-07:
@Rex700 Can you provide the requested `resource show all` when using RC6?

Rex700, 2019-04-08:
**Resource of 1468:**
### # resource
resource BEEPER 1 A14
resource MOTOR 1 A03
resource MOTOR 2 B01
resource MOTOR 3 A02
resource MOTOR 4 B00
resource MOTOR 5 B14
resource MOTOR 6 B15
resource LED_STRIP 1 B07
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 3 C03
resource LED 1 B05
resource LED 2 B04
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 A03
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource OSD_CS 1 A04
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A15

**Resource on RC6**

# resource
resource BEEPER 1 A14
resource MOTOR 1 A03
resource MOTOR 2 B01
resource MOTOR 3 A02
resource MOTOR 4 B00
resource MOTOR 5 B14
resource MOTOR 6 B15
resource LED_STRIP 1 B07
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 3 C03
resource LED 1 B05
resource LED 2 B04
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 A03
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource OSD_CS 1 A04
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A15

mikeller, 2019-04-08:
@Rex700: That's not the output of typing `resource show all` and [Enter] in CLI. Please try again.

Rex700, 2019-04-08:
RC6:

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 3
A03: MOTOR 1
A04: OSD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: BEEPER
A15: GYRO_CS 1
B00: MOTOR 4
B01: MOTOR 2
B02: FREE
B03: FREE
B04: LED 2
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: INVERTER 3
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#1468: not available

mikeller, 2019-04-09:
@Rex700: The fact that your DMA allocation does not show any usage for motors suggests that you did the `resource show all` with an unconfigured flight controller. For it to actually allow any diagnostics, please repeat it with the configuration that is not working for you.

mikeller, 2019-04-09:
@Rex700: This is super weird. Motors 1 and 3 should show up on DMA1 Stream 7 and DMA1 Stream 6 respectively.

Can you also do `timer a3`, timer a3 list`, `timer a2`, and `timer a2 list` and copy the output?

Rex700, 2019-04-11:
@mikeller will upload later
Sorry for delay😞

mikeller, 2019-04-11:
@Rex700: No problem.

Rex700, 2019-04-11:
4.0 Official Release also not working

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 3
A03: MOTOR 1
A04: OSD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: BEEPER
A15: GYRO_CS 1
B00: MOTOR 4
B01: MOTOR 2
B02: FREE
B03: FREE
B04: LED 2
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: SERIAL_RX 3
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: INVERTER 3
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: MOTOR 2
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# timer a3
timer A03 1

# timer a3 list
# 0: TIM2 CH4
# 1: TIM5 CH4
# 2: TIM9 CH2

# timer a2
timer A02 1

# timer a2 list
# 0: TIM2 CH3
# 1: TIM5 CH3
# 2: TIM9 CH1

mikeller, 2019-04-11:
@Rex700: Thanks for that. Unfortunately this does not help with solving this, as everything seems to be assigned as it was specified to be - I do not know why timer 5 isn't working here. I also do not have the hardware to test, so a final resolution will probably have to wait for @MJ666.

As a workaround, can you try if all motors work when you set `dshot_burst = on`, or `timer a3 0` and `timer a2 0`, or both of these combined?

MJ666, 2019-04-11:
Mhhh, also don't have the hardware. Will ping FFPV to check.

mikeller, 2019-04-11:
@MJ666: Thanks. In general, FF hardware seems to be scarce with developers. Can we ask them to provide us with some? After all we are the first contact point for a lot of customers with issues.

Rex700, 2019-04-15:
At least please can you upload the 1468 to somewhere? Didnt found on the configurator anymore😔

Asizon, 2019-04-15:
@rex700
https://ci.betaflight.tech/job/Betaflight/1468/artifact/obj/

Rex700, 2019-04-15:
@asizon thanks!

MJ666, 2019-04-18:
@Rex700 can

 you give this build a try:

[betaflight_4.1.0_FF_PIKOF4OSD.hex.txt](https://github.com/betaflight/betaflight/files/3095828/betaflight_4.1.0_FF_PIKOF4OSD.hex.txt)

@mikeller FFPV agreed to send a set of boards for firmware testing. Are you ok the send them to me so i can continoue maintaining and testing there firmware?

Rex700, 2019-04-18:
@mj666 thanks i will try ASAP

MJ666, 2019-04-19:
@Rex700 this was tested at FFPV. The former version i send had motor 2 and 4 swaped. Please can you confirm. Build with #8049.
[betaflight_4.1.0_FF_PIKOF4OSD.hex.txt](https://github.com/betaflight/betaflight/files/3098676/betaflight_4.1.0_FF_PIKOF4OSD.hex.txt)

Rex700, 2019-04-19:
awesome @MJ666  thankyou i will try in less thanm an hour and tell here

Rex700, 2019-04-19:
@MJ666 tested on the bench and all works! tomorrow remaiden and we wil see

thankyou!

woobie4, 2019-04-30:
Hey guys I just flashed bf4.0 on a pikoosd, and I have the same issue too. Will this new file be commuted so when I load online from the configurator it will work? If not do I just download the hex.txt here and load it to the fc using the configurator? 
Thanks 
Jeremy

woobie4, 2019-05-01:
NM looks like this is being targeted for the next milestone. Furious dropped these boards anyway I just liked them for hex builds but we have other 20x20 options so no buggy.

MJ666, 2019-05-01:
@woobie4 yes you can donnload the file from here, remove the additional .txt extension and flash with the configurator. Feedback wold be helpful. The board is discontinued by FFPV but there is a reasonble amount out in the field. Finally this fix will go into the next release and the 4.0.x maintenance branch.

MJ666, 2019-05-04:
@Rex700 can you confirm the order of motor pins 2 and 4 has been swapped between 3.5.x and the 4.1.0 build i send you. Thanks.

woobie4, 2019-05-04:
Yes after I flashed with your new code, but did not have to remap any motors. The only thing I had to do was use blheli again to alter the rotation of one. I’m using a spedix 25 amp blheli_s 4 in 1 on this build so only Dshot600. But I’m sure blheli_32 would work good too. My esc is orientated correctly in that each signal wire is connected to a the corresponding esc number. Works great thank you so much.

PR #8049 (2019-05-04): Furious FPV PIKOF4OSD motor timer fix
Fixes #7928. Tested at FFPV.
closed 2019-05-04 by mikeller