betaflight/betaflight #7940: Kakute F7 wont go into DFU mode

JustinBluntFPV, 2019-04-08:
when i hit flash the light goes faint green and it disconnects from the computer port without switching to dfu mode.  i was swapping motor locations in RESOURCE and might have messed up the pin resources.  i think if i took a fresh DUMP (LOL) from a working Kakute F7 maybe that would fix things. what do you think? can anyone post one? thanx

etracer65, 2019-04-08:
Use the boot button to put it in DFU mode. Follow the directions in the "Recovery / Lost communication" section of the Firmware Flasher tab.

Also make sure you don't have any connected receivers or GPS modules power. If they're "talking" to the flight controller they can interfere with entering DFU.

dadoribeiro, 2019-04-08:
Connecting pushing the boot button does not put in DFU mode?

JustinBluntFPV, 2019-04-08:
No luck using the boot button either. 

-Justin

etracer65, 2019-04-08:
Did you disconnect any receivers and/or gps modules (or make sure they have no power) as I mentioned? Also, did you install the necessary drivers as mentioned on the Welcome tab?

> Latest STM USB VCP Drivers can be downloaded from here
Latest Zadig for Windows USB driver installation can be downloaded from here

Also the ImpulseRc driver fixer is useful for helping fix connectivity issues *after* you've installed the drivers.

mikeller, 2019-04-09:
@JustinBluntFPV: I am finding it hard to get the KAKUTEF7 (bare) into boot mode as well - even with the boot button pushed. Several repeated attempts of unplugging / plugging with the boot button pushed usually gets me there. I suspect the design of their boot circuitry isn't quite right.

jflyper, 2019-04-09:
@mikeller If the board is inherently difficult to go into the boot loader mode, I suspect a bad reset timing case. I experienced this on a H7 prototype board; sampling of BOOT pin seems to fail if the pin is sensed before Vdd is fully stable. A RC delay circuit on NRST pin solved the problem, which almost all the flight controller does not have; they usually connect NRST directly to Vdd and rely on Vdd startup timing for reset timing.

mikeller, 2019-04-09:
@jflyper: Could well be - the target uses a STM32745XG, which is also relatively uncommon, well possible that this MCU is particularly sensitive to BOOT pin timing.

AndersHoglund, 2019-04-09:
Is this related to the known unsolved problem with SPRACINGF7DUAL? A generic F7/H7 problem?
#6159

jflyper, 2019-04-09:
@AndersHoglund : I think they are unrelated, as #6159 is mainly about invocation of the boot loader on software reset (well, there are some talks about cap on reset, which might be related in terms of how reset "circuit" should be implemented.)

closed 2019-05-16 by stale[bot]

ricardodeazambuja, 2020-02-10:
I was also having problems to boot it into DFU mode and I solved it by simply disconnecting the battery and powering the board only from the micro USB.