betaflight/betaflight #7951: Cannot set receiver mode to PWM

Scalle0479, 2019-04-09:
**Describe the bug**
I cannot set the Receiver Mode to PWM RX input.
In betaflight 3.5.7 everything is fine.

**To Reproduce**
In the configuration tab, in the receiver window, I open the dropdown menu, set the mode to PWM RX input and hit the "Save and Reboot" Button. When the FC is rebooted, the receiver mode dropdown is set back to  PPM RX input. Alternatively, in CLI, when I type feature RX_PARALLEL_PWM, I get the reply : "Enabled RX_PARALLEL_PWM". After "save" the FC reboots. In configuration tab the dropdow is back to PPM en in when I type feature in CLI, I get "Enabled: RX_PPM"

**Expected behavior**
PWM should remain enabled so that I can fly my quad

**Flight controller configuration**
I didn't change anything, except for the receiver mode. 
```
# diff
# Betaflight / SPRACINGF3 (SRF3) 4.0.0 Apr  3 2019 / 14:27:16 (22b9f3453) MSP API: 1.41 / FEATURE CUT LEVEL 6
batch start

profile 0

rateprofile 0

batch end
```

watageek, 2019-04-09:
I think it was dumped in the feature cut level for f3. You might need to ask help on the facebook / rc group/ any forum to have a feature reenabled. You can do it with cygwin or linux easely.

Scalle0479, 2019-04-09:
Thanks, I'll try the FB group.

swilkens, 2019-04-10:
If I'm not mistaken, you can add `#define USE_PWM` to the `target.h` file of the `SPRACINGF3` target and re-compile.

It should fit in flash:
```
Linking SPRACINGF3
Memory region         Used Size  Region Size  %age Used
           FLASH:      249250 B       252 KB     96.59%
    FLASH_CONFIG:          0 GB         4 KB      0.00%
             RAM:       32884 B        40 KB     80.28%
             CCM:          2 KB         8 KB     25.00%
       MEMORY_B1:          0 GB         0 GB     -1.#J%
```

Scalle0479, 2019-04-10:
This works! Thanks a lot!!

swilkens, 2019-04-10:
No problem. If you have problems like this in the future, it's fairly easy to build Betaflight yourself:

 - Building in Windows 
https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Windows.md

 - Customized version
https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

Both these documents need some updating I see, I will look at that. The basic principle explained in those documents remains the same though.

If this answers your questions, please close this issue report 👍

Scalle0479, 2019-04-10:
Great! Thanks a lot!

closed 2019-04-10 by Scalle0479

AlekseyTroitck, 2019-04-21:
Да в версии 4.0.0 имеет место не выставление PWM (у меня 7 канальный приемник Futaba F607FS). Я попробовал предыдущую прошивку 3.5.7 - там все нормально.

swilkens, 2019-04-21:
@AlekseyTroitck Какой контроллер полета у вас есть?

AlekseyTroitck, 2019-04-21:
SP Racing F3

swilkens, 2019-04-21:
@AlekseyTroitck 

[betaflight_4.1.0_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/3101442/betaflight_4.1.0_SPRACINGF3.zip)

hex-plex, 2020-03-02:
I actually want to ask how are you editing the hex file if you could say that as this file is causing a throttle problem that the throttle if once increased it doesn't decrease so i would like to make the edit by my self I have spracing f3 acro

swilkens, 2020-03-08:
> I actually want to ask how are you editing the hex file if you could say that as this file is causing a throttle problem that the throttle if once increased it doesn't decrease so i would like to make the edit by my self I have spracing f3 acro

Sorry for the slow reply, this is a very old topic.

I did this modification by editing the build configuration file for this board and adding `#define USE_PWM` back to it. However, you should be aware that:

- F3 based boards are not supported in current betaflight, you would have to use an older release.

- The configuration method has changed since last year.

I think the last version you can use is probably 4.0.6 or maybe 4.1.0.

To do this, you will have to setup a private build environment and build the latest version of betaflight that still provides support for your F3 board. Find the configuration file and add PWM support back, then build.