betaflight/betaflight #7956: RSSI Channel Betaflight 3.5.6

ido555, 2019-04-10:
Problem: Rssi channel resets to disabled on reboot
i have a dys f4 v2 FC and when i try to set the RSSI channel to anything then hit refresh or reboot the drone it always goes back to disabled. 
i tried asking on reddit and some other forums no solutions so far.
i think this is a betaflight issue.

```
# version
# Betaflight / DYSF4PRO (DYS4) 3.5.6 Feb 16 2019 / 12:55:16 (c28ca22f9) MSP API: 1.40

board_name DYSF4PRO
manufacturer_id 

# name
name Martian 215

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature RSSI_ADC

# beeper
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1500 2100 0 0
aux 1 1 2 900 1200 0 0
aux 2 2 2 1400 1600 0 0
aux 3 13 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = -66,18,21
set min_check = 1035
set max_check = 1950
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 20
set serialrx_provider = IBUS
set blackbox_p_ratio = 16
set blackbox_device = NONE
set dshot_idle_value = 250
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set vbat_min_cell_voltage = 32
set small_angle = 180
set deadband = 12
set yaw_deadband = 10
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1500
set osd_vbat_pos = 2519
set osd_rssi_pos = 2483
set osd_tim_1_pos = 2145
set osd_tim_2_pos = 2113
set osd_flymode_pos = 2093
set osd_anti_gravity_pos = 2452
set osd_g_force_pos = 2465
set osd_throttle_pos = 2497
set osd_current_pos = 2486
set osd_mah_drawn_pos = 2454
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2513
set osd_stat_bb_no = OFF

# profile
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set throttle_boost = 25
set f_pitch = 55
set f_roll = 55
set f_yaw = 55

# rateprofile
rateprofile 1

set thr_mid = 100
set thr_expo = 65
set roll_rc_rate = 132
set pitch_rc_rate = 132
set yaw_rc_rate = 130
```

etracer65, 2019-04-10:
Disable RSSI_ADC on the Configuration tab. Or do it in the CLI with `feature -RSSI_ADC`.

You can't use analog RSSI and receiver channel RSSI at the same time.

ido555, 2019-04-10:
that solved it, thanks!.
hopefully now the next person that has this problem finds this.

closed 2019-04-10 by ido555