betaflight/betaflight #7959: Kakute F7 - Frsky-FPort - on BF 4.0 dont' work (UART6) ?

Mavy1704, 2019-04-10:
Today I installed BF 4.0 RC 6 with the Betaflight-Configurator 10.5.0 on my Kakute F7 FC. Previously I ran my R-XSR receiver from Frsky over BF 3.5.7 with FPort on UART6. Everything worked well before that, including telemetry and LUA script. 

I have made all settings at BF 4.0 new (no 'dump' on CLI). Now my control commands of the transmitter do not arrive in the FC. Only the RSSI can be seen on the Teranis X7S (FPort typical). Also, no telemetry comes on. Also, I tried the command 'set serialrx_inverted = ON / OFF', but to no avail. 

Then I instaled BF 3.5.7 again and everything worked again. 

Can someone help me or is this a bug? 

PS: Below I attached my two dump files from the CLI with BF 3.5.7 and BF 4.0 RC 6.

[BF.3.5.7 - BTFL_cli_GEP-KHX5_20190409_120900-BF.3.5.7.txt](https://github.com/betaflight/betaflight/files/3062092/BF.3.5.7.-.BTFL_cli_GEP-KHX5_20190409_120900-BF.3.5.7.txt)
[BF.4.0.0 - BTFL_cli_GEP-KHX5_20190409_153041-BF.4.0.0.txt](https://github.com/betaflight/betaflight/files/3062093/BF.4.0.0.-.BTFL_cli_GEP-KHX5_20190409_153041-BF.4.0.0.txt)

thenurgle, 2019-04-10:
you need to set serialrx_inverted = ON and set serialrx_halfduplex = ON

Mavy1704, 2019-04-11:
Oooooh...it was so easy. Thanks for help, it works. :-D

closed 2019-04-11 by Mavy1704

smpotter, 2019-05-07:
Not working for me, in fact the receiver no lo her connect when making these changes.

infornatico, 2019-08-23:
I've Kakute F7 connected to Frsky R-XSR with F.port on UART6 also. The receiver and the transmitter are bound together (D16 mode), green led on receiver is active. The Telemetry works fine, but my control commands on transmitter do not arrive to BF (4.0.5). I made the setting like Thenurgle told but no change. I flashed the receiver with the last F.port firmware also. Can you help me? Thankx