betaflight/betaflight #7965: Drifting of new FCs

davidT54, 2019-04-11:
After few secs my nano (75 mm) quadcopter starts to drift. Tx trimmering helps few seconds and then it starts to drift again. Accelerometrs are calibrated; the  FCs (Noxe and Matek 411) are new; the motors are not noisy and not very hot, i.e this seem not to be a vibration issue, also I try to use soft mouning.; the SG seem to be OK too; and finally all was working perfectly on cheap (no name) FC untill I try to make further PID tunning and USB connector dismounted from the board.  I substituted with "better" FCs and this drift isue started up. Other new thing I use is the Betaflight 10.4 (for no name FC I use 3.5, or 3.7)

I assume the probability that both new FC got a faulty accelerometer is very low. Search gave number of unsolved reports. Any Idea what can be wron?

Thanks, 
David

etracer65, 2019-04-11:
Assuming you properly calibrated the accelerometer, then this is commonly caused by excessive vibrations from the motors/props. The accelerometer is very sensitive to vibrations because they are in fact tiny accelerations happening very quickly. You need to capture a blackbox log to identify the issue with noise so that you can adjust the filtering as needed to reduce.

The first step is to be sure to start with the default settings for filtering and PID settings.

davidT54, 2019-04-11:
Thank you for the suggestion on black box log I will try it.

davidT54, 2019-04-14:
Mini FC boards need external to get  blackbox log, so I try all possible filter obtions. As I expected the default was the optimal. to loww or too high filtering was causing low frequency oscillations and strong motor heating. Probably 75 mm frame is too small for high rpm 1104 motors - the vibrations are too strong. I will try 90 mm frame...

davidT54, 2019-04-14:
Let me add that this one of the most common , hard to solve problem is not frequently discussed (see [)](https://github.com/betaflight/betaflight/issues/4573)

closed 2019-05-21 by stale[bot]