betaflight/betaflight #7969: Betaflight 4.0 versions bug on Matek f405 with motor speed run away..... when giving input.. when bench testing the new firmware... older 3.5.7 works fine

90000000000383737Drew, 2019-04-11:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

etracer65, 2019-04-11:
No information provided.

closed 2019-04-11 by etracer65

georiege, 2019-04-17:
I have exactly the same problem. With betaflight 4 the quad does not take off. It is arm normally but when you pitch a little the throttle, the motors spins with very high speed and i get runaway at OSD. With betaflight 3.5.x work fine. The diff that i am putting at betaflight 4 is attached.
[Bakcup17_4_2019.txt](https://github.com/betaflight/betaflight/files/3089765/Bakcup17_4_2019.txt)

etracer65, 2019-04-17:
Check the motion of the model on the Setup tab. It's likely that the board alignment is incorrect. Adjust this on the Configuration tab so that physical motion matches the movement in the Setup tab.

georiege, 2019-04-17:
Thank you  etracer65, i did that and worked fine. You where right, i had to change the gyro degrees from 270 CW to 180 CW for the drone to face correctly at setup figure. Thank you for your fast reply !

pancho303, 2019-05-12:
I had the same problem and has to do with Betaflight 4.0.2 setting this board with 270cw at default settings.  Can't call it a bug maybe, but should be changed into 180 into the next bf release to avoid runaways.