betaflight/betaflight #7971: RSSI value drops from OSD (Without RXLOSS)

pljeroen, 2019-04-11:
**Describe the bug**
During mid-flight and sometimes after battrery swap without turning transmitter off, my OSD's RSSI value drops to 0 and starts flashing. A transmitter reboot is required to regain RSSI values in OSD. Other telemetry values are unaffected.

**To Reproduce**
Random behaviour. Will fly 3 packs straight without problems, and then all of a sudden manifest again. I even loose the value mid-flight, without RXLOSS and I can just continue flying. 

**Expected behavior**
I expect this to NOT hapen, as there is no RXLOSS.

**Flight controller configuration**
'''# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Mar 26 2019 / 19:07:01 (173e958da) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name
name PLJ Rev3

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 27 3 900 1300 0 0
aux 4 13 5 1875 2100 0 0
aux 5 28 0 1700 2100 0 0
aux 6 35 2 1700 2100 0 0
aux 7 36 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_trim_pitch = -4
set acc_calibration = -12,35,13
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = KILL
set bat_capacity = 1500
set beeper_frequency = 0
set yaw_motors_reversed = ON
set yaw_deadband = 1
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 2500
set osd_vbat_pos = 2120
set osd_rssi_pos = 2138
set osd_remaining_time_estimate_pos = 2092
set osd_flymode_pos = 2113
set osd_g_force_pos = 2476
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2497
set osd_current_pos = 2127
set osd_mah_drawn_pos = 2520
set osd_craft_name_pos = 2506
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set vtx_band = 6
###ERROR: CORRUPTED CONFIG: vtx_band = 6###
set vtx_power = 2
set vtx_freq = 0

# profile
profile 2

# rateprofile
rateprofile 1

# end the command batch
batch end'''

**Setup / Versions**
 - CL RACING F4S v1.5 Betaflight 4 RC4, but happened on 3.5.6 as well.
 - r-xsr receiver with smartport hooked up as per CL RACING manual. FCC180921.frk
 - X-Lite transmitter with LUA script installed, OpenTX2.3.3

**Additional context**
It happens at 1:35: https://www.youtube.com/watch?v=NOYSGuwPBi4
[image]
[image]
[image]

I'm new to reporting bugs or issues here, so please let me know this needs more info, and in that case, what. At your service for debugging purposes.

etracer65, 2019-04-11:
Also since the flight controller is getting the RSSI signal being fed back from the TX to the flight controller, it's perfectly normal for there to be drop outs. The telemetry link from the TX with the Lua script is very low bandwidth and subject to signal dropouts. You would get better results changing to using the F.port firmware on the receiver and setting the flight controller to that instead of SBUS. F.port supplies the RSSI directly from the receiver to the flight controller. Optionally you could configure your radio to put the RSSI value on an unused aux channel and use that as your RSSI source.

pljeroen, 2019-04-11:
> Also since the flight controller is getting the RSSI signal being fed back from the TX to the flight controller, it's perfectly normal for there to be drop outs. The telemetry link from the TX with [...]

Regarding this: My other telemetry values are fine. It's only the RSSI one that drops to 0 **and will not recover** unless I restart the transmitter. Any insight on why that would be ?

> 
> 
> You have corruption in your flight controller config as noted by:
> 
> ```
> ###ERROR: CORRUPTED CONFIG: vtx_band = 6###
> ```
> 
> Upgrade to the official 4.0 release and do a full-chip erase and set up from scratch. Don't try and paste a `diff` or `dump` from a previous version - even from earlier 4.0 dev or RC versions.

I did a clean setup for this one, without any copy and paste. Flash and erase. But yeah, thats there, so I dunno how that happened, will do a fresh install. However, this issue can not be BF4 related, as I had this in 3.5.x as well.

pljeroen, 2019-04-11:
BF4.0 now, redone the complete config from scratch:
```# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr 11 2019 / 07:01:44 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name
name PLJ Rev3

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 27 3 900 1300 0 0
aux 4 13 5 1700 2100 0 0
aux 5 28 0 1700 2100 0 0
aux 6 35 2 1700 2100 0 0
aux 7 36 3 1700 2100 0 0
aux 8 39 4 1925 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -17,38,38
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 10
set failsafe_switch_mode = KILL
set bat_capacity = 1500
set yaw_motors_reversed = ON
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_vbat_pos = 14409
set osd_rssi_pos = 14425
set osd_remaining_time_estimate_pos = 14380
set osd_flymode_pos = 14402
set osd_g_force_pos = 14765
set osd_throttle_pos = 14658
set osd_vtx_channel_pos = 14786
set osd_current_pos = 14415
set osd_mah_drawn_pos = 14807
set osd_craft_name_pos = 14795
set osd_stick_overlay_left_pos = 14661
set osd_stick_overlay_right_pos = 14674

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end```

pljeroen, 2019-04-11:
Issue remains after upgrade to 4.0 final.
Since rebooting my X-lite fixes this, should I be posting on opentx or the Lua script githubs?

etracer65, 2019-04-11:
If rebooting your transmitter fixes it then my guess would be that the Lua script has crashed (or has otherwise stopped running). You could confirm that no other Lua functions work after the problem starts.

I would suggest opening an issue against the Lua scripts repo. The  OpenTX guys wouldn't know anything about the Betaflight Lua scripts. It may turn out that there's some incompatibility with the X-lite but too early to tell.

pljeroen, 2019-04-11:
Reason for me posting here is that I don't see how the Lua script could impact a pure fc function like displaying the rssi value from smartport telemetry on OSD, or am I not under standing the dependencies and how it actually works?

etracer65, 2019-04-11:
The Lua script runs continuously in the background on your radio and periodically sends the RSSI value back to the flight controller over telemetry. So if the RSSI stops that's a good indication that the Lua script has crashed or otherwise stopped running.

etracer65, 2019-04-11:
If instead you used an actual RSSI source between your receiver and flight controller then the Lua script would have no impact. Examples would be analog RSSI, receiver outputting RSSI on a channel, or F.port.

pljeroen, 2019-04-12:
@etracer65 which route would you recommend, fport?

BradB111, 2019-04-12:
> @etracer65 which route would you recommend, fport?

Sorry to jump in, sounds like something isn't set up correctly but not 100% sure as you say it works fine for a given time (very odd). If you are sending RSSI as a channel value to the FC there shouldn't be any LUA involvement to my knowledge.

I personally use F.port over SBUS whenever possible. for RSSI it sort of cuts out the middleman. the Rx sends the FC the RSSI value directly without needing to map a channel. That RSSI would the be sent via Telemetry as normal to your TX like voltage would for example as opposed to an analogue channel value.

This also can help to reduce any errors as a result of incorrect configuration. that may result in OSD showing the wrong value (like in your case) or the wrong value being sent back to your Tx etc...

I'd say give F.port a go and see if you experience the same. Make sure you remove any analogue RSSI channel settings in your TX and remove the channel selection for RSSI in the configuration on the RC tab so its not trying to read a value that now doesn't exist. (or clean flash, F.port shouldn't require any configuration such as mapping channels etc..)

Duplicate the model in your Tx so you can always revert back.

pljeroen, 2019-04-12:
No sensor lost, and not always repeatable either. It just so happened to be near myself this time. Other times it's when I do a flip.

I'm certain it is configured correctly for this setup. I would need to resolder a couple of wires to get f.port running, I could do that. 

There is no rssi channel setup atm, as I think this is using the Lua script ro get rssi back to my OSD. I would not know how to setup a channel for it either to be honest.

All other telemetry is fine, it's just the OSD value going into an error state or something. I suspect a curropt packet messing things up or something.

pljeroen, 2019-04-12:
Or my X-lite crashing the Lua script, or vise versa.

BradB111, 2019-04-12:
> Or my X-lite crashing the Lua script, or vise versa.

Possibly. Setting up a channel to be sent back to the rx showing RSSI value is fairly simple to do once you understand whats going on (couldn't get my head round it at first I will admit).
Below is a tutorial on how to achieve this.
https://www.youtube.com/watch?v=t-evOAS9Mkg

I would still recommend F.port if its an option for you. Loads of benefits besides the RSSI being simpler to set up also.

It can be tricky to get F.port working on an F4 board. I am not personally familiar with using the CLRacing F4. You need to locate a bi-directional UART, depending on if its inverted or non-inverted will determine if you need to use the non-inverted pad on your Rx, remember to set Rx inversion and half duplex accordingly in CLI if necessary. Bit of trial and error sometimes required until you get the hang of what options do what:

set serialrx_halfduplex = ON/OFF
set serialrx_inverted = ON/OFF

For example Matek brand F4 boards it always seems to be UART 3 and its non-inverted so I solder to the TX3 pad from the non inverted F.port pad (would be the non inverted smartport pad when F.porn firmware not in use). Of course with F.port its only 1 wire so everything is done through the TX3 pad in my case.

Here is a good guide on setting up / using F.port:
https://oscarliang.com/setup-frsky-fport/

pljeroen, 2019-04-12:
If I can leave my sbus wire connected, with uart1 disabled, and just change uart3 to fport (it having a buildin hardware inverter on this board, just as the sbus pad), I should be able to just flash the receiver with the fport version and make the BF changes, right ?

(sorry for using this as a help chat unrelated to a BF bug .. I will be closing this at the end of this)

etracer65, 2019-04-12:
Since you already have Smartport telemetry working on UART3 (uninverted SPORT pad on receiver connected to TX3), all you need to do is flash your receiver with the F.port firmware. Then change UART3 on the Ports tab to have the Serial Rx instead of UART1. Finally switch the serial receiver protocol on the Configuration tab to FrsSky FPort. You can leave the SBUS connection on RX1 if you want but it won't do anything.

BradB111, 2019-04-12:
> If I can leave my sbus wire connected, with uart1 disabled, and just change uart3 to fport (it having a buildin hardware inverter on this board, just as the sbus pad), I should be able to just flash [...]

Exactly right. + @etracer65 's suggestion of leaving the SBUS connected may be a good idea at least for now if you need / want to revert back for whatever reason.

Let us know how it goes.

pljeroen, 2019-04-14:
Transition was painless. So far, a few bench tests proved promising. 

I disabled smartport for uart3, enabled rx on it, disabled it on 1, and switched sbus to frsky fport for the receiver type.

pljeroen, 2019-04-21:
Issue gone using f.port! Thanks for the suggestion guys!

closed 2019-04-21 by pljeroen