betaflight/betaflight #7973: Buzzer madness on F3 brushed

nightflasher, 2019-04-11:
On betaflight 4.0.0 (and RC6, didn't test more) the buzzer goes crazy when the kwad is armed. It stops beeping when i raise the throttle and sometimes it stops by adding Airmode (on a switch). 

I do test new versions on my brushed kwads, because they don't harm as much as brushless (my smallest one is 3") one.

First time I had this issue was while the kwad was connected to the configurator and it said "accelerometer calibration", but that can't be true and I couldn't find any option to turn off. Even turning the Buzzer for those features didn't stop the buzzer madness.

swilkens, 2019-04-11:
A constant beep after arming is a concequence of the FC being armed while you have "motor stop" enabled, using DSHOT. This beeping stops when the motors start turning, which also explains why airmode will cause it to stop.

[image]

You can disable that here:

[image]

nightflasher, 2019-04-11:
Yeah, but you missed that I am on BRUSHED no DSHOT!

swilkens, 2019-04-11:
MOTOR_STOP is available on all protocols. I also assume this feature will also cause a beep when the motors don't spin, regardless of motor stop, while the quad is armed?

Have you tried disabling the ARMED beeper feature?

mikeller, 2019-04-11:
As @swilkens says, this is by design, if the beeper mode 'ARMED' is enabled. This probably does not make much sense for brushed setups, so 'ARMED' should not be enabled for brushed setups if 'MOTOR_STOP' is enabled.

nightflasher, 2019-04-12:
Found the solution: disabled motor_stop
Well, this is not what I wanted to have.

Next step: raise minimum throttle when armed
Another thing I didn't want to have. But if this is the only way to get it working like my brushless ones, it's acceptable somehow. 

Thanks everybody, back to work.

swilkens, 2019-04-12:
If you disable the 'ARMED' beeper feature you can keep 'MOTOR_STOP' enabled without the beeping.

I assume you just don't want to hear the beep, but keep the motors stopped while armed at 0 throttle. Unless I misunderstood.

JumaAnher, 2019-05-08:
Hello
I have this issue in two different boards: CrazybeeF4DX and MATEK F405.
I originally used the Crazybee with Betaflight version 3.5.5, and when I updated to version 4.0.0 I began to suffer this issue.
I have confirmed that the continuous beep (beeeeeeeeeeeeeeeeeeeeeeeep... not beep...beep...beep) is caused by the 'ARMED' beeper feature, and if I disable it or I disable MOTOR_STOP, the buzzer does not beep when ARMED.
This feature (ARMED) had a different behavour in version 3.5.5, and the beeping was  not continuous, it was beep... beep... beep... until the motors begin to run.

I updated the MATEK F405 to he 4.0.0 version and observed the same behavour: when ARMED and MOTOR_STOP, continuous beep. Of course, in version 3.5.5 there's no continuous beep.

Is there any reason for this change?

Best regards

swilkens, 2019-05-08:
@JumaAnher This is a safety feature to alarm the user to the fact that his / her quad is armed. Since the props are not spinning with `MOTOR_STOP` a user might not realize the quad is armed. If he / she then approaches the quad, bad things might happen.

nightflasher, 2019-05-08:
But, seriously! It's rare to have a buzzer these days (because of dshot commands), this feature needs another solution. On the other hand, call it 'natural selection'! Without a buzzer and motor_stop feature (without air mode), there isn't any beep from the motors with dshot commands.
Btw.: by selecting any dshot protocol, I think the dshot beacon should be enabled by itself.
Maybe an option to turn off and remind to have a buzzer installed.

JumaAnher, 2019-05-08:
Ok, I can understand that this is a good reason @swilkens but I think that, as mentioned @nightflasher , there must be a better way to warn about this than this.
For example, I've just disabled the ARMED feature because of this, but I'm not confident without an ARMED signal; I'm not very experienced with betaflight, and don't know if it would be possible to configure the led strip to warn about this. I'll investigate this possibility.

Regarding to the DSHOT commands, I had it configured, but in the Trashcan the sound is very low so I finally added a buzzer.

Thank you for your answers and best regards.

closed 2019-06-15 by stale[bot]