betaflight/betaflight #7974: 4.0 Tuning Notes

scon, 2019-04-11:
I tried to apply the suggestet settings for 7" drones.

set dyn_lpf_gyro_min_hz = 70
set dyn_lpf_gyro_max_hz = 350
set dyn_notch_range = LOW
set dyn_notch_min_hz = 70
set dyn_lpf_dterm_min_hz = 70
set dyn_lpf_dterm_max_hz = 150
set dterm_lowpass2_hz = 120
set d_yaw = 0

But i get an error that this value is out of range.

# set dyn_notch_min_hz = 70
###ERROR: INVALID VALUE###
Allowed range: 100 - 1000

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

atomiclama, 2019-04-12:
@scon  Is that from the wiki? Just correct it.

ctzsnooze, 2019-04-12:
My bad.

When I wrote that wiki entry I didn't realise the CLI check limited the lowest value that could be entered was 100hz.  

Hopefully we can get that lowest possible limit set to 60hz.  In the meantime I'll fix the wiki note to say 100hz.

See:

https://github.com/betaflight/betaflight/pull/7984

haslinghuis, 2022-05-01:
Fixed in #7984

closed 2022-05-01 by haslinghuis