betaflight/betaflight #7975: Add Angle/Horizon settings to Rate Profiles

JustAskDave, 2019-04-12:
**Is your feature request related to a problem? Please describe.**
I am setting up 140mm micro FPV racing drones for school students. I have set up a Beginner/Intermediate/advanced switch that changes between 3 Rate Profiles. This is set to adjust TX Rates, TX Expo and Thottle Max position (now in BF4.0!!) to tame them down while they get used to it (first time flyers)

However, I would like to include the max angle in Angle mode in each switch position, ie, 25° for beginner, 40° for intermediate and the original 55° for advanced.

**Describe the solution you'd like**
I would love for these to be included in the Rate Profile instead of the Profile, as it currently means a switch for one and stick combos to change the other

**Describe alternatives you've considered**
If the Rate Profile could be linked to the Profile (ie, Profile 1 uses Rate Profile 1, Profile 2 uses Rate Profile 2 etc) then I could just use the stick combo method to swap between, not ideal since there is no visual indication.

**Additional context**

mikeller, 2019-04-14:
This is related to #2452 - what is needed is a general overhaul of how the profiles are structured.

JustAskDave, 2019-04-14:
Ahh.. ok, so no solution at this point in time.   Cool, was work asking.

Just read through #2452 and it mentions it needs to reboot to change the Profiles?  Landing, using the Stick Commands to change Profiles, then take off again instantly - the FC does not appear to reboot nor is it down for any noticeable length of time,, is it still rebooting as described back then?

mikeller, 2019-04-14:
@JustAskDave: I think the rebooting part has been dropped in the meantime. But it still re-initialises a bunch of filters, so allowing PID profile switches while flying could potentially lead to crashes even if the before / after filters were flyable in principle.

JustAskDave, 2019-04-14:
@mikeller - couldn't it just be set so that the profiles could not be changed when armed? It's currently like that anyway with the stick commands, just ignore switch commands if already armed.

mikeller, 2019-04-14:
@JustAskDave: That's definitely a possibility - one potentially problematic side effect of this would be that a profile switch would not come into effect until the pilot disarmed, at which point in time they probably will have forgotten that they tried to switch profile.

As I said, what is needed is a general overhaul of 'what goes where' and 'how is the UX for this' for all things profile - this aspect should be part of the big picture.

readej, 2022-01-23:
I was looking for this as well. I have similar need, and would like to change amount of angle limit via a switch. Even if it could only be done when disarmed that would still work for me but would not be the preferred solution.

How do I change profiles with stick commands now? Also how would I know what profile I'm in? The drone I'm using does not have FPV.

Regarding @JustAskDave 's need to "tame" down the beginner pilots, what does work right now is setting the "rate profiles" via  a switch in the adjustments tab. You can setup 3 rate profiles with different levels of throttle limit percentages. like 60%, 80%, 100% for the beginner, intermediate, and advanced pilots.

I'm trying to do a combo of the two things, setting both angle limit and throttle limit at the same time with a switch.

JustAskDave, 2022-01-23:
Hey John,Yep, that's exactly what I have done, throttle limit as a percentage so the stick still has full travel combined with rates changes.Then the profiles I have max angle set to 15 degrees on profile 1, 35 degrees on profile 2 and 50 degrees on profile 3.To change with sticks is left stick to bottom left and hold, right stick left for profile 1, right stick forward for profile  right for profile 3.Cheers, Dave. David Baker0429 328 432http://www.justaskdave.com.auhttps://au.linkedin.com/in/justaskdaveSent from my Samsung Galaxy smartphone.

closed 2026-09-26 by blckmn