betaflight/betaflight #7979: Smart Audio 2.1 Unify Evo not working

jtroutt19, 2019-04-12:
Flashed BF 4.0.0 release tried to use osd menu to switch VTX channels. Does not function

```# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Apr 11 2019 / 07:20:40 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name
name 7hatGuy

# resources
resource MOTOR 1 B01
resource MOTOR 2 E11
resource MOTOR 3 B00
resource MOTOR 4 E09

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 2 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 1650 0 0
aux 1 46 0 1950 2100 0 0
aux 2 13 1 1975 2025 0 0
aux 3 26 1 1300 1600 0 0
aux 4 35 2 1950 2100 0 0

# adjrange
adjrange 0 0 3 1475 1525 29 3 0 0

# rxrange

# vtx

# rxfail

# display_name
display_name N9JTT

# master
set gyro_lowpass_type = PT1
set dyn_notch_width_percent = 0
set mag_hardware = NONE
set rssi_scale = 200
set rc_smoothing_derivative_hz = 150
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set min_throttle = 1000
set dshot_bidir = ON
set use_unsynced_pwm = ON
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 16000
set current_meter = ESC
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 1500
set osd_alt_alarm = 1000
set osd_vbat_pos = 14720
set osd_rssi_pos = 14732
set osd_tim_2_pos = 6519
set osd_vtx_channel_pos = 2104
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 6551
set osd_mah_drawn_pos = 14743
set osd_craft_name_pos = 10251
set osd_display_name_pos = 4448
set osd_gps_speed_pos = 6163
set osd_gps_lon_pos = 4097
set osd_gps_lat_pos = 4129
set osd_gps_sats_pos = 6159
set osd_home_dir_pos = 6509
set osd_home_dist_pos = 6510
set osd_altitude_pos = 4119
set osd_warnings_pos = 14377
set osd_esc_tmp_pos = 2048
set osd_stick_overlay_left_pos = 8423
set osd_stick_overlay_right_pos = 8430
set osd_stat_max_alt = ON
set osd_profile = 3
set debug_mode = FFT
set scheduler_optimize_rate = ON
set vtx_band = 1
set vtx_freq = 5865
set displayport_max7456_wht = 1

# profile
profile 2

# rateprofile
rateprofile 1

# end the command batch
batch end

# 
```

Kakute F7 TBS unify EVO with most current firmware flashed.

moonlitfpv, 2019-04-19:
My Smartaudio doesn't work either after updating to 4.0. TBS Unify Pro HV Race

matbhe, 2019-04-23:
My Smartaudio doesn't work either after updating to 4.0. TBS Unify Pro 5G8 HV Race 2

PR #8276 (2019-05-25): [VTX] SmartAudio conversion to VTX table and V2.1 implementation
This PR aims to further issue #7299 by transitioning SmartAudio to use the new VTX table facility introduced in #7251 and using it to implement SmartAudio V2.1. It should also fix #7979 and #7336.

The following features have been implemented:
- PIT mode enable and disable without power cycle
- dbm-based power selection
- persistant RACE and FREE opmodels that cause the vtx to power up in PIT mode or normal mode

At this time, vtxtables are NOT set up automatically. Therefore the user has to manually set them correctly.

For TBS  [... cut]
closed 2019-05-25 by mikeller